esphome/feature-requests #189: Add LED driver PCA9633

gordonthree, 2019-04-23:
**Describe the problem you have/What new integration you would like**

Requesting support for the NXP PCA9633 four channel LED driver. It is much simpiler device than the PCA9685 and only has four outputs rather than 16. Also it is only 8 bit instead of 12 bit. 

The datasheet can be found here:
https://www.nxp.com/products/analog/interfaces/ic-bus/ic-led-controllers/4-bit-fm-plus-ic-bus-led-driver:PCA9633

**Please describe your use case for this integration and alternatives you've tried:**

I am trying to reverse engineer the existing PCA9685 module but not having much luck yet. 

**Additional context**

I have written a simple Arduino library for the chip here. It works on ESP8266 and some Arduino boards.

https://github.com/gordonthree/pca9633

max246, 2023-05-17:
@gordonthree  is this still neded?

gordonthree, 2026-01-19:
wow, a blast from the past. if anyone is still using PCA9633 in the wild, I'd be happy to help get it working in ESP Home.

mcassaniti, 2026-01-20:
> wow, a blast from the past. if anyone is still using PCA9633 in the wild, I'd be happy to help get it working in ESP Home.

I've added a thumbs up. If anyone else is using a PCA9633 I'd suggest a thumbs up on the quoted post.