esphome/feature-requests #193: Alternative way to handle CWWW led bulb

NutTom, 2019-04-25:
**Describe the problem you have/What new integration you would like**

I bought couple of "SL06" cold white/warm white led bulb, instead of having 2 parallel channels for cold and warm they put them in serial, from electronic point of view one mosfet transistor is connected to both warm led and to second mosfet transistor controlling cold led. It's impossible to light cold led alone, it's impossible to bright cold led more than warm led. To resume, there is a PWM channel for overall brightness and a second PWM channel too "cool down" the light temperature.

**Please describe your use case for this integration and alternatives you've tried:**

Actual workaround is to declare the cwww light bulb as a RGB light bulb with the following setup:
red channel for PWM1 (the overall brightness)
green channel for PWM2 (the cold led)
blue channel to dummy pin (not wired)

When setting the light color to 100% red, 0% green, 0%blue I have the warmest possible color.
By adding green without changing red I'm cooling down the light.
The brightness control work fine in this case.

**Additional context**

xdss, 2019-09-14:
Was there any progress on this. Having the same issue with the Briliant downlight, GPOIO5 is brightness and GPIO13 is CW/WW. Not sure how to add them neatly into a component other than the hack above.

https://blakadder.github.io/templates/brilliant_downlight.html

OttoWinter, 2019-10-21:
@xdas Well there's some progress as in you can now create custom light components (https://esphome.io/components/light/custom.html) easily that can support any possible configuration you want.

xdss, 2019-10-27:
thanks, I did try to go down this path but I think at the moment it is a step too far for me.

dieselrabbit, 2019-12-03:
Another use case for this is the newest revision of the Feit Smart bulb BPA800/RGBW/AG/2(P), where GPIO4 controls the overall white brightness, and GPIO5 controls the color temp, or mix between CW/WW.

eidorb, 2019-12-15:
Similarly, the [Arlec Grid Connect CWWW](https://blakadder.github.io/templates/arlec_GLD112HA.html) bulb uses GPIO5 for brightness, and GPIO13 for colour temperature. 

I will try the hack above for now.

ozeraser, 2020-01-12:
I built a custom light output for these CWWW bulbs with a temp and brightness pin, unlike the standard CWWW pinout. Give it a go.

https://github.com/ozeraser/ESPHome/tree/master/CWWW%20Custom%20light%20Tuya_Arlec%20Grid%20Connect

dieselrabbit, 2020-01-12:
I was finally able to wrap my head around creating a custom light component for the Feit Smart Bulb. I changed it up a little bit so that when the user sets the light to pure white, it'll change over the the CWWW leds and turn off the RGB leds (and vise-versa) so both are not on at the same time. The tricky part for me was cross-fading the RGB with the White without access to the progress from the LightTransformer. More a proof-of-concept for me than something ready for public consumption (not commented), but don't know how much more I'm actually going to get to do to it so I posted it anyways.
https://github.com/dieselrabbit/FeitGen2_ESPHome

eidorb, 2020-01-12:
> I built a custom light output for these CWWW bulbs with a temp and brightness pin, unlike the standard CWWW pinout. Give it a go.
> 
> https://github.com/ozeraser/ESPHome/tree/master/CWWW%20Custom%20light%20Tuya_Arlec%20Grid%20Connect

Nice write-up, @ozeraser.

NutTom, 2020-01-12:
> > I built a custom light output for these CWWW bulbs with a temp and brightness pin, unlike the standard CWWW pinout. Give it a go.
> > https://github.com/ozeraser/ESPHome/tree/master/CWWW%20Custom%20light%20Tuya_Arlec%20Grid%20Connect
> 
> Nice write-up, @ozeraser.

Nice ! You are my hero ! Now I understand the new custom way to properly implement this kind of smart bulb ! Thanks you !

rcruikshank, 2023-10-10:
> I built a custom light output for these CWWW bulbs with a temp and brightness pin, unlike the standard CWWW pinout. Give it a go.
> 
> https://github.com/ozeraser/ESPHome/tree/master/CWWW%20Custom%20light%20Tuya_Arlec%20Grid%20Connect

I tried to use your code but I'm getting the following errors. Not sure what I'm doing wrong:

_Compiling .pioenvs/test-down-light/lib01f/ESPAsyncTCP-esphome/SyncClient.cpp.o
In file included from src/main.cpp:34:
src/CWWWSinglepintempandbrightnessLightOutput.h: In member function 'virtual esphome::light::LightTraits CWWWSinglepintempandbrightnessLightOutput::get_traits()':
src/CWWWSinglepintempandbrightnessLightOutput.h:74:16: error: 'class esphome::light::LightTraits' has no member named 'set_supports_brightness'; did you mean 'get_supports_brightness'?
   74 |         traits.set_supports_brightness(true);
      |                ^~~~~~~~~~~~~~~~~~~~~~~
      |                get_supports_brightness
src/CWWWSinglepintempandbrightnessLightOutput.h:75:16: error: 'class esphome::light::LightTraits' has no member named 'set_supports_rgb'; did you mean 'get_supports_rgb'?
   75 |         traits.set_supports_rgb(false);
      |                ^~~~~~~~~~~~~~~~
      |                get_supports_rgb
src/CWWWSinglepintempandbrightnessLightOutput.h:76:16: error: 'class esphome::light::LightTraits' has no member named 'set_supports_rgb_white_value'; did you mean 'get_supports_rgb_white_value'?
   76 |         traits.set_supports_rgb_white_value(false);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                get_supports_rgb_white_value
src/CWWWSinglepintempandbrightnessLightOutput.h:77:16: error: 'class esphome::light::LightTraits' has no member named 'set_supports_color_temperature'; did you mean 'get_supports_color_temperature'?
   77 |         traits.set_supports_color_temperature(true);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                get_supports_color_temperature
Compiling .pioenvs/test-down-light/lib01f/ESPAsyncTCP-esphome/tcp_axtls.c.o
Archiving .pioenvs/test-down-light/lib01f/libESPAsyncTCP-esphome.a
Compiling .pioenvs/test-down-light/libaf0/Hash/Hash.cpp.o
*** [.pioenvs/test-down-light/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.24 seconds ==========================_

My yaml is attached.
Thanks in advance.
[brilliant.txt](https://github.com/esphome/feature-requests/files/12851842/brilliant.txt)