esphome/feature-requests #201: Nextion sensor

bigstation17, 2019-04-29:
**Describe the problem you have/What new integration you would like**

Possibility to read value from nextion slider component

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I would like to be able to use nextion slider for dimming light component in esphome.

Andoramb, 2019-10-02:
I have a 7inch display, I would gladly participate in testing if needed

Andoramb, 2019-11-19:
Sorry to bump this. How feasible would be to implement this feature? Thanks for the work!

gianakmal, 2019-12-03:
its clear ?

petosak, 2020-03-29:
I need this feature too.

rspaargaren, 2020-04-23:
I am working on an extention of the current Nextion code within ESP-Home to enable sensor input and a switch based on dual state button. Anybody wants to test the code?

Andoramb, 2020-04-23:
That's great to hear! Unfortunately I cannot test it atm, but please keep us updated :)

petosak, 2020-04-23:
that sounds great. i have available some EPSs and 2 Nextion displays. I can try.

rspaargaren, 2020-04-26:
Ok I have added a sensor and a switch to the nextion package. unfortunately some code has to be added to the item in Nextion as well when creating to enable it. You can have a preview and please test on my github:
https://github.com/rspaargaren/esphome/tree/NextionUpdate/esphome/components/nextion
I have added a README with some explanation. Looking forward for your feedback.

I have also tried to work on a light etc. but this is not following ESP HOME philosophy. So maybe a binary output and float output with UART output to Slider. But basically this can already been done in the DISPLAY with the labda.

mirkokrajco, 2020-05-02:
rspaargaren thank  you. 
It is good job.

rspaargaren, 2020-05-02:
Thanks I will create the related pull request and update docs soon so it will be integrated in the dev env.

wernight, 2021-01-04:
Let me briefly comment that this is a pretty important feature. The bug when setting colors (extra quotes) can be worked around using the `send_command_printf` (which arguably is actually easier as it's what you put in Nextion debugger), however this one has no workaround. Sliders and retrieving values is currently impossible. Also knowing on which page you are is nearly impossible.

Andoramb, 2021-01-05:
@wernight, knowing on which page you are can be done by [this ](https://github.com/Andoramb/Nextion-wall-mount/blob/881dc3802800e97aa4b5edf9be3a9ae5b0b33792/YAML/NextionDisplay.yaml#L323)method

rspaargaren, 2021-01-05:
The PR I have created is enabling you to get the values of the sliders, etc. But at this moment an more improved code is developed by Senex which makes it even more flexible! Reading which page you are on is as easy as sending the page number as a variable to esphome or polling the variable from the esp. (In the new code)

rspaargaren, 2021-01-12:
Can everybody check the New Pull request:
https://github.com/esphome/esphome/pull/1464
This is a combination of various pulls in relation to the Nextion. The more activity and feedback the faster it gets pushed.