esphome/feature-requests #202: generic_thermostat implementation of climate component

DutchDynamics, 2019-04-29:
**Describe the problem you have/What new integration you would like**
Implementation of climate component is new, and only Bang Bang variant is available.
The Bang Bang climate component is a two way control. 
Looking for the 1 way thermostat

**Please describe your use case for this integration and alternatives you've tried:**
Trying to use ESPHome to automate/integrate the control of my jacuzzi into HA. The Jacuzzi only has a heater and temperature sensor. Tried to use the new Bang Bang climate implementation, but some functions are not available and only need the one point setup. Now used ESPHome to expose temperature sensor and switch for the heater. Configured the generic_thermostat in HA. 

**Additional context**

glmnet, 2019-04-30:
Actually this should be the `template_climate` component.

OttoWinter, 2019-05-01:
> The Bang Bang climate component is a two way control.

It is not, the bang_bang climate platform supports both one-way and two-way control (by omitting `heat_action` or `cool_action`).

The bang_bang climate platform is actually very much like generic_thermostat, just with a different (and IMO better) config interface.

closed 2019-05-01 by OttoWinter

glmnet, 2019-05-01:
Shouldn’t it expose in the traits if it supports heat and/or cooling?