esphome/feature-requests #203: Add password to ESPHome dashboard

ReneHonig, 2019-04-30:
**Describe the problem you have/What new integration you would like**

Thanks for a great development!
Would it be possible to improve the security of the dashboard, by adding the option to require a password to access the dashboard? 
**Please describe your use case for this integration and alternatives you've tried:**

It is great that this option exists for OTA updates, but access to the dashboard also provides access to the Yaml files with the password data.
**Additional context**

OttoWinter, 2019-05-01:
Already exists, see `esphome <folder> dashboard -h` - the `--password`option.

closed 2019-05-01 by OttoWinter

vamichael, 2019-05-10:
> Already exists, see esphome <folder> dashboard -h - the --passwordoption.

@OttoWinter I don't understand what you mean by this.  Is password protecting the dashboard documented somewhere?  I'm using the docker `esphone/esphome` image, so perhaps this is different than using the non-docker version.

OttoWinter, 2019-05-10:
For docker the command is more or less the same, just substitute `esphome` at beginning with `docker run [...] -it esphome/esphome`. so for docker compose it just means adjusting the command option.

vamichael, 2019-05-13:
> For docker the command is more or less the same, just substitute `esphome` at beginning with `docker run [...] -it esphome/esphome`. so for docker compose it just means adjusting the command option.

Could you provide a sample docker-compose?  I’ve spent a couple of hours trying and still haven’t had any luck.  Much appreciated.