esphome/feature-requests #204: device_tracker vs. binary_sensor in esp32_ble_tracker

ReneHonig, 2019-04-30:
**Describe the problem you have/What new integration you would like**

Thanks for a great development!
Would it be possible to add a device _tracker entity type for the` esp32_ble_tracker` component (in addition to the binary_sensor)? That way the integration with other tracker functionality in HA (like persons) would work better.
**Please describe your use case for this integration and alternatives you've tried:**

current setup does not integrate well with HA device_tracker functionality.
**Additional context**

OttoWinter, 2019-05-01:
Yeah I know. However last time I looked at the device_tracker in HA it looked really badly architected (can't remember the exact reasons, but it looked a lot like features were bolted on top of each other).

So for that to get into ESPHome, that architecture would first need to be untangled a bit.

kirichkov, 2019-06-12:
I came across the same problem, and here's my workaround:

The home assistant configuration:
1. Add to known devices in home assistant the BT device you want to track:

```
tracked_device:
  mac: BLE_AA:BB:CC:DD:EE:FF
  name: Tracked device
  track: yes
  hide_if_away: no
```

2. Add the tracked device to a Person

The ESPHome configuration:

```
binary_sensor:
  - platform: ble_presence
    mac_address: AA:BB:CC:DD:EE:FF
    name: Tracked device
    id: tracked_device
    on_state:
      then:
        - if:
            condition: 
              binary_sensor.is_on: tracked_device
            then:
              - homeassistant.service:
                  service: device_tracker.see
                  data:
                    dev_id: tracked_device
                    location_name: 'home'
            else:
              - homeassistant.service:
                  service: device_tracker.see
                  data:
                    dev_id: tracked_device
                    location_name: 'not_home'
```

This way the esphome binary sensor will call the HASS service that a device is in/out of range and update the device_tracker entity, which in turn will update the Person object.

OttoWinter, 2019-06-12:
@kirichkov FYI, you don't need to use condition here - just use `on_press` and `on_release`

closed 2020-07-04 by ReneHonig

HepoH3, 2020-09-24:
Indeed.

> 
> 
> This should be revisited i think, because a _lot_ has changed in Home Assistant on the device_tracker front.

axwell, 2020-10-14:
i am looking for the same thing..known devices is deprecated.

ministryofsillywalks, 2021-03-09:
Would also love to see this updated to support device_tracker.
I track my keys with a BLE sensor and it would be easiest to just add them to a person but it's not possible to add a binary_sensor to a person.
Like someone mentioned. known_devices has been deprecated so currently there is no good way to get this working afaik?