esphome/feature-requests #205: ESP32 Camera access mjpeg outside homeassistant

nhorvath, 2019-04-30:
**Describe the problem you have/What new integration you would like**

I want to be able to feed the mjpeg stream into motioneye, not just the home assistant api. Is there a way to access the MJPEG stream from the esp32 camera?

**Please describe your use case for this integration and alternatives you've tried:**

Currently the camera stream is only available inside of home assistant. I tried to use the camera proxy url and tested that I could view it by using vlc network stream but that only works while I'm viewing the stream in a browser (the token changes after I close it), and it doesn't work at all in motioneye for some reason.

**Additional context**

OttoWinter, 2019-05-01:
Duplicate of https://github.com/esphome/issues/issues/245

closed 2019-05-01 by OttoWinter

DavoudTaghawiNejad, 2019-05-21:
nhorvath, you can generate a permanent token. http://you-home-assistant-ip:8123/profile. Or find profile via the menues

nhorvath, 2019-05-21:
@DavoudTaghawiNejad thanks for the tip, I didn't know about that. But when I replaced the token= parameter from the url to the camera and replaced it with the generated one it says unauthorized. Any ideas?

https://example.com/api/camera_proxy_stream/camera.esp32_cam?token=[generated token]

baaswietse, 2019-05-30:
@DavoudTaghawiNejad like to know the answer aswell