esphome/feature-requests #209: Add support for Xiaomi Mi Scale V1 and V2

tiagofreire-pt, 2019-05-02:
**Describe the problem you have/What new integration you would like**
There is no easy way to integrate this BLE scale into Home Assistant, on Hassio.

**Please describe your use case for this integration and alternatives you've tried:**
Integrating weight and BMI on esphome for Home Assistant.

The python code is available for both V1 and V2 Xiaomi Mi Scales: https://github.com/lolouk44/xiaomi_mi_scale/blob/master/Xiaomi_Scale.py

**Additional context**
It's just a small code increment on xiaomi_ble.cpp parser, already available on esphome.

tiagofreire-pt, 2019-05-02:
More info for parsing data at: https://community.home-assistant.io/t/integrating-xiaomi-mi-scale/9972/81

tiagofreire-pt, 2019-05-08:
`# blescan -v
Scanning for devices...
Running  /usr/local/lib/python3.6/dist-packages/bluepy/bluepy-helper
Sent:  le on

Got: '# bluepy-helper.c version 1.3.0 built at 19:35:35 on Mar 12 2019\n'
Got: 'rsp=$mgmt\x1ecode=$success\n'
Sent:  scan

Got: 'rsp=$mgmt\x1ecode=$success\n'
Got: "rsp=$stat\x1estate=$scan\x1edst='(null)\x1emtu=h0\x1esec='low\n"
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4F\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
    Device (new): c4:7c:8d:6a:49:16 (public), -79 dBm 
	Flags: <06>
	Incomplete 16b Services: <0000fe95-0000-1000-8000-00805f9b34fb>
	16b Service Data: <95fe712098007d16496a8d7cc40d0710030a0300>
	Complete Local Name: 'Flower care'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h3E\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF884006094D4942435309FF5701CCE75C1BAC6B\n'
    Device (new): cc:e7:5c:1b:ac:6b (random), -62 dBm 
	Flags: <06>
	Incomplete 16b Services: <0000181b-0000-1000-8000-00805f9b34fb>
	16b Service Data: <1b1802a4e3070508132e2efdff8840>
	Complete Local Name: 'MIBCS'
	Manufacturer: <5701cce75c1bac6b>
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4C\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h39\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF884006094D4942435309FF5701CCE75C1BAC6B\n'
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4C\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
Got: 'rsp=$scan\x1eaddr=bC47C8D6A45B5\x1etype=h1\x1erssi=h58\x1eflag=h0\x1ed=b020106030295FE141695FE7120980078B5456A8D7CC40D0910026F000C09466C6F7765722063617265\n'
    Device (new): c4:7c:8d:6a:45:b5 (public), -88 dBm 
	Flags: <06>
	Incomplete 16b Services: <0000fe95-0000-1000-8000-00805f9b34fb>
	16b Service Data: <95fe7120980078b5456a8d7cc40d0910026f00>
	Complete Local Name: 'Flower care'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h3C\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF8840\n'
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4F\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h3B\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF884006094D4942435309FF5701CCE75C1BAC6B\n'
Select timeout
Sent:  scanend

Got: 'rsp=$mgmt\x1ecode=$success\n'
Stopping  /usr/local/lib/python3.6/dist-packages/bluepy/bluepy-helper`

tiagofreire-pt, 2019-05-08:
> Some `VERY_VERBOSE` logs of the mi scale BLE payload would be useful. Could you provide those?

`# blescan -v
Scanning for devices...
Running  /usr/local/lib/python3.6/dist-packages/bluepy/bluepy-helper
Sent:  le on

Got: '# bluepy-helper.c version 1.3.0 built at 19:35:35 on Mar 12 2019\n'
Got: 'rsp=$mgmt\x1ecode=$success\n'
Sent:  scan

Got: 'rsp=$mgmt\x1ecode=$success\n'
Got: "rsp=$stat\x1estate=$scan\x1edst='(null)\x1emtu=h0\x1esec='low\n"
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4F\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
    Device (new): c4:7c:8d:6a:49:16 (public), -79 dBm 
	Flags: <06>
	Incomplete 16b Services: <0000fe95-0000-1000-8000-00805f9b34fb>
	16b Service Data: <95fe712098007d16496a8d7cc40d0710030a0300>
	Complete Local Name: 'Flower care'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h3E\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF884006094D4942435309FF5701CCE75C1BAC6B\n'
    Device (new): cc:e7:5c:1b:ac:6b (random), -62 dBm 
	Flags: <06>
	Incomplete 16b Services: <0000181b-0000-1000-8000-00805f9b34fb>
	16b Service Data: <1b1802a4e3070508132e2efdff8840>
	Complete Local Name: 'MIBCS'
	Manufacturer: <5701cce75c1bac6b>
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4C\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h39\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF884006094D4942435309FF5701CCE75C1BAC6B\n'
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4C\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
Got: 'rsp=$scan\x1eaddr=bC47C8D6A45B5\x1etype=h1\x1erssi=h58\x1eflag=h0\x1ed=b020106030295FE141695FE7120980078B5456A8D7CC40D0910026F000C09466C6F7765722063617265\n'
    Device (new): c4:7c:8d:6a:45:b5 (public), -88 dBm 
	Flags: <06>
	Incomplete 16b Services: <0000fe95-0000-1000-8000-00805f9b34fb>
	16b Service Data: <95fe7120980078b5456a8d7cc40d0910026f00>
	Complete Local Name: 'Flower care'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h3C\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF8840\n'
Got: 'rsp=$scan\x1eaddr=bC47C8D6A4916\x1etype=h1\x1erssi=h4F\x1eflag=h0\x1ed=b020106030295FE151695FE712098007D16496A8D7CC40D0710030A03000C09466C6F7765722063617265\n'
Got: 'rsp=$scan\x1eaddr=bCCE75C1BAC6B\x1etype=h2\x1erssi=h3B\x1eflag=h0\x1ed=b02010603021B1810161B1802A4E3070508132E2EFDFF884006094D4942435309FF5701CCE75C1BAC6B\n'
Select timeout
Sent:  scanend

Got: 'rsp=$mgmt\x1ecode=$success\n'
Stopping  /usr/local/lib/python3.6/dist-packages/bluepy/bluepy-helper`

Xeroxxx, 2019-06-01:
I love to see this supported too.

zsiti, 2019-06-15:
I don't own a Xiaomi Mi Scale, but I would buy one if it would be possible to read out data from it using ESPHome. Hope this would be implemented eventually.

tiagofreire-pt, 2019-07-11:
@ottowinter, do you need some help on this?

tsunglung, 2019-08-05:
I already implement this feature, How can I upload the code?

diraimondo, 2019-08-05:
> I already implement this feature, 

Do you mean a working ESPHome component?

> How can I upload the code?

You could create a PR for the main branch or provide the code to test in custom-component folder.

tsunglung, 2019-08-05:
[xiaomi_scale.diff.zip](https://github.com/esphome/feature-requests/files/3468444/xiaomi_scale.diff.zip)

I am fresh to github. no idea to create PR. I provide the diff code for test.

glmnet, 2019-08-05:
I see you're trying to create the PR 🎉️
If you find issues you can find good info [here](https://esphome.io/guides/contributing.html#)
Feel free to ask

diraimondo, 2019-08-05:
@tsunglung I can confirm that your patch works and my Mi Scale 2 reports the weight (in kgs).
It looks that the battery level is not available: do you know if it is supported by the device/protocol?

For the PR: it is not too difficult but, I you prefer, I could make the request for you.

tsunglung, 2019-08-06:
@diraimondo Thanks for confirming.
From @tiagofreire-pt  provided info/protocol, I did not see the battery level info. So I did not implement battery level. 
I will try to create PR later if I have time.

tiagofreire-pt, 2019-08-08:
@tsunglung, OK, for this first approach I think is may be good enough.

I'm looking forward into this new PR. If you need testers, please, just say it. :)

tiagofreire-pt, 2019-08-08:
@tsunglung, please, check if it's everything alright. My knowledge on github and C++ are quite limited. :)

Thanks for this awesome work!

tsunglung, 2019-08-08:
[xiaomi_scale.diff.zip](https://github.com/esphome/feature-requests/files/3481496/xiaomi_scale.diff.zip)

@tiagofreire-pt Thanks for creating commit, I checked it and need some modification.

tiagofreire-pt, 2019-08-08:
@tsunglung, can you do it directly or may I need to do it for you?

If so, please, point where are the modifications needed.

Thanks!

glmnet, 2019-08-08:
@tiagofreire-pt you must create only one PR for the whole feature, not a PR for every file. Please fix that. Close PR that will not be required to be merged.

tiagofreire-pt, 2019-08-08:
@glmnet , it seems that I don't have permissions for:

> You’re not authorized to merge this pull request. 

I'm following this FAQ: https://help.github.com/en/articles/merging-a-pull-request

glmnet, 2019-08-08:
it's not your role to do that, you are welcome to create a pull request, however as I said your PR is not acceptable as there is one pull request per file and that is not how it's done.

see for example this PR https://github.com/esphome/esphome/pull/679 it has 5 file changes 
you provided 6 PR with one file change each.

We (actually nobody) will accept this.

For a clean approach I will suggest you close all your PR and submit a new one with all the changes. Thanks.

tiagofreire-pt, 2019-08-10:
@glmnet, I guess I finally managed to properly submit the PR: https://github.com/esphome/esphome/pull/691

cohe4ko, 2020-03-05:
How to use this feature in actual project?

pattyland, 2020-05-06:
@cohe4ko As you can see #691 is still not merged so you can't use it.
There seems to good progress for Xiaomi BLE devices in #552 so maybe this will help this issue

pattyland, 2021-02-22:
Looks like #1368 finally added the support for the xiaomi scales.

Thank you @dckiller51! I don't know if you noticed: https://www.bountysource.com/issues/73647454-add-support-for-xiaomi-mi-scale-v1-and-v2

ppanagiotis, 2021-05-11:
Hello! I am trying to use this feature but no input data appears at esphome logs or at home assistant. My yaml file is:
```
esphome:
  name: ble_gw_test
  platform: ESP32
  board: nodemcu-32s
  arduino_version: dev

wifi:
  ssid: "foo"
  password: "bar"

api:

ota:

## Enable logging
logger:

esp32_ble_tracker:
  
sensor:
  - platform: xiaomi_miscale2
    mac_address: 5C:CA:D3:XX:XX:XX
    weight:
      name: "Xiaomi Mi Scale Weight"
      id: weight_miscale
      on_value:
        then:
          - lambda: |-
              if (id(weight_miscale).state >= 65 && id(weight_miscale).state <= 74.49) {
                return id(weight_user1).publish_state(x);}
              else if (id(weight_miscale).state >= 74.50 && id(weight_miscale).state <= 83) {
                return id(weight_user2).publish_state(x);}
              else if (id(weight_miscale).state >= 46 && id(weight_miscale).state <= 64) {
                return id(weight_user3).publish_state(x);}
              else if (id(weight_miscale).state >= 28 && id(weight_miscale).state <= 45) {
                return id(weight_user4).publish_state(x);}
              else if (id(weight_miscale).state >= 5 && id(weight_miscale).state <= 20) {
                return id(weight_user5).publish_state(x);}

    impedance:
      name: "Xiaomi Mi Scale Impedance"
      id: impedance_xiaomi
      on_value:
        then:
          - lambda: |-
              if (id(weight_miscale).state >= 65 && id(weight_miscale).state <= 74.49) {
                return id(impedance_user1).publish_state(x);}
              else if (id(weight_miscale).state >= 74.50 && id(weight_miscale).state <= 83) {
                return id(impedance_user2).publish_state(x);}
              else if (id(weight_miscale).state >= 46 && id(weight_miscale).state <= 64) {
                return id(impedance_user3).publish_state(x);}
              else if (id(weight_miscale).state >= 28 && id(weight_miscale).state <= 45) {
                return id(impedance_user4).publish_state(x);}
              else if (id(weight_miscale).state >= 5 && id(weight_miscale).state <= 20) {
                return id(impedance_user5).publish_state(x);}

  - platform: template
    name: Weight User1
    id: weight_user1
    unit_of_measurement: 'kg'
    icon: mdi:weight-kilogram
    accuracy_decimals: 2
  - platform: template
    name: Impedance Panagiotis
    id: impedance_user1
    unit_of_measurement: 'ohm'
    icon: mdi:omega
    accuracy_decimals: 0
  - platform: template
    name: Weight Siham
    id: weight_user2
    unit_of_measurement: 'kg'
    icon: mdi:weight-kilogram
    accuracy_decimals: 2
  - platform: template
    name: Impedance Siham
    id: impedance_user2
    unit_of_measurement: 'ohm'
    icon: mdi:omega
    accuracy_decimals: 0
  - platform: template
    name: Weight Théo
    id: weight_user3
    unit_of_measurement: 'kg'
    icon: mdi:weight-kilogram
    accuracy_decimals: 2
  - platform: template
    name: Impedance Théo
    id: impedance_user3
    unit_of_measurement: 'ohm'
    icon: mdi:omega
    accuracy_decimals: 0
  - platform: template
    name: Weight Sacha
    id: weight_user4
    unit_of_measurement: 'kg'
    icon: mdi:weight-kilogram
    accuracy_decimals: 2
  - platform: template
    name: Impedance Sacha
    id: impedance_user4
    unit_of_measurement: 'ohm'
    icon: mdi:omega
    accuracy_decimals: 0
  - platform: template
    name: Weight Noham
    id: weight_user5
    unit_of_measurement: 'kg'
    icon: mdi:weight-kilogram
    accuracy_decimals: 2
  - platform: template
    name: Impedance Noham
    id: impedance_user5
    unit_of_measurement: 'ohm'
    icon: mdi:omega
    accuracy_decimals: 0
``` 
Can someone help me?

PricelessToolkit, 2021-06-30:
ppanagiotis :   same things everything worked for me, but after the update it does not work any more

edenhaus, 2021-09-08:
@jesserockz As this feature is already implemented (esphome/esphome#1368), I think we can close this issue

closed 2021-09-08 by jesserockz