esphome/feature-requests #211: Wiegand interface

remygiraud, 2019-05-03:
**Describe the problem you have/What new integration you would like**
I am using wiegand access. It would be nice to be able to integrate this kind of equipment to ESPHome

**Please describe your use case for this integration and alternatives you've tried:**
At the moment, I am using a home made script. Having one solution for all would be nice.

**Additional context**

Zebble, 2019-05-23:
The RDM6300 seems to use a standard UART protocol.  Wiegand is its own beast.  It would require some interrupt-driven communication for accuracy, but does seem fairly simple.

There's a good library that can figure out the number of bits in the data automagically.  If I had time, I'd port it.  Any takers?

https://github.com/paulo-raca/YetAnotherArduinoWiegandLibrary

nerdosity, 2019-06-21:
That would be very good. I have a weigand reader outside, knowing the tag read would be perfect for integrating it into Home Assistant.
Not knowing the tag I have mapped the relay as an ON/OFF, and I will have to put another rfid reader integrated for automations since I want some automation just for my tag and my girlfriend's, not for other tags.

beikeland, 2019-07-15:
I have a couple of Siemens AR6182-MX readers (w/keypads) that can be configure to use a Wiegand interface, would be interesting to get this hooked up to EspHome.

beikeland, 2019-07-17:
This fixes the library used in esp-rfid as linked to above
https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino/pull/38

Unfortunately I need other bitcounts than standard depending on card type and optional pin length so I'll probably do a AR6182-MX library instead of a generic Wiegand.

nerdosity, 2019-07-18:
@beikeland I am using esp-rfid project on my nodeMCU since a couple of weeks, just use version 1.0.2 since latest one has problems with MQTT messages for tags.

beikeland, 2019-07-18:
My primary motivation for looking into using ESPhome was to not rely on MQTT. I've got the library working and sorted out separating Pin, Card and Card+Pin in an Arduino sketch running on a NodeMCU-devkit. So just have to wrap my head around the abstraction needed to write a new component for the Wiegand bus and a sensor or something on top of that, or find that someone beats me to it while trying to understand the ESPhome code:)

nerdosity, 2019-07-18:
I understand that, I just replied to the fact that it wasn't booting on your nodeMCU ;)

beikeland, 2019-07-18:
That was due to the missing ICACHE attribute for the interrupt handler in
the underlying library. Fixed it.

beikeland, 2019-07-20:
Still trying to work out how to add the components to ESPHome, maybe one day.
Sharing a gist with the working arduino sketch for esp8266 and AR6182 reader. Reader uses an electric wiegand interface, but custom protocol.
https://gist.github.com/beikeland/145b20b1fc585049ea8951eddfc5732e

CountParadox, 2019-09-14:
I have a bunch of HID Prox readers, would love a component to use them with ESPHome! Currently using them with https://github.com/TomHarkness/homeID

pricard32, 2020-01-10:
Any news on that? I've bought a keypad that use this protocol and would like to wiring it to an ESP32 running ESPHOME

edwardscaleb9, 2020-01-11:
Also interested in using wiegand keypads connected to an ESP8266/ESP32.

pricard32, 2020-01-11:
Can you help me building the plugin ?

edwardscaleb9, 2020-01-11:
I’d be happy to give it a go, but I’m not much of a programmer these days. More of a scripter I’d say.

While I’m thinking about it, one feature I’d like to see is the ability to set codes/card numbers in HomeAssistant (or whatever frontend) and have them cached locally on the ESP.

CountParadox, 2020-01-12:
You must need the card number to be passed to home assistant, then you can
do what you want with it.

Zebble, 2020-01-12:
If you want local cache / standalone control, you might want to look at the esprfid project.  It can also talk to Mqtt so can also be manually integrated with stuff like Hass.

https://github.com/esprfid/esp-rfid

You don't need their special board, any esp8266 will do.  I used it to handle card reading/users and had Hass control the lock.  Worked quite well.

Still would be good to have weigand in esphome, to keep everything on one platform.

Sent from Nine<http://www.9folders.com/>

pricard32, 2020-01-12:
I'm not interest of just RFID. I would like the complete wiegand protocol. Either Keypad of RDIF will be pass to HA.

And like @edwardscaleb9 said, it would be great to program the keypas via HA.

I've found online a library for Weigand, but dont have the skills to implement it with ESPHOME.

Someone with those skill would want to help us ?

gdoerr, 2020-01-27:
I was able to implement a CustomAPIDevice using code from [this Github repo](https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino) without any issues. The code works as is (I had to drop the `IRAM_ATTR` on the interrupt handlers in order to get it to link but it's working well for me.

I can post the sample custom device code if anyone is interested.

pricard32, 2020-01-27:
@gdoerr i’m really interested to see how you did it !!

gdoerr, 2020-01-27:
Save the code below as wiegand_device.h and include it as a custom component in your YAML file.
```
custom_component:
  - lambda: |-
       auto wiegand = new WiegandReader(pin0, pin1, "script.to_call");
       return {wiegand};
```

`pin0` should be the GPIO connected to the D0 signal and `pin1` is the GPIO connected to the D1 signal. Replace `"script.to_call"` with the name of the script or service to call. The code entered will be passed as the parameter `code`.

```
#include "esphome.h"

/**
 * Wiegand Reader Custom Device
 *
 * Copied from https://github.com/monkeyboard/Wiegand-Protocol-Library-for-Arduino
 * Implemented by Greg Doerr (https://github.com/gdoerr)
 *
 * In my example, hooked to an Olimex ESP32-POE device connected to a Retekess H1EM-W
 * Wiegand keypad. This device calls a service on Home Assistant when a code is available
 *
 * Samples key presses every 200ms and stores the key in a string until:
 *   [1] - the user presses a '#' which sends the code immediately
 *   [2] - the user presses nothing for 2,000ms (inter-digit timer) which then sends the code
 *   [3] - the user presses a '*' which sends the '*' immediately
 */
class WiegandReader : public PollingComponent, public CustomAPIDevice {

public:
    WiegandReader(int pinD0, int pinD1, std::string serviceName)
        : PollingComponent(200),
        pinD0(pinD0), pinD1(pinD1),
        serviceName(serviceName) {
    }

    /**
     * Initial setup
     */
    void setup() override {
        _lastWiegand = 0;
        _cardTempHigh = 0;
        _cardTemp = 0;
        _code = 0;
        _wiegandType = 0;
        _bitCount = 0;

        // Configure the input pins
        pinMode(pinD0, INPUT);
        pinMode(pinD1, INPUT);

        // Attach the interrupts
        attachInterrupt(digitalPinToInterrupt(pinD0), ReadD0, FALLING);  // Hardware interrupt - high to low pulse
        attachInterrupt(digitalPinToInterrupt(pinD1), ReadD1, FALLING);  // Hardware interrupt - high to low pulse
    }

    void update() override {
        // See if we have a valid code
        noInterrupts();
        bool rc = DoWiegandConversion();
        interrupts();

        if(rc) {
            if(_code < 10) {
                // We have a digit, make it ASCII for convenience
                keyCodes += (_code + 0x30);
            } else if(_code == 11) {
                // The user pressed '#', send the accumulated code and reset for the next string
                callHAService(keyCodes);
                keyCodes = "";
[... 150 lines cut ...]
                    _cardTemp=0;
                    _cardTempHigh=0;
                    _code=cardID;
                    return true;
                }
            } else {
                // well time over 25 ms and bitCount !=8 , !=26, !=34 , must be noise or nothing then.
                _lastWiegand=sysTick;
                _bitCount=0;
                _cardTemp=0;
                _cardTempHigh=0;
                return false;
            }
        } else
            return false;
    }
};

volatile unsigned long WiegandReader::_cardTempHigh = 0;
volatile unsigned long WiegandReader::_cardTemp = 0;
volatile unsigned long WiegandReader::_lastWiegand = 0;
volatile int WiegandReader::_bitCount = 0;
unsigned long WiegandReader::_code = 0;
int WiegandReader::_wiegandType = 0;
```

pricard32, 2020-01-27:
Wow @gdoerr , that's a greate job you did! Thanks !
Since i'm pretty new to this, and i've buy the exact same keypad as you, can you help me with something more.
I'll like to push the data into a text_sensor: can you explain me what I have to that to the yaml and .h file?

Thanks again!

pricard32, 2020-01-27:
And i'm using a ESP-32 for now, can I use any GPIO pin ?

gdoerr, 2020-01-27:
You'd have to restructure the Wiegand class into a TextSensor. I don't recommend this because you're not interested in state, you really want the event which is why I call a service. You should be able to find the example code on esphome to implement a custom text sensor.

You should be able to use any GPIO pins on the ESP32 but the order is important...swap D0 and D1 if your're getting weird results.

Good Luck!

pricard32, 2020-01-27:
What would you recommend to write in the script to call to received as a event ?

gdoerr, 2020-01-27:
You can use either a [yaml script](https://www.home-assistant.io/integrations/script/) or a [python script](https://www.home-assistant.io/integrations/python_script/).

I use a python script to validate the code sequences against a dedicated outlook.com calendar to control access to my garage.

pricard32, 2020-01-27:
Or maybe only share your how implentation, that's help me reverse engineering and understand or to implement it for my use.

pricard32, 2020-01-28:
I've tried to change the class declaration from CustomApiDevice to TextSensor and add PublishState(keyCode) to CallHaService without success

gdoerr, 2020-01-28:
@pricard32 Rather than mucking up this feature request thread with a specific problem, perhaps you can post your question on Stack Overflow which is better suited and then post a link to the problem here.

I'm happy to help but you'll need to provide more information than you tried something without success. When you ask the question, be sure and post exactly what you tried and what errors you're getting. Otherwise it's difficult to help you without writing the software for you...

Zebble, 2020-01-28:
@gdoerr great work!  Solves the immediate problem and really helped me understand how custom components work in esphome.  Well done, and thank you.

pricard32, 2020-01-29:
@gdoerr Right, I understand. I'm pretty new here.
In fact, the way you use the keypad is exactly how I need it, except I have to found a way to received the keycode in a sensor, or text input, or any kind of event that i'll use late in an automation.
This is why I was a little excited when I saw that you've sucessfully implement it.

I'm a Lua programmer and when I try to modify C++ code, it always end with synthax error. I need practice.

So my goal is this : received the keycode to use it in automation without having to write another script in a language that I don't know.

Luisiado, 2020-02-03:
hello, first of all thank for your code I test it and works fine, just I would add to the description:
```
esphome:
  includes:
    - wiegand_device.h
```

I am trying to catch the code in a python script.
my ESPhome yaml is:

`custom_component:
  - lambda: |-
       auto wiegand = new WiegandReader(21, 23, "python_scripts.door_mqtt_publish.py");
       return {wiegand};`

in **config/python_scripts/door_publish_mqtt.py** I have a simple code just to test it 
```
logger.info("PYTHON SCRIPT CALLED")
name = data.get('code')
logger.info("HELLOOOOOOOOO %s", name)
hass.bus.fire(name, {"test": "from a Python script!"}
)```

and I think that services is not mandatory
**config/python_scripts/services.yaml**
```
door_mqtt_publish:
  description: publish a rfid code
  field:
    code:
      description: code format
      example: 9451293
```

In the ESP home logger I can see that the code is read perfectly but I can not trigger the python script to send the payload with the code on it. After read a code nothing appears in the HA logger.

HA logger:

> 2020-02-03 12:25:05 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you do experience issues with Home Assistant.
> 2020-02-03 12:25:05 WARNING (MainThread) [homeassistant.loader] You are using a custom integration for customizer which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you do experience issues with Home Assistant.
> 2020-02-03 12:25:36 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.1908784496] Error handling message: Unauthorized

gdoerr, 2020-02-03:
You cannot call a python script directly through Home Assistant [see here](https://www.home-assistant.io/integrations/python_script/)

You have to specifically allow each script (likely for security reasons) and call them through a service as was implemented in the example I posted.

pricard32, 2020-03-01:
When I connect the Wiegand device to the GPIO of the ESP32, the board failed to boot every time.
Someone have a clue why? (for now it connetected to 21/23, but i've try with other pins too)

CountParadox, 2020-03-01:
Check the power connections on the device

pricard32, 2020-03-01:
The power is correct, since when I disconnect the PIN between the ESP and the keypad, the board boot as usual

Luisiado, 2020-03-02:
do you have grounds (RFID, ESP32, power supply) short circuited? all
grounds should be connected between themselves.

pricard32, 2020-03-02:
@Luisiado i'm ashamed... ! Forgot the ground between the ESP and the keypad and that cause the ESP to fail to boot... ! Thanks !

pricard32, 2020-03-02:
@Luisiado what did you do to solve your issue? how did you publish the code via MQTT to HA ?

Gio-dot, 2020-03-02:
> @Luisiado what did you do to solve your issue? how did you publish the code via MQTT to HA ?

Hi, what kind of ESP do you have? It is 3.3V right? 
What is your wiegand device? It is 5.0V on Do and D1 lines right?
Did you use any level shifter to connect wiegand device to ESP device?

Luisiado, 2020-03-02:
There are some articles which say that ESP is 5V tolerant. The proper way
to connect the rfid reader is through voltage divider. I am not using the
voltage divider and my system is working fine but I can't recommend you to
do the same ;)

Gio-dot, 2020-03-02:
Ok, thank you. I'm titally new on esphome and i don't know pyton. I know quite well yaml. Problem is that i have no idea how script to catch the code should be. Did you manage to have it working?

nerdosity, 2020-03-02:
> There are some articles which say that ESP is 5V tolerant. The proper way to connect the rfid reader is through voltage divider. I am not using the voltage divider and my system is working fine but I can't recommend you to do the same ;)
> […](#)

I can testify that it's 5v tolerant, even esp8266 was. It's semi-official, since an engineer form expressif commented this on Facebook saying that it was tolerant.

Luisiado, 2020-03-02:
I opened a issue here but I didn´t get answer. Maybe now we are lucky
because I published the code.
https://community.home-assistant.io/t/how-to-read-a-variable-from-a-custom-component-wiegand/168588

Gio-dot, 2020-03-31:
> @Luisiado i'm ashamed... ! Forgot the ground between the ESP and the keypad and that cause the ESP to fail to boot... ! Thanks !

Hi, i can't understand: gdoerr code works only with wiegand keyboards or even with wiegand rfid tag readers? I'm trying it wth an rfid tag reader but i can't see nothing in esp log when i read a tag. My reader works in an arduino system...

pricard32, 2020-03-31:
@Gio-dot The rfid tag reader doesnt seems to work for me too.

Gio-dot, 2020-03-31:
> @Gio-dot The rfid tag reader doesnt seems to work for me too.
This is my setup. I use an ESP32 (D0 pin 23 D1 pin 21). When i read a tag nothing happens in the log...
[image]

Gio-dot, 2020-04-20:
> Save the code below as wiegand_device.h and include it as a custom component in your YAML file.
>
> ```
> custom_component:
>   - lambda: |-
>        auto wiegand = new WiegandReader(pin0, pin1, [...]

I gdoerr, can you help to make it working? I used your code but nothing happen when i read a tag. The same reader work well when connected to an arduino with the monkeyboard library. Thank you.

gdoerr, 2020-04-20:
@Gio-dot Since your keypad works with your Arduino I assume it's either the hardware interface (you don't mention the board you're using) or with the Home Assistant service call.

I suggest adding some log statements in the update loop and see if you're getting anything in that loop. That will narrow down if it's a keypad interface or Home Assistant interface issue.

Good luck

Gio-dot, 2020-04-20:
> @Gio-dot Since your keypad works with your Arduino I assume it's either the hardware interface (you don't mention the board you're using) or with the Home Assistant service call.
>
> I suggest adding [...]

I'm using an esp32 with a wiegand id tag reader like the picture 2 post above. Which esp are you using? Can you suggest how ha script should be? I never used a script that manage a parameter until now.. Thank you.

pashdown, 2020-10-03:
With the use of homeassistant.tag, it would be nice to get this incorporated into the main esphome code.

pashdown, 2020-10-03:
I have an immediate need for Wiegand support and I'd be happy to pay/contract someone to integrate it.  Using the example above throws my Wemos d1 Mini into safe mode if I do this:
```
custom_component:
  - lambda: |-
      auto wiegand = new WiegandReader(D6, D7, "homeassistant.tag_scanned");
      return {wiegand};
```

The "[Yet Another Arduino Wiegand Library](https://github.com/paulo-raca/YetAnotherArduinoWiegandLibrary)" is nice because it not only reports different bit lengths, it knows when the reader is connected and disconnected, upon which another sensor could be built.  However, my attempts at integrating their example have failed spectacularly. Here is where I'm currently at, which doesn't compile.

And yes, I've tried esp-rfid, which is fine, but I'd rather be able to write automations and control the RFID reader inside esphome.
```
#include "esphome.h"
#include <Wiegand.h>

// These are the pins connected to the Wiegand D0 and D1 signals.
#define PIN_D0 12
#define PIN_D1 13

class WiegandReader : public PollingComponent, public TextSensor {
public:
  // The object that handles the wiegand protocol
  Wiegand wiegand;

  WiegandReader() : PollingComponent(100), TextSensor() { }

  // Initialize Wiegand reader
  void setup() override {
    Serial.begin(9600);
  
    //Install listeners and initialize Wiegand reader
    wiegand.onReceive(receivedData, "Card read: ");
    wiegand.onReceiveError(receivedDataError, "Card read error: ");
    wiegand.onStateChange(stateChanged, "State changed: ");
    wiegand.begin(Wiegand::LENGTH_ANY, true);

    //initialize pins as INPUT
    pinMode(PIN_D0, INPUT);
    pinMode(PIN_D1, INPUT);
  }

  // Continuously checks for pending messages and polls updates from the wiegand inputs
  void loop() override {
    // Checks for pending messages
    wiegand.flush();

    // Check for changes on the the wiegand input pins
    wiegand.setPin0State(digitalRead(PIN_D0));
    wiegand.setPin1State(digitalRead(PIN_D1));
  }

  // Notifies when a reader has been connected or disconnected.
  // Instead of a message, the seconds parameter can be anything you want -- Whatever you specify on `wiegand.onStateChange()`
  void stateChanged(bool plugged, const char* message) {
    Serial.print(message);
    Serial.println(plugged ? "CONNECTED" : "DISCONNECTED");
    ESP_LOGD("custom", "%s %s", message, plugged ? "CONNECTED" : "DISCONNECTED");
  }

  // Notifies when a card was read.
  // Instead of a message, the seconds parameter can be anything you want -- Whatever you specify on `wiegand.onReceive()`
  void receivedData(uint8_t* data, uint8_t bits, const char* message) {
    Serial.print(message);
    Serial.print(bits);
    Serial.print("bits / ");

    //Print value in HEX
    uint8_t bytes = (bits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(data[i] >> 4, 16);
        Serial.print(data[i] & 0xF, 16);
    }
    ESP_LOGD("custom", "%s %d bits / %X", message, bits, data);
  }

  // Notifies when an invalid transmission is detected
  void receivedDataError(Wiegand::DataError error, uint8_t* rawData, uint8_t rawBits, const char* message) {
    Serial.print(message);
    Serial.print(Wiegand::DataErrorStr(error));
    Serial.print(" - Raw data: ");
    Serial.print(rawBits);
    Serial.print("bits / ");

    //Print value in HEX
    uint8_t bytes = (rawBits+7)/8;
    for (int i=0; i<bytes; i++) {
        Serial.print(rawData[i] >> 4, 16);
        Serial.print(rawData[i] & 0xF, 16);
    }
    ESP_LOGD("custom", "%s %s - Raw data: %d bits / %X", message, Wiegand::DataErrorStr(error), rawBits, rawData);
  }
};
```

pashdown, 2020-10-04:
As mentioned previously ICACHE attribute is needed to keep it from crashing on esp8266:

```
    static void ICACHE_RAM_ATTR ReadD0() {

    static void ICACHE_RAM_ATTR ReadD1() {
```
I'm still willing to bounty someone to integrate YetAnotherArduinoWiegandLibrary into esphome.

Gio-dot, 2020-10-04:
> As mentioned previously ICACHE attribute is needed to keep it from crashing on esp8266:
> 
> ```
>     static void ICACHE_RAM_ATTR ReadD0() {
> 
>     static void ICACHE_RAM_ATTR ReadD1() {
> ```
> 
> I'm still willing to bounty someone to integrate YetAnotherArduinoWiegandLibrary into esphome.

I'm hoping too that someone can help with WIegand. These readers are much better than RC522 or PN532..

reidprojects, 2020-10-07:
Hey everyone, it's not ready to share yet, but just to let you know that I just got a success integrating YetAnotherArduinoWiegandLibrary in a custom ESPHome integration. Now, each tags et keypresses are sent through the Home Assistant Tag API. I'll continue to work on this in the upcoming days and let you know once I have something more complete !

[image]

CountParadox, 2020-10-07:
That's amazing, thank you and keep up the hard work!

Gio-dot, 2020-10-07:
> Hey everyone, it's not ready to share yet, but just to let you know that I just got a success integrating YetAnotherArduinoWiegandLibrary in a custom ESPHome integration. Now, each tags et [...]

Great! I have some wiegand reader an various esp32- esp8266 devices. If you need some test i will be happy to try your component. Thank you.

Gio-dot, 2020-10-07:
Great, i have some wiegand reader and various esp32/8266 devices. If you need someone that can make some test i will be happy to help. Thank you.

nrodgers1987, 2020-11-03:
@reidprojects Is there anything I can test or try out? I'm building a temperature station and need to ability to scan cards using a HID reader, this is the missing link to my project. Any help would be much appreciated.

miccico, 2020-12-30:
@reidprojects - did you have any success with your approach? as i am currently looking for that same functionality it would be great if you could share an intermediate state of your code. Thank you very much!

Gio-dot, 2020-12-30:
> @reidprojects - did you have any success with your approach? as i am currently looking for that same functionality it would be great if you could share an intermediate state of your code. Thank you very much!

Hi, since nobody create a component to integrate Wiegand in Esphome, i realized this (working) solution: https://github.com/Gio-dot/Esphome-Wiegand-reader

[85 more replies]

closed 2023-06-03 by nagyrobi