esphome/feature-requests #213: Wifi serial bridge

adrianmihalko, 2019-05-04:
**Describe the problem you have/What new integration you would like**

It would be great to see a serial to Wifi bridge, like esp-link:

https://github.com/jeelabs/esp-link

Why it would be this a nice component?

1. You can wirelessly access your serial devices / output
2. You can make a wireless Rflink 433mhz reader/transmitter if you connect to arduino serial output, then use it in Home Assistant in TCP mode: https://www.home-assistant.io/components/rflink/

Esp-link has more functionality, but for a start a simple read serial on XY pins and forward to TCP port would be more than enough.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

KiLLeRRaT, 2019-05-06:
I'm a bit new to all this, but I think this would be a pretty good feature to have.

I guess if there is a bin output from ESPHome then you can use that bin file and flash it manually using ESPLink the first time.  Once that's done, you can just OTA using ESPHome from there onwards?

OttoWinter, 2019-05-07:
ESPHome is not a collection of projects, it's a collection of components that communicate through defined interfaces (like `sensor`, `binary_sensor`). A component is required to not manage both the front-end (like web server, TCP) and backend (like UART). A wifi-serial would break this abstraction. One potential way would be to have a *stream* abstraction so it would be serial -> stream -> wifi. But I don't really see enough of a use-case to get this into the core.

Sure it would be possible in custom code (in fact the custom UART interface is already documented and on the website).

closed 2019-05-07 by OttoWinter