esphome/feature-requests #214: High accuracy temp sensor mcp9808

JiriHolubcik, 2019-05-05:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

chard62, 2019-09-10:
Higher accuracy temperature sensor, works with I2C. All the other common sensors have been supported.

tarbax, 2019-12-28:
does this work ?

cbvicious, 2020-10-21:
What exactly are we waiting with for a merge? I have a thermostat set up that I'd love to move over to ESPHome, but it has an MCP9808

chard62, 2020-10-21:
It is a nice module, but I did not have the energy to proceed. I was able to get some very nice pre-built bluetooth battery operated units from Xiaomi that do the trick for me. Already ESPHome integrated too. About $7 USD each.Richard

closed 2020-10-26 by jesserockz