esphome/feature-requests #215: Add support for Heltec wifi lora32 "V2"

TEfjLONjDON, 2019-05-06:
**Describe the problem you have/What new integration you would like**

I own a (hiletgo) Heltec wifi lora32 V2 purchased from here "HiLetgo ESP32 LoRa SX1278 0.96 inch OLED Display Development Board WIFI Bluetooth Dual Core 240MHz CP2102 and 433/470MHz Antenna for Arduino Smart WIFI LoRa 32" https://www.amazon.com/dp/B07FYWFH4C/ref=cm_sw_r_cp_apa_i_8y7ZCbA6NGAQY
**Please describe your use case for this integration and alternatives you've tried:**

The list of boards in the Esphome dashboard has "Heltec wifi lora32" but not "Heltec wifi lora32 V2"
**Additional context**

My goal is to use this device as a RF hub. ( Remote_Receiver / Remote_Transmitter ) currently there are issues with the gpio pins recognized incorrectly. The device doesn't show any RF codes in logs when set to remote_receiver dump: all

OttoWinter, 2019-05-07:
That list is just a copy from the platformio-espressif32 package. If that particular board is different from v1, then it will have to be added to that package.

Anyway, the board setting essentially only affects pin aliases. So if you use the `GPIOx` pin names all should be ok.

> The device doesn't show any RF codes in logs when set to remote_receiver dump: all

That would be separate problem, please see issues tagged with remote_receiver first.

Closing as already supported plus this is a feature request for an upstream package.

closed 2019-05-07 by OttoWinter

mseefeldt, 2019-06-05:
The problem here is a bit different:
The board is already listed on platform.io [Heltec WiFi LoRa 32 (V2)](http://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_lora_32_V2.html#board-espressif32-heltec-wifi-lora-32-v2)
but it is not possible to select it (at least in version 1.12.2. ESPHome complains about an unknown board since it is not listed as a choice.

One difference of the boards is the available memory (what i can see from a first glance).

So will the board be included in a new version of ESPHome automatically?