esphome/feature-requests #216: Add special support for the PICO-D4-MINI's up to 40Mhz LED PWM

Joshfindit, 2019-05-07:
**Describe the problem you have/What new integration you would like**

A "platform" that supports the ultra-high PWM of the PICO-D4-MIni, something like:
`platform: ledc_d4`

**Please describe your use case for this integration and alternatives you've tried:**

Use-case: High-PWM LED control with ESPHome
Alternatives: I've only directly worked with the TTGO T7 with the D4, and have tested controlling the LEDs from >1Mhz to <20kHz using ledc.

This is a feature request, and as-such the only "problem" is that Espressif have not properly documented the PWM limits of the ESP32 or the PICO-D4-MINI.
As Espressif's LEDC library controls them both there is some chance that I'm wrong and that the ESP32 can go as high as the D4, but I suspect there's a much smaller limit to the ESP32.

**Additional context**

The ESP32 itself does have PWM output, but it seems to be limited compared to the PICO-D4-MINI (citation needed)
Very likely because of the upgrade's integrated 40Mhz crystal oscillator.

*Do note: "40Mhz LED PWM" is a little click-baity as it does PWM the LED at 40Mhz, but all messages are "ON". it does 1 bit at 20Mhz and so-on. My personal sweet spot is 10-bit (0-1023) which can run as high as 300,000kHz, but I run it at 25kHz as I have found no flicker at 25kHz even at a brightness of `1`*

OttoWinter, 2019-05-07:
> Espressif have not properly documented the PWM limits of the ESP32 or the PICO-D4-MINI.

I think they did: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/ledc.html#ledc-api-supported-range-frequency-duty-resolution

> As Espressif's LEDC library controls them both there is some chance that I'm wrong and that the ESP32 can go as high as the D4, but I suspect there's a much smaller limit to the ESP32.

Actually the high-speed base-clock of the ledc peripheral also runs at 40MHz (and with a configurable bit depth). So from what I see that will have the same range as the D4-mini module.

ESPHome even has a validation for this. Try setting a bit_depth/frequency in the ledc config and it will print an error if the requested range cannot be achieved.

OttoWinter, 2019-05-07:
Oh now I understand, the PICO-D4-Mini *is* an ESP32. No then this feature request is not valid - the LEDC is the same as with all ESP32 modules. The LEDC perihperal is even exactly the same as with the other ESP32 modules (same silicon).

closed 2019-05-07 by OttoWinter

Joshfindit, 2019-05-08:
Nope, you're right again.
I pulled apart my lamp and tried different values to see what failed, and your math holds up. I was apparently mis-remembering tests done at *8-bit* and 300kHz.

Thanks again
Onwards and upwards to ESPHome!