esphome/feature-requests #217: Support for VEML6070 i2c UV light sensor

rradar, 2019-05-08:
**Describe the problem you have/What new integration you would like**

VEML6070 Support

**Please describe your use case for this integration and alternatives you've tried:**

Measure UVA & UVB

**Additional context**
 
- https://github.com/adafruit/Adafruit_VEML6070
- https://www.adafruit.com/products/2899

nirkons, 2019-06-03:
I second this request :)

thx

gaiar, 2019-07-16:
Yes, it is cheap and popular.

pearfire575, 2019-08-03:
Yes please. Cheap and popular to create meteo stations!

Nornode, 2019-09-18:
@rradar - Did you figure out any workaround using a Custom I2C component?
[Custom I2C](https://esphome.io/custom/i2c.html) ?

tilman, 2019-10-01:
As a workaround you can use the Custom Sensor component. It works well with other i2c components on the same bus. I only tested the code with an ESP8266.

File: veml6070_custom_sensor.h
````c++
#include "esphome.h"
#include "Adafruit_VEML6070.h"

class VEML6070CustomSensor : public PollingComponent, public Sensor {
 public:
  Adafruit_VEML6070 uv = Adafruit_VEML6070();
  VEML6070CustomSensor() : PollingComponent(15000) {}
  void setup() override {
    Wire.begin();
    uv.begin(VEML6070_1_T);
  }
  void update() override {
    uint16_t cur_uv = uv.readUV();
    ESP_LOGD("custom", "The value of sensor is: %i", cur_uv);
    publish_state(cur_uv);
  }
};
````

Espconfig:
````yaml
esphome:
  name: wetterstation
  platform: ESP8266
  board: esp01_1m
  includes:
    - veml6070_custom_sensor.h
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6070"

...

i2c:
  sda: GPIO5
  scl: GPIO4

...

sensor:
  - platform: custom
    lambda: |-
      auto veml6070 = new VEML6070CustomSensor();
      App.register_component(veml6070);
      return {veml6070};
    sensors:
      name: "VEML6070 Custom Sensor"

````

Nornode, 2019-10-08:
Awesome, thanks! 
I know this isn't a support forum so I'll try to keep to the topic. 
I don't think this is the preferred way of integrating new sensors should be integrated into ESPhome? - The ease of customizing ESP's with YAML is why ESPhome is gaining so much traction from any other ESP/Arduino project. Fail rate of similar implementations are just very much higher than "standard" YAML.

I tried the example above and failed as my sensor is reading the value "65535" every single reading.

ways, 2019-10-09:
I have a 7700 (lux sensor). Tried this code and get 65535.

```
[18:08:42][D][wifi:272]: Starting scan...
[18:08:43][D][custom:015]: The value of sensor is: 65535
[18:08:43][D][sensor:092]: 'VEML6070 Custom Sensor': Sending state 65535.00000  with 0 decimals of accuracy
```

Don't know if this supposed to work with all the VEML sensors...

ways, 2019-10-09:
I see now it's not. I'll try to adapt it myself and create a new thread.

arfrater, 2019-11-01:
> As a workaround you can use the Custom Sensor component. It works well with other i2c components on the same bus. I only tested the code with an ESP8266.
>
> File: veml6070_custom_sensor.h
>
> ```c++ [...]

What about the VEML6075 will this work around work for this sensor

minsuke, 2019-11-04:
I have VEML6075 and applied your code, but I also always 65535 
but thanks for your sharing and I wish can be applied someday.. ^^

martin3000, 2019-11-06:
I think 65535 is -1 and this means it is dark.

martin3000, 2019-11-07:
For ESP32:

```
#include "esphome.h"
#include "Adafruit_VEML6070.h"

#define I2C_SDAa 16
#define I2C_SCLa 33

#define I2C_SDAb 21
#define I2C_SCLb 22

TwoWire I2Cb = TwoWire(0);
TwoWire I2Ca = TwoWire(1);

class VEML6070CustomSensor : public PollingComponent, public Sensor {
 public:
  Adafruit_VEML6070 uv = Adafruit_VEML6070();
  VEML6070CustomSensor() : PollingComponent(15000) {}
  void setup() override {
    I2Ca.begin(I2C_SDAa,I2C_SCLa,50000);
    I2Cb.begin(I2C_SDAb,I2C_SCLb,50000);
    uv.begin(VEML6070_1_T, &I2Ca);
  }
  void update() override {
    uint16_t cur_uv = uv.readUV();
    ESP_LOGD("custom", "The value of sensor is: %i", cur_uv);
    publish_state(cur_uv);
  }
};
```

```
  - platform: custom
    lambda: |-
      auto veml6070 = new VEML6070CustomSensor();
      App.register_component(veml6070);
      return {veml6070};
    sensors:
      name: "UV"
      unit_of_measurement: "mW/cm²"
      accuracy_decimals: 2
      filters:
      - delta: 0.01
      - multiply: 0.03
      - filter_out: 65535
      - lambda: |-
          if (x > 600) {
            return 0;
          } else {
            return x;
          }

```

SeByDocKy, 2019-11-23:
> For ESP32:
> 
> ```
> #include "esphome.h"
> #include "Adafruit_VEML6070.h"
> 
> #define I2C_SDAa 16
> #define I2C_SCLa 33
> 
> #define I2C_SDAb 21
> #define I2C_SCLb 22
> 
> TwoWire I2Cb = TwoWire(0);
> TwoWire I2Ca = TwoWire(1);
> 
> class VEML6070CustomSensor : public PollingComponent, public Sensor {
>  public:
>   Adafruit_VEML6070 uv = Adafruit_VEML6070();
>   VEML6070CustomSensor() : PollingComponent(15000) {}
>   void setup() override {
>     I2Ca.begin(I2C_SDAa,I2C_SCLa,50000);
>     I2Cb.begin(I2C_SDAb,I2C_SCLb,50000);
>     uv.begin(VEML6070_1_T, &I2Ca);
>   }
>   void update() override {
>     uint16_t cur_uv = uv.readUV();
>     ESP_LOGD("custom", "The value of sensor is: %i", cur_uv);
>     publish_state(cur_uv);
>   }
> };
> ```
> 
> ```
>   - platform: custom
>     lambda: |-
>       auto veml6070 = new VEML6070CustomSensor();
>       App.register_component(veml6070);
>       return {veml6070};
>     sensors:
>       name: "UV"
>       unit_of_measurement: "mW/m²"
>       accuracy_decimals: 2
>       filters:
>       - delta: 10
>       - multiply: 0.03
>       - filter_out: 65535
>       - lambda: |-
>           if (x > 600) {
>             return 0;
>           } else {
>             return x;
>           }
> ```

Why you have 

#define I2C_SDAa 16
#define I2C_SCLa 33

#define I2C_SDAb 21
#define I2C_SCLb 22

TwoWire I2Cb = TwoWire(0);
TwoWire I2Ca = TwoWire(1);

? 

You defined two I2C ports ? if yes why ?

martin3000, 2019-11-23:
Oh I have two I2C devices and I was too lazy to use them on the same I2C port. Wiring was eaysier this way.

SeByDocKy, 2019-12-06:
> Oh I have two I2C devices and I was too lazy to use them on the same I2C port. Wiring was eaysier this way. Of course you need only one.

another question

#define I2C_SDAa 16
#define I2C_SCLa 33

correspond to which GPIO ?  GPIO 16 & 33

martin3000, 2019-12-06:
#define I2C_SDAa 16 
#define I2C_SCLa 33
define SDA on pin GPIO16 und SCL on pin GPIO 33.

SeByDocKy, 2019-12-06:
> #define I2C_SDAa 16
> #define I2C_SCLa 33
> define SDA on pin GPIO16 und SCL on pin GPIO 33.

Ok it's working fine now :)

Thanks for sharing

Johan-Du-Preez, 2020-02-05:
Did anybody ever get this working with the 6075 ? as this is the github library https://github.com/adafruit/Adafruit_VEML6075 and not the one posted above

jirisrba, 2020-02-09:
yes, the VEML6075 works fine. Change the name of the library to github repo and slighly modify the code to read the measurement data.

As VEML6075 can read both of UVA and UVB values, you can publish them as a separate sensor. This is my example using the library SparkFun_VEML6075_Arduino_Library.h

```
#include "esphome.h"
#include <SparkFun_VEML6075_Arduino_Library.h>

class VEML6075CustomSensor : public PollingComponent, public Sensor {
 public:
  VEML6075 uv = VEML6075();

  Sensor *uva_sensor = new Sensor();
  Sensor *uvb_sensor = new Sensor();

  VEML6075CustomSensor() : PollingComponent(15000) {}
  void setup() override {
    Wire.begin();
    uv.begin();
    uv.setIntegrationTime(VEML6075::IT_100MS);
  }

  void update() override {
    float uva = uv.rawUva();
    float uvb = uv.rawUvb();
    ESP_LOGD("custom", "The value of sensor uva is: %.0f", uva);
    ESP_LOGD("custom", "The value of sensor uvb is: %.0f", uvb);
    publish_state(uva);
    uva_sensor->publish_state(uva);
    uvb_sensor->publish_state(uvb);
  }
};
```

Johan-Du-Preez, 2020-02-10:
Thanks for this, what is your config in esphome is it the same as posted above ?

jirisrba, 2020-02-10:
espconfig.yaml:

```
sensor:
  - platform: custom
    lambda: |-
      auto veml6075 = new VEML6075CustomSensor();
      App.register_component(veml6075);
      return {veml6075->uva_sensor, veml6075->uvb_sensor};
    sensors:
    - name: "UVA"
      id: zelva_uva
      unit_of_measurement: "µW/cm²"
      accuracy_decimals: 0
    - name: "UVB"
      id: zelva_uvb
      unit_of_measurement: "µW/cm²"
      accuracy_decimals: 0
```

Johan-Du-Preez, 2020-02-10:
Thanks for that think my sensor is broken as it just keeps on throwing
[20:50:35][D][custom:021]: The value of sensor uva is: 65532
[20:50:35][D][custom:022]: The value of sensor uvb is: 65532

Nornode, 2020-02-10:
Please Mr jirisrba, 
Would you please also include you head of your ESPhome YAML definition?
I'm battling with the includes and libraries part.
Currently mine are looking like this:
```
  name: mains_power_meter
  platform: ESP8266
  board: nodemcuv2
  includes:
    - customfiles/VEML6075_Arduino_Library.h
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6075"
```

Which is giving me a build error:
```
In file included from src/VEML6075_Arduino_Library.h:3:0,
from src/main.cpp:22:
.piolibdeps/Adafruit VEML6075 Library/Adafruit_VEML6075.h:22:32: fatal error: Adafruit_I2CDevice.h: No such file or directory

****************************************************************************
* Looking for Adafruit_I2CDevice.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Adafruit_I2CDevice.h"
* Web  > https://platformio.org/lib/search?query=header:Adafruit_I2CDevice.h
*
****************************************************************************

#include <Adafruit_I2CDevice.h>
^
compilation terminated.
*** [.pioenvs/mains_power_meter/src/main.cpp.o] Error 1
```

jirisrba, 2020-02-11:
I tested two VEML6075 libraries

- https://github.com/sparkfun/SparkFun_VEML6075_Arduino_Library
- https://github.com/adafruit/Adafruit_VEML6075

Both of them works fine. My example above was from SparkFun. 

This is my configuration and Custom Sensor component prepared on D1 Wemos:

veml6075_custom_sensor.h:
```
#include "esphome.h"
#include "Adafruit_VEML6075.h"

class VEML6075CustomSensor : public PollingComponent, public Sensor {
 public:
  Adafruit_VEML6075 uv = Adafruit_VEML6075();

  Sensor *uva_sensor = new Sensor();
  Sensor *uvb_sensor = new Sensor();

  VEML6075CustomSensor() : PollingComponent(15000) {}
  void setup() override {
    Wire.begin();
    uv.begin();
    uv.setIntegrationTime(VEML6075_100MS);
  }

  void update() override {
    float uva = uv.readUVA();
    float uvb = uv.readUVB();
    ESP_LOGD("custom", "The value of sensor uva is: %.0f", uva);
    ESP_LOGD("custom", "The value of sensor uvb is: %.0f", uvb);
    publish_state(uva);
    uva_sensor->publish_state(uva);
    uvb_sensor->publish_state(uvb);
  }
};
```

espconfig.yaml:
```
esphome:
  name: ${esphome_name}
  platform: ESP8266
  board: d1_mini
  includes:
    - veml6075_custom_sensor.h
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6075"

i2c:
  - sda: D2
    scl: D1
    scan: True

sensor:
  - platform: custom
    lambda: |-
      auto veml6075 = new VEML6075CustomSensor();
      App.register_component(veml6075);
      return {veml6075->uva_sensor, veml6075->uvb_sensor};
    sensors:
    - name: "zelva UVA"
      id: zelva_uva
      unit_of_measurement: "µW/cm²"
      accuracy_decimals: 0
    - name: "zelva UVB"
      id: zelva_uvb
      unit_of_measurement: "µW/cm²"
      accuracy_decimals: 0
```

Note that I'm not sure about the units of UVA and UVB measurements .. maybe I am wrong.

SeByDocKy, 2020-02-11:
Anybody managed to tweak this sensor to be waterproof ?

Nornode, 2020-02-11:
>   libraries:
>     - "https://github.com/adafruit/Adafruit_VEML6075"

I was lucky with your config @jirisrba so thank you very much!
However, I had to make a little tweak to your YAML definition:

```
  includes:
    - VEML6075_Arduino_Library.h
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6075"
    - "https://github.com/adafruit/Adafruit_BusIO"
```

**Observe the last library added!**
I tried both in the ESPhome container as well as desktop (Mac OS) without success but working on both with that last library added!

SeByDocKy, 2020-02-11:
I used some sillicone on a VEML6070..... unfortunatly the sensor died after the first rain..... the mems sensor didn't like :( ... I iordered e VEML6075 now thanks to your work :)

Johan-Du-Preez, 2020-02-11:
> > libraries:
> > - "https://github.com/adafruit/Adafruit_VEML6075"
> 
> I was lucky with your config @jirisrba so thank you very much!
> However, I had to make a little tweak to your YAML definition:
> 
> ```
>   includes:
>     - VEML6075_Arduino_Library.h
>   libraries:
>     - "https://github.com/adafruit/Adafruit_VEML6075"
>     - "https://github.com/adafruit/Adafruit_BusIO"
> ```
> 
> **Observe the last library added!**
> I tried both in the ESPhome container as well as desktop (Mac OS) without success but working on both with that last library added!
> 
> @SeByDocKy: I just melted a lot of hotglue on to it...
> I'm waiting for some light to show readings and see if this actually works!

Thanks this works with @jirisrba config combined it is showing readings

Johan-Du-Preez, 2020-02-12:
@Nornode Did u get the correct readings ? As mine just shows 440 mW/cm² or over 1000 mW/cm²
not the uv rating as shown in this link [https://learn.adafruit.com/adafruit-veml6075-uva-uvb-uv-index-sensor/arduino-test](url) not shure if we have to add something else i see this is the arduino code

```/*!
 * @file veml6075_fulltest.ino
 *
 * A complete test of the library API with various settings available
 * 
 * Designed specifically to work with the VEML6075 sensor from Adafruit
 * ----> https://www.adafruit.com/products/3964
 *
 * These sensors use I2C to communicate, 2 pins (SCL+SDA) are required
 * to interface with the breakout.
 *
 * Adafruit invests time and resources providing this open source code,
 * please support Adafruit and open-source hardware by purchasing
 * products from Adafruit!
 *
 * Written by Limor Fried/Ladyada for Adafruit Industries.  
 *
 * MIT license, all text here must be included in any redistribution.
 *
 */
 
#include <Wire.h>
#include "Adafruit_VEML6075.h"

Adafruit_VEML6075 uv = Adafruit_VEML6075();

  // Set the integration constant
  uv.setIntegrationTime(VEML6075_100MS);
  // Get the integration constant and print it!
  Serial.print("Integration time set to ");
  switch (uv.getIntegrationTime()) {
    case VEML6075_50MS: Serial.print("50"); break;
    case VEML6075_100MS: Serial.print("100"); break;
    case VEML6075_200MS: Serial.print("200"); break;
    case VEML6075_400MS: Serial.print("400"); break;
    case VEML6075_800MS: Serial.print("800"); break;
  }
  Serial.println("ms");

  // Set the high dynamic mode
  uv.setHighDynamic(true);
  // Get the mode
  if (uv.getHighDynamic()) {
    Serial.println("High dynamic reading mode");
  } else {
    Serial.println("Normal dynamic reading mode");
  }

  // Set the mode
  uv.setForcedMode(false);
  // Get the mode
  if (uv.getForcedMode()) {
    Serial.println("Forced reading mode");
  } else {
    Serial.println("Continuous reading mode");
  }

  // Set the calibration coefficients
  uv.setCoefficients(2.22, 1.33,  // UVA_A and UVA_B coefficients
                     2.95, 1.74,  // UVB_C and UVB_D coefficients
                     0.001461, 0.002591); // UVA and UVB responses
}

void loop() {
  Serial.print("Raw UVA reading:  "); Serial.println(uv.readUVA());
  Serial.print("Raw UVB reading:  "); Serial.println(uv.readUVB());
  Serial.print("UV Index reading: "); Serial.println(uv.readUVI());

  delay(1000)```

Unfortunately i dont know anything about coding so im not sure at what i am looking at

Nornode, 2020-02-13:
Yeah, looking at the sensor values it spits out now, it's not really looking correct.
It might just be me, don't knowing UAx readings, but how can it be negative?

This is my Grafana output: [Image link](https://ibb.co/fddt7Wk)

@jirisrba Where are you located and do you have similar readings?

jirisrba, 2020-02-18:
@Nornode the VEML6075 sensor is located inside the terrarium close to the UVB lamp. The measurement generates a positive numbers between 100 and 500.

Shamshala, 2020-10-20:
> @Nornode the VEML6075 sensor is located inside the terrarium close to the UVB lamp. The measurement generates a positive numbers between 100 and 500.
Just out of curiosity what UVB lamp do you have and what is your readings? Thanks!

Nornode, 2020-10-28:
> Just out of curiosity what UVB lamp do you have and what is your readings? Thanks!

I got mine outside... Values depend on the weather and the position of the sensor. It tops put at approximately 8000 mw/cm2.

SeByDocKy, 2020-10-28:
> > Just out of curiosity what UVB lamp do you have and what is your readings? Thanks!
> 
> I got mine outside... Values depend on the weather and the position of the sensor. It tops put at approximately 8000 mw/cm2.

how you waterproofed it ?

Shamshala, 2020-10-28:
> > Just out of curiosity what UVB lamp do you have and what is your readings? Thanks!
> 
> I got mine outside... Values depend on the weather and the position of the sensor. It tops put at approximately 8000 mw/cm2.
OMG despite i've quoted the right guy above, now i noticed it didn't pinned the right name. 😆 @jirisrba 

@Nornode Thanks but the problem is for me that my VEML6075 readings look overall ok but according to my testing with UV lamp (with known specs) at least UVB part doesn't correspond.

Shamshala, 2020-10-28:
> > Just out of curiosity what UVB lamp do you have and what is your readings? Thanks!
> 
> I got mine outside... Values depend on the weather and the position of the sensor. It tops put at approximately 8000 mw/cm2.

OMG despite i've quoted the right guy above, now i noticed it didn't pinned the right name. 😆 @jirisrba 

@Nornode Thanks but the problem is for me that my VEML6075's readings look overall ok but according to my testing with UV lamp (with known specs) at least UVB part doesn't correspond.
Readings from outside is kinda useless to determine whether it measures ok, no offense.

Nornode, 2020-10-28:
@Shamshala 
> Readings from outside is kinda useless to determine whether it measures ok, no offense.

No need to point that out. Everyone's use is as good as any. I'm not using it for exact values anyway.

nagyrobi, 2022-05-16:
Trying to compile for ESP8266 with ESPHome 2022.04 and results in error:
```
Compiling /data/sensors-padlas/.pioenvs/sensors-padlas/lib8b5/Adafruit VEML6070 Library/Adafruit_VEML6070.cpp.o
/data/sensors-padlas/.piolibdeps/sensors-padlas/Adafruit VEML6070 Library/Adafruit_VEML6070.cpp:36:10: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://registry.platformio.org/search?q=header:Wire.h
*
**************************************************************

   36 | #include "Wire.h"
      |          ^~~~~~~~
compilation terminated.
*** [/data/sensors-padlas/.pioenvs/sensors-padlas/lib8b5/Adafruit VEML6070 Library/Adafruit_VEML6070.cpp.o] Error 1
========================= [FAILED] Took 53.48 seconds =========================
```

Tried adding:
```
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6070"
    - "https://github.com/adafruit/Adafruit_BusIO"
    - "https://github.com/esp8266/Arduino"
```
No successs.

nagyrobi, 2022-05-16:
OK found the problem, it's caused by https://github.com/esphome/esphome/pull/2608.
For ESP8266 the currently correct configuration needs this, in order to compile:
```
  libraries:
    - Wire
    - "https://github.com/adafruit/Adafruit_VEML6070"
```

nagyrobi, 2022-05-16:
> Anybody managed to tweak this sensor to be waterproof ?

Boxed together with a BH1750 in a [car side light fixture](https://www.aliexpress.com/item/1005002974199373.html):
[image]
[image]
[image]
[image]
[image]
[image]

SeByDocKy, 2022-05-16:
> > Anybody managed to tweak this sensor to be waterproof ?
> 
> Boxed together with a BH1750 in a [car side light fixture](https://www.aliexpress.com/item/1005002974199373.html).
> 
> It's up on the roof for more than 2 years now.
> 
> [image] [image] [image] [image] [image] [image]

Very interesting packaging.... I was sure that UV would be block by the transparent covert.... but  seems not the case

nagyrobi, 2022-05-16:
Well I'm sure it affects the measurement a bit (a few %), but certainly doesn't block it. Usually UV blocking transparent parts are specially designed to do that and usually cost more. If you choose the cheapest product, that will 99.9% NOT blocking UV.

nagyrobi, 2022-06-01:
Here's my working config with a lambda to filter out sudden value decreases due to clouds passing by within 30 minutes. Also text sensors providing level and warning.

```
sensor:
- platform: template
  name: last_good_value
  id: last_good_value
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  disabled_by_default: true

- platform: template
  name: counter
  id: counter
  icon: "mdi:memory"
  entity_category: diagnostic
  state_class: measurement
  disabled_by_default: true

- platform: custom
  lambda: |-
    auto veml6070 = new VEML6070CustomSensor();
    App.register_component(veml6070);
    return {veml6070};
  sensors:
    name: "UV Index"
    icon: "mdi:sun-wireless-outline"
    id: uvindex
    accuracy_decimals: 1
    state_class: "measurement"
    filters:
    - filter_out: 65535
    - heartbeat: 60s
    - multiply: 0.0102  # based on trial and error, result compared to official meteorology institute public values
    - lambda: |-  # filter out sudden value decreases due to clouds passing by within 30 minutes
        if (id(counter).state < 30 && id(last_good_value).state != 0 && (id(last_good_value).state - x) >= 1.0) {
            id(counter).publish_state(id(counter).state + 1);
            return {};
        }
        id(last_good_value).publish_state(x);
        id(counter).publish_state(0);
        return x;
    on_value:
        then:
          - if:
              condition:
                lambda: 'return ( (x < 0.1) );'
              then:
                - text_sensor.template.publish:
                    id: text_uv_level
                    state: "None"
                - text_sensor.template.publish:
                    id: text_uv_warn
                    state: "No measurable value"
          - if:
              condition:
                lambda: 'return ( (x >= 0.1) && (x <= 2.9) );' # "Low"       = sun->fun
              then:
                - text_sensor.template.publish:
                    id: text_uv_level
                    state: "Low"
                - text_sensor.template.publish:
                    id: text_uv_warn
                    state: "Not dangerous"
          - if:
              condition:
                lambda: 'return ( (x >= 3.0)  && (x <= 4.9) );' # "Mid"       = sun->glases advised
              then:
                - text_sensor.template.publish:
                    id: text_uv_level
                    state: "Mid"
                - text_sensor.template.publish:
                    id: text_uv_warn
                    state: "Sunglasses recommended"
          - if:
              condition:
                lambda: 'return ( (x >= 5.0)  && (x <= 6.9) );' # "High"      = sun->glases a must
              then:
                - text_sensor.template.publish:
                    id: text_uv_level
                    state: "High"
                - text_sensor.template.publish:
                    id: text_uv_warn
                    state: "Sunglasses needed, skin protection recommended"
          - if:
              condition:
                lambda: 'return ( (x >= 7.0)  && (x <= 7.9) );' # "Danger"    = sun->skin burns Level 1
              then:
                - text_sensor.template.publish:
                    id: text_uv_level
                    state: "Very high"
                - text_sensor.template.publish:
                    id: text_uv_warn
                    state: "Skin burn 1 degree"
          - if:
              condition:
                lambda: 'return ( (x >= 8.0) && (x <= 25.0) );' # "BurnL1/2"  = sun->skin burns level 1..2
              then:
                - text_sensor.template.publish:
                    id: text_uv_level
                    state: "Danger!"
                - text_sensor.template.publish:
                    id: text_uv_warn
                    state: "Skin burn 2-3 degree"
          - if:
              condition:
                lambda: 'return ( (x > 25) );' # "OoR"       = out of range or unknown
              then:
                - text_sensor.template.publish:
                    id: text_uv_level
                    state: "Error"
                - text_sensor.template.publish:
                    id: text_uv_warn
                    state: "Sensor error, or outer space"

text_sensor:
- platform: template
  name: "UV Level"
  icon: "mdi:weather-sunny-alert"
  id: text_uv_level
- platform: template
  name: "UV Warning"
  icon: "mdi:weather-sunny-alert"
  id: text_uv_warn
```

BatrakovSV, 2024-02-16:
> - multiply: 0.03

Hello.
Based on what data do you apply the multiplication by 0.03?

[image]

matthiasbirkich, 2024-02-18:
https://github.com/adafruit/Adafruit_VEML6075/blob/master/Adafruit_VEML6075.h
#define VEML6075_DEFAULT_UVA_RESPONSE 0.001461 ///< Default for no coverglass
#define VEML6075_DEFAULT_UVB_RESPONSE 0.002591 ///< Default for no coverglass

matthiasbirkich, 2024-02-18:
I would use VEML7700 for ambient light. The range is from 0 - 120 lux.
The resolution is 0.0036 lx/ct (Lux per count)
I would use ML8511 for differenciation between UVA and UVB.
There is a multifunctional sensor 5 in one from Adafruit, which is doing it:
https://www.dfrobot.com/product-2522.html
It will work with UART or I2C and gives you the values in mW/m2 for UVA and UVB