esphome/feature-requests #218: Support for VEML7700 High range (0-120k) Lux Sensor

rradar, 2019-05-08:
**Describe the problem you have/What new integration you would like**

VEML7700 Support.

**Please describe your use case for this integration and alternatives you've tried:**

Measure lux outside and therefor need a high range >100k for direct sun light.

**Additional context**

- https://github.com/adafruit/Adafruit_VEML7700
- https://www.adafruit.com/product/4162

rradar, 2020-01-19:
As a work'a'round I use the (already supported) bh1750 with [this custom code](https://github.com/esphome/feature-requests/issues/148#issuecomment-549578132) (maybe included in next release?) to support measurements up to 121556 Lux :tada:

ways, 2020-04-01:
It can be used like this: [https://0p.no/2020/02/29/veml7700_in_esphome.html](0p.no)

closed 2022-07-07 by nagyrobi