esphome/feature-requests #223: syslog logging

leo-b, 2019-05-11:
**Describe the problem you have/What new integration you would like**
It would be nice if the logger component would support logging via [syslog](https://en.wikipedia.org/wiki/Syslog).

**Please describe your use case for this integration and alternatives you've tried:**
I'd like to use esphome for a device that should work autonomously (without home assistant and mqtt broker) most of the time.
Configuring and running an MQTT broker just to be able to see the log output is cumbersome.

**Additional context**
Maybe [this library](https://github.com/arcao/Syslog) could help.

OttoWinter, 2019-05-11:
I see the value in this, however I think it will be low priority.

The problem with creating a new logging provider is that all loggers need to be quite well tested - when the logger has a bug much of the rest of the system can't work. In some cases even OTA safe mode will fail.

Configuring an MQTT broker is one option - but if using the native API you can also just create an `esphome node.yaml logs` process and that will also show all logs.

TheStaticTurtle, 2021-02-10:
For what it is worth I also wanted one, so I spent the night and made a custom component, you can see it here: https://github.com/TheStaticTurtle/esphome_syslog
The component works fine with my setup (rsyslog + pimpmylogs).
I would love your input on what could be done better

atus42, 2021-03-04:
It works for me. I modified your code a little to be able to natively log to syslog, not just through the logger component.
You may define a new variable like attach_to_logger which defaults to true, and honor it's value like you did width `#ifdef USE_LOGGER` in syslog_component.cpp line 34.

I gave the component an id, like my_syslog, and use `my_syslog->log(LOG_INFO, "${my_esp_name}", "some message");` in a lambda to send "some message" to my syslog.

TheStaticTurtle, 2021-03-06:
Here you go @atus42 it's implemented and should be working 😄 , btw there is an action for sending logs no need to create a lambda function. Maybe open an issue on my repo next time 😅.

ashrafmn, 2021-12-13:
Dears,

i came across this while i was teaching for a remote log solution,, thanks for putting this together.

unfortunately i'm encountering compilation error as below,, I'm using Hassos and esphome add-on 2021.12.0
any suggestion to fix this??

=================================================================
INFO Reading configuration /config/esphome/smini_danasmallroom.yaml...
WARNING 'smini_danasmallroom': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'Asia/Riyadh'
INFO Generating C++ source...
INFO Compiling app...
Processing smini_danasmallroom (board: esp8285; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Syslog> 2.0.0
|-- <ESP8266mDNS> 1.2
Linking /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/syslog/syslog_component.cpp.o: in function `esphome::syslog::SyslogComponent::loop()':
syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponent4loopEv+0x0): multiple definition of `esphome::syslog::SyslogComponent::loop()'; /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/esphome/components/syslog/syslog_component.cpp.o:syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponent4loopEv+0x0): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/syslog/syslog_component.cpp.o: in function `esphome::syslog::SyslogComponent::get_setup_priority() const':
syslog_component.cpp:(.text._ZNK7esphome6syslog15SyslogComponent18get_setup_priorityEv+0x4): multiple definition of `esphome::syslog::SyslogComponent::get_setup_priority() const'; /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/esphome/components/syslog/syslog_component.cpp.o:syslog_component.cpp:(.text._ZNK7esphome6syslog15SyslogComponent18get_setup_priorityEv+0x4): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/syslog/syslog_component.cpp.o: in function `esphome::syslog::SyslogComponent::SyslogComponent()':
syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponentC2Ev+0x14): multiple definition of `esphome::syslog::SyslogComponent::SyslogComponent()'; /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/esphome/components/syslog/syslog_component.cpp.o:syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponentC2Ev+0x14): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/syslog/syslog_component.cpp.o: in function `esphome::syslog::SyslogComponent::SyslogComponent()':
syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponentC2Ev+0x14): multiple definition of `esphome::syslog::SyslogComponent::SyslogComponent()'; /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/esphome/components/syslog/syslog_component.cpp.o:syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponentC2Ev+0x14): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/syslog/syslog_component.cpp.o: in function `esphome::syslog::SyslogComponent::log(unsigned char, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponent3logEhRKSsS3_+0xc): multiple definition of `esphome::syslog::SyslogComponent::log(unsigned char, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'; /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/esphome/components/syslog/syslog_component.cpp.o:syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponent3logEhRKSsS3_+0xc): first defined here
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/syslog/syslog_component.cpp.o: in function `esphome::syslog::SyslogComponent::setup()':
syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponent5setupEv+0x40): multiple definition of `esphome::syslog::SyslogComponent::setup()'; /data/smini_danasmallroom/.pioenvs/smini_danasmallroom/src/esphome/components/syslog/syslog_component.cpp.o:syslog_component.cpp:(.text._ZN7esphome6syslog15SyslogComponent5setupEv+0x40): first defined here
collect2: error: ld returned 1 exit status
*** [/data/smini_danasmallroom/.pioenvs/smini_danasmallroom/firmware.elf] Error 1
========================= [FAILED] Took 11.15 seconds =========================

TheStaticTurtle, 2021-12-13:
I have 0 clues what is wrong, my code hasn't changed for a very long time. Plus I just updated and everything (esphome, hassio, ....) and tried to re-compile, and it works without any issue.
Verify that you have everything up to date
Also, please open an issue on my repo if you have an issue with my code as to not flood esp-home maintainers.

ashrafmn, 2021-12-14:
thanks for your your response,, will do..:)

closed 2022-07-01 by nagyrobi