esphome/feature-requests #224: current_color in AddressableLambdaLightEffect

Roncoroni, 2019-05-11:
**Describe the problem you have/What new integration you would like**

It would be nice to access the current color of a Neopixelbus Light in a lambda effect.
The `current_color` is available in the function `apply` but will not provided to the call to the lambda.

**Please describe your use case for this integration and alternatives you've tried:**

I want to write multiple complex animations which use the actual color.
Until now, if i change the color of the stripe, the animation has no possibility to get this update.

**Additional context**

nope

OttoWinter, 2019-05-12:
You can always check the current color yourself with `id(my_light).get_current_values()`

Roncoroni, 2019-05-20:
ok, thanks for this hint. It works for me.

But i think, to provide the parameter current_value to the lambda is much clearer and easer reusable.

OttoWinter, 2019-05-25:
Yeah but the problem is that it cannot be added easily - the function can only accept one argument, and that is `it`.

Closing as already possible and not easy to add.

closed 2019-05-25 by OttoWinter