esphome/feature-requests #227: Uptime for switches

AalianKhan, 2019-05-13:
**Describe the problem you have/What new integration you would like**

I want a sensor that counts how long a switch is on.
**Please describe your use case for this integration and alternatives you've tried:**

although you can do this with Lambdas and template sensors, It will be nice to implement it in a few lines. And I would like to see some sort of human-readable values instead of just seconds. For example ``1 day, 4 hours and 24 minutes`` or ``01:04:24:16``. This will also be useful if you don't have energy monitoring plugs but know the wattage of the appliance it's connected, Then you can get an estimate of how much watt-hours is being used by that appliance by multiplying the watts by the time it has been on. 
**Additional context**

 This Arduino [Sketch](https://hackaday.io/project/7008-fly-wars-a-hackers-solution-to-world-hunger/log/25043-updated-uptime-counter) records the uptime and converts it to human readable values.
P.S. I want the Human readable values as an option or only a front end thing.

AalianKhan, 2019-05-20:
Does anyone know how to get the state of a switch

OttoWinter, 2019-05-27:
I'd rather have this as a template thing for now. Especially because this component would require so many settings for how data is displayed (for numeric value: seconds, minutes etc; and for textual of course much much more)

If I see this being requested a lot, I might change my mind - but I don't see too much value for this to be built-in.

closed 2019-05-27 by OttoWinter

AalianKhan, 2019-05-27:
Question: will the template sensor be accurate  enough to be used

OttoWinter, 2019-05-27:
Accurate? I'm not sure what you mean there.

Template sensors use the same data all other sensors use as well: floats - usually they provide plenty of precision.

AalianKhan, 2019-05-27:
As in the use for the uptime for switches. The sensor will have to increment every second and I see in the uptime sensor you used millisecond precision. So my question is would the sensor become out of sync

OttoWinter, 2019-05-27:
The way I would do is is use `on_turn_on` on the switch to store when the switch was last turned on in some global variable (in milliseconds). In the template sensor you then just need to substract that value from the current time in milliseconds - and that's it.

AalianKhan, 2019-06-06:
for anyone interested, this is how I did it. the accuracy is to a second.
First added a time component (any component may work)
```
time:
  - platform: sntp
    timezone: American/Toronto
    id: time
```
Then added a sensor 
```
  - platform: template
    name: Heater Uptime
    id: heat_uptime
    update_interval: 1s
    lambda: |-
      if (id(heater).state){
        return id(time).now().time - id(heater_last_turn_on);
      } else {
        return id(heat_uptime).state;
      }
```