esphome/feature-requests #228: Load only referenced libraries

vamichael, 2019-05-13:
**Describe the problem you have/What new integration you would like**

I have a simple 1 second update interval reporting an incrementing number to HA. When power is first applied, it takes about 30-60 seconds for the device reports the first number.  I’d like that to be short as possible.  With an Arduino project, a WiFi connection (even with DHCP) takes just about 10 seconds to connect and report to HA with MQTT.

My only guess is that ESPHome might be loading and initializing code that is not needed.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-05-13:
> My only guess is that ESPHome might be loading and initializing code that is not needed.

That is incorrect. ESPHome only includes code in the binary that is actually used. Plus that would never get to the time range you're talking about.

Please add more info about your setup - if it's native API, use MQTT (native API currently the ESP is the server, and HA is polling every 30-60s for it to be back online).

vamichael, 2019-05-13:
I’m using the Native API.  Do you think MQTT would respond faster after a WiFi connection is established?

OttoWinter, 2019-05-13:
Yes, until https://github.com/esphome/feature-requests/issues/46 is ready.

vamichael, 2019-05-14:
@OttoWinter well somehow I got mqtt working, and it is indeed *a lot* faster.  However, the docs mention upgrading from mqtt to the api, not the other way around so I'm having a tiny bit of trouble.

Unless I specify in HA's configuration.yaml that 
```
mqtt: 
   discovery: false 
```
I end up with duplicate entities (trailed by _2).  I don't want to turn HA discovery off because I'm using it for other things.  Shouldn't one of the [MQTT Client Component](https://esphome.io/components/mqtt.html?highlight=discovery) configuration variables not even present the MQTT items for discovery in the first place?  I tried using  `discovery:false` but that still brought in two of everything.

Note:  I started with `api:` and only added this block:
```
mqtt:
  discovery: true
  discovery_prefix: homeassistant
  client_id: home-assistant-1
  broker: 172.16.68.62
  username: mqtt
  password: mqttpass
```
to the following:

```
esphome:
  name: car_bmw
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "MyWiFi"
  password: "01234567890"
  fast_connect: true  
  manual_ip:
    static_ip: 172.16.68.150
    gateway: 172.16.68.1
    subnet: 255.255.255.0
    
logger:

api:
  password: 'password'
  
ota:
  password: 'password'
  
binary_sensor:
  - platform: homeassistant
    name: "Garage Opened BMW"
    entity_id: input_boolean.bmw_opened_garage
    id: garage_opened_bmw
   
sensor:
  - platform: template
    name: "BMW"
    lambda: |-
      static int prevCount = 0;
      static int num_executions = 0;
      if (!(id(garage_opened_bmw).state)) {
        num_executions += 1;
      }
      else {
        num_executions = 1;
      }
      if (prevCount != num_executions) {
        prevCount = num_executions;
        return num_executions;
      }
    update_interval: 1s
```

vamichael, 2019-05-26:
@OttoWinter do you see any issues with my code?  Sometimes my sensor is returning values to HA in as fast as 6 seconds, sometimes over a minute.  The ESPHome web page detects a connection really fast, which is what I’d like for HA to do.

OttoWinter, 2019-05-27:
1. Disable API and switch over to MQTT
2. Remove all ESPHome entries from the integrations screen
3. Go to the entity registry (configuration -> entity registry) and remove old entities.

I believe you are still using API (or at least the non `_2` entities are). Follow the steps above.

closed 2022-07-01 by nagyrobi