esphome/feature-requests #229: Support for SGP30 Gas/Air Quality Sensor

Skelly82, 2019-05-13:
**Describe the problem you have/What new integration you would like**

It would be nice to include support for SGP30 sensors as they are relatively cheap way to measure amount CO2 gas in the environment. This sensor is also available as i2c breakout board i.e. this from Adafruit:
https://www.adafruit.com/product/3709

**Please describe your use case for this integration and alternatives you've tried:**

To measure CO2 and also TVOC concentration in environment.

**Additional context**

Adafruit already provides library for Arduino:
https://github.com/adafruit/Adafruit_SGP30

OttoWinter, 2019-05-13:
A very similar sensor has just been merged https://github.com/esphome/esphome/pull/536

Do you think that could also work?

Skelly82, 2019-05-13:
As soon as the sensor shows up I'll test and let you know. But as these sensors are from different manufacturers and the i2c addresses are also different I doubt it will work.

OttoWinter, 2019-05-13:
Nah that integration very likely won't work for this sensor.

What I meant is that the CCS811 is very similar, and so a viable option as an alternative to this sensor.

Skelly82, 2019-05-13:
Yes, the CCS811 is an alternative, but I already bought SGP30. I'll stick with Tasmota for now and I'll see if I could manage to create PR for SGP30.

Aleksahek, 2019-09-29:
/config/esphome/sgp30.h
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

`#include "esphome.h"
#include "Adafruit_SGP30.h"

uint32_t getAbsoluteHumidity(float temperature, float humidity) {
    // approximation formula from Sensirion SGP30 Driver Integration chapter 3.15
    const float absoluteHumidity = 216.7f * ((humidity / 100.0f) * 6.112f * exp((17.62f * temperature) / (243.12f + temperature)) / (273.15f + temperature)); // [g/m^3]
    const uint32_t absoluteHumidityScaled = static_cast<uint32_t>(1000.0f * absoluteHumidity); // [mg/m^3]
    return absoluteHumidityScaled;
}

class SGP30 : public PollingComponent {
 public:
  Adafruit_SGP30 sgp;
  SGP30() : PollingComponent(15000) { }
  Sensor *rawH2_sensor = new Sensor ();
  Sensor *TVOC_sensor = new Sensor ();
  Sensor *rawEthanol_sensor = new Sensor ();
  Sensor *eCO2_sensor = new Sensor ();
  

  
  void setup() override {
    sgp.begin();

  }
  
  int counter = 0;
  
  void update() override {
    if (! sgp.IAQmeasure()) {
      Serial.print("Measurement failed");
      return;
    }
    uint16_t eCO2 = sgp.eCO2;
    eCO2_sensor->publish_state(eCO2);
    uint16_t TVOC = sgp.TVOC;
    TVOC_sensor->publish_state(TVOC);
    uint16_t rawH2 = sgp.rawH2;
    rawH2_sensor->publish_state(rawH2);
    uint16_t rawEthanol = sgp.rawEthanol;
    rawEthanol_sensor->publish_state(rawEthanol);
    
    delay(1000);
    
    counter++;
    if (counter == 30) {
      counter = 0;

      uint16_t TVOC_base, eCO2_base;
      if (! sgp.getIAQBaseline(&eCO2_base, &TVOC_base)) {
        Serial.print("Failed to get baseline readings");
        return;
      }
    Serial.print("****Baseline values: eCO2: 0x"); Serial.print(eCO2_base, HEX);
    Serial.print(" & TVOC: 0x"); Serial.println(TVOC_base, HEX);
  }
  }
};`

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
SGP30_sensor.yaml
`esphome:
  includes: 
    - SGP30.h
  libraries:
    - "Adafruit SGP30 Sensor"
  name: qwality_sensor
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: " "
  password: " "

# Enable logging
logger:
 
# Enable Home Assistant API
api:
ota:
i2c:
  sda: GPIO4
  scl: GPIO5
#  scan: True
  id: bus_a
  frequency: 400000Hz
  
sensor:
- platform: custom
  lambda: |-
    auto SGP30_kitch = new SGP30();
    App.register_component(SGP30_kitch);
    return {SGP30_kitch->eCO2_sensor, SGP30_kitch->TVOC_sensor, SGP30_kitch->rawH2_sensor, SGP30_kitch->rawEthanol_sensor};

  sensors:
  - name: "eCO2_k"
    unit_of_measurement: ppm
    accuracy_decimals: 0
  - name: "TVOC_k"
    unit_of_measurement: ppb\t
    accuracy_decimals: 0
  - name: "rawH2_k"
    unit_of_measurement: ch
    accuracy_decimals: 0
  - name: "rawEthanol_k"
    unit_of_measurement: ch
    accuracy_decimals: 0
`

scop, 2019-10-21:
https://github.com/esphome/esphome/pull/679

closed 2019-10-21 by OttoWinter