esphome/feature-requests #231: add variable time format for UPTIME sensor

theniz, 2019-05-14:
**Describe the problem you have/What new integration you would like**

for now, uptime sensor supports only seconds. 
with the seconds, I can not calculate how long it works instantly.
**Please describe your use case for this integration and alternatives you've tried:**

in HA, I used to converte minutes to time format like 1day, 22hours, 14minutes.
my yaml code is like this,
HA Uptime to Date format - HA Online
    ha_online:
      friendly_name: "HA Online"
      unit_of_measurement: "time"
      value_template: >-
        {% set time = (((states.sensor.ha_uptime.state) | round ) * 60 | int) | int %}
        {% set minutes = ((time % 3600) / 60) | int %}
        {% set hours = ((time % 86400) / 3600) | int %}
        {% set days = (time / 86400) | int %}
        {%- if time < 60 -%}
          Less than a minute
        {%- else -%}
          {%- if days > 0 -%}
            {{ days }}d
          {%- endif -%}
          {%- if hours > 0 -%}
            {%- if days > 0 -%}
              {{ ' ' }}
            {%- endif -%}
            {{ hours }}h
          {%- endif -%}
          {%- if minutes > 0 -%}
            {%- if days > 0 or hours > 0 -%}
              {{ ' ' }}
            {%- endif -%}
            {{ minutes }}m
          {%- endif -%}
        {%- endif -%}

It worked very smooth. 
but when I applied same code to esphome, it didn't work. in my understanding i needed to change some codes for esphome. i am noob for c++, i can not.

**Additional context**

i am very appreciated that uptime sensor provide data as time format which couldl be easily understood.
thank you.

pavolholes, 2019-05-24:
I have more simple sensor to do this in Home Assistant, you can try to use it in a meantime:

**configuration.yaml:**
```yaml
- platform: template
  sensors:
    workroom_uptime:
      icon_template: mdi:clock
      device_class: timestamp
      value_template: "{{ (as_timestamp(now())  - (states('sensor.workroom_uptime')) | float) | timestamp_custom('%Y-%m-%dT%H:%M:%S+01:00',True) }}"
```
**Result:**
[image]
No credits to me, I found it somewhere at Discord :)

OttoWinter, 2019-05-25:
Yeah I'd recommend @pavolholes's solution here. This is not going to be simple in ESPHome lambda's (possible, but more lines than the suggested solution).

I also don't see this as an ESPHome feature request - the uptime sensor is more meant for measuring the uptime so that it can be used in automations; for those cases, measuring in seconds makes a lot of sense.

theniz, 2019-05-27:
> I have more simple sensor to do this in Home Assistant, you can try to use it in a meantime:
>
> **configuration.yaml:**
>
> ```yaml
> - platform: template
>   sensors:
>     workroom_uptime: [...]

thank you for your reply. I know it could do in homeassistant.
I will study your code.

theniz, 2019-05-27:
> Yeah I'd recommend @pavolholes's solution here. This is not going to be simple in ESPHome lambda's (possible, but more lines than the suggested solution).
>
> I also don't see this as an ESPHome [...]

OK...I see
I asked if there were any tips.
your reply helped me to study from where.
I appreciate you comment very much.
you could close this thread. Thank you.

closed 2019-05-27 by OttoWinter