esphome/feature-requests #232: Add support for North BLE plant monitor

FireBall1725, 2019-05-15:
**Describe the problem you have/What new integration you would like**

There is a BLE plant sensor similar to the Xiaomi MiFlora BLE sensor called the North plant sensor. It appears to be the exact same hardware as the Xiaomi sensor, including being able to use the same MiFlora config in Home Assistant.

**Please describe your use case for this integration and alternatives you've tried:**

I have been using Home Assistant to monitor my garden, however the plant sensors seem to be right at the edge of the BLE range, as I was only getting updates from my plants every few hours so I was looking for a bridge that I could use to relay the data into Home Assistant. ESPHome works great with the authentic Xiaomi MiFlora sensors I have, but ignores the North ones.

**Additional context**

North's website: http://weliveupnorth.com/plant-monitor

GoPosi, 2019-05-23:
I have a bunch of these that I use with ESPHome, you just need to update the firmware to the latest Xiaomi MiFlora.

Download the Xiaomi Flower Care app on your phone, then you can add the North planet monitor to the app just like a normal MiFlora.  Once connected, update it to the latest MiFlora firmware shown in the app.  It will then work with ESPHome like the Xiaomi branded ones.

FireBall1725, 2019-05-23:
@GoPosi Thanks for this information, I will give that a shot when I get home from work.

FireBall1725, 2019-05-23:
It looks like the plant care app is only for Android, the iOS app from Xiaomi doesn't list the plant monitor device in the list of devices, ill have to acquire an android device to test

FireBall1725, 2019-05-24:
Updating the plant sensors to the Xiaomi firmware worked as expected from @GoPosi's Post

closed 2019-05-24 by FireBall1725