esphome/feature-requests #233: Disable individual LEDs in FastLED

poldim, 2019-05-15:
**Describe the problem you have/What new integration you would like**
Need to permanently disable an array of individual LEDs in a string for various reasons.  This is is a great example: https://youtu.be/q1G4UurGZfQ?t=378

**Please describe your use case for this integration and alternatives you've tried:**
Using Rob's code is great but I'm trying to move things over to ESPHome and am not aware of a way to override LEDs to always being off.

**Additional context**
In Rob's code he calls these annoying LEDs and sets them to black on every color write.

OttoWinter, 2019-05-15:
See light partitions

closed 2019-05-15 by OttoWinter

poldim, 2019-05-16:
Perfect, thanks! I did not come across this while searching!