esphome/feature-requests #234: WS2815 - Add support for this newer addressable LED IC

poldim, 2019-05-15:
**Describe the problem you have/What new integration you would like**

WS2815 appears to be a 12v brother of the 5v WS2812/3.  I have ordered some WS2815 strips but have not received them.
Spec sheet: http://www.witop-tech.com/wp-content/uploads/2017/06/WS2815-12v-addressable-led-chip-specification-.pdf

**Please describe your use case for this integration and alternatives you've tried:**

Once I get the strips I will try running them with the WS2812/3 to see if there are any issues.

**Additional context**

OttoWinter, 2019-05-15:
From [this page](https://forum.arduino.cc/index.php?topic=576510.0) (german) it sounds like it's the same protocol as the WS2812.

poldim, 2019-05-16:
I'll let you know once they get here from China.  I just ordered them so
they should be here in about a month.

yanikdude, 2019-06-12:
Im interested in WS2815 connected to ESP8266 using ESPHome. 

Can the current libraries / configs can cater for RGBW and not just RGB.

OttoWinter, 2019-06-13:
@yanikdude Please see neopixelbus integration.

Closing issue as pretty likely WS2815 already works. If not, reopen issue

closed 2019-06-13 by OttoWinter