esphome/feature-requests #235: ESP8266 UART0 used for uart component when logging disabled

p12J, 2019-05-17:
**Describe the problem you have/What new integration you would like**

Currently on esp 8266 UART0 is used for the logger. I would like to be able to use it with the uart component if logging is disabled
**Please describe your use case for this integration and alternatives you've tried:**

I have a four relay board from aliexpress. It uses a mcu to communicate with an esp8266 (via uart0) that comes with it. This is quite a common board on axp, and it would be nice if I can use the hardware as is with no cross connecting of pins. For this I would need uart0 to be the peripheral used by the uart component in esphome. It would be also good to have an option of sending different things for switch on and switch off.
**Additional context**

brandond, 2019-05-17:
You can't entirely disable logging, there are a bunch of boot messages output by both the chip itself and the bootloader, but you can at least prevent ESPHome from using it by doing something like this:

```yaml:
logger:
  level: NONE
  hardware_uart: UART1
```

p12J, 2019-05-17:
Ok, perfect. This works. How about the second comment about the switch staying on after the data has been sent, and be able to send some other data when switching off? Or should this be in its own request?

brandond, 2019-05-17:
For that I'd probably just set up multiple uart switches. They're a bit odd in that they only send the specified data when the switch is turned on, so for what you want I'd probably use them behind a template switch that toggles the real switches on and off to send the right values. Alternately you could put in a PR to have the uart switches accept both an on and an off value.

https://esphome.io/components/switch/uart.html

p12J, 2019-05-18:
- platform: template
    name: "RE_1"
    turn_on_action:
      - switch.turn_on: RE_1_on
    turn_off_action:
      - switch.turn_on: RE_1_off
       
 #functional relay switches
  - platform: uart
    id: "RE_1_on"
    data: [0x02, 0xAA, 0x02, 0x01, 0x00]

The template switch used above gets toggled off the same way as the uart switch does.

OttoWinter, 2019-05-27:
I think the base issue has been resolved - logging can be disabled via the hardware_uart option (or alternatively you can disable it via `baud_rate: 0` as the docs say).

As for the other questions I'm not sure what the OP means. Anyway, the base issue is resolved, so I'll close this issue.

closed 2019-05-27 by OttoWinter