esphome/feature-requests #238: Support for MAX44009 High range (0-188k) Lux Sensor

rradar, 2019-05-20:
**Describe the problem you have/What new integration you would like**

MAX44009 Support (i2c)

**Please describe your use case for this integration and alternatives you've tried:**

Measure lux outside and therefor need a high range >100k for direct sun light.

No integration right now supports to measure sun with esphome :sun_with_face: :disappointed: 
**Additional context**

https://datasheets.maximintegrated.com/en/ds/MAX44009.pdf
https://github.com/dantudose/MAX44009

Esiravegna, 2019-06-17:
Maybe this code could save you some time.
https://github.com/esphome/feature-requests/issues/29#issuecomment-447176550

OttoWinter, 2019-06-17:
@Esiravegna Thanks for the link

Marking this as duplicate of https://github.com/esphome/feature-requests/issues/29

closed 2019-06-17 by OttoWinter