esphome/feature-requests #239: Support for  Waveshare Three-color E-Ink Displays

ASMfreaK, 2019-05-21:
**Describe the problem you have/What new integration you would like**

References #77. 
I've got Waveshare's 3-color e-ink display and it would be nice to have some support for them.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

ASMfreaK, 2019-05-26:
After some fiddling around, I can report that implementing color in 7.50 modules is quite hard due to memory.
Consider this (numbers in braces are reported by PIO): 
* The ESP I use have around 80K(81920) of RAM.
* ESPHOME in minimal configuration consumes around 30K(31300) of RAM.
* Waveshare Display in BW mode consumes ![\dfrac{640 \cdot 384}{8} = 30720](https://latex.codecogs.com/png.latex?%5Cdfrac%7B640%20%5Ccdot%20384%20%7D%7B8%7D%20%3D%2030720%20%7E%5Ctext%7Bbytes%7D), totaling in ![81920 - (31300 + 30720) = 19900 ~\text{bytes}](https://latex.codecogs.com/png.latex?81920%20-%20%2831300%20&plus;%2030720%29%20%3D%2019900%20%7E%5Ctext%7Bbytes%7D) of free space
* A naive color implementation (2bits) needs a buffer of ![\dfrac{640 \cdot 384}{4} = 61440 ~\text{bytes}](https://latex.codecogs.com/png.latex?%5Cdfrac%7B640%20%5Ccdot%20384%7D%7B4%7D%20%3D%2061440%20%7E%5Ctext%7Bbytes%7D), resulting in allocation fail - ![81920 - (31300 + 61440) = -10820 ~\text{bytes}](https://latex.codecogs.com/png.latex?81920%20-%20%2831300%20&plus;%2061440%29%20%3D%20-10820%20%7E%5Ctext%7Bbytes%7D).

I'll try to use flash memory for this. This result will also affect our ability to implement color LCD displays.

OttoWinter, 2019-05-27:
@ASMfreaK Hmm yeah that's true.

Of course the ESPHome memory usage could be optimized a bit (though most of that number probably comes from the SDK wifi system etc), but that would not solve the problem for longer.

At least on ESP32s this will be a much smaller problem (I think they had like 6 times as much RAM, but don't quote me on that). So if it turns out that it can't run on ESP8266s let's just make it ESP32-only.

sbussinger, 2019-06-19:
I'd also like support for WaveShare's 3 color e-ink displays, but I'm interested in the smaller displays (1.5" 200x200). So let's not make it ESP32 only since the smaller buffer required should fit on an ESP8266.

ASMfreaK, 2019-07-07:
@sbussinger I've written and tested the code for 7.5in color display. I could extend it for all color displays, but I need someone to test the code.

tommy-hartmann, 2019-07-23:
I have a waveshare 4.20in BWR Display and i'd like to test your code.

minsuke, 2019-10-26:
> @sbussinger I've written and tested the code for 7.5in color display. I could extend it for all color displays, but I need someone to test the code.

you developed 3 color? If I can use 3 color I'll buy it, I wich you can share the code with esphome library. you great !!!

Cabbo, 2020-08-23:
What about implementing the smaller color ones? I could test 1.54in-c

shacal, 2020-11-08:
Any update on this, I have hardware to test this out.

lbonherbe, 2021-01-19:
Hello all!

Since then Waveshare has released a flurry of epaper screens - the list is long but a lot of good ones, with partial refresh, almost instant refreshes etc. We can see the list here for example: [https://www.waveshare.com/product/displays/e-paper/epaper-2/3.7inch-e-paper.htm](url)
It would be good to add those in ESPHome as they seem to be changing the game of epaper display!

brettp, 2021-03-13:
Is there a ticket to track color support for these three-color displays? 

I can only find a few references about specific displays (#892, [a reference in a PR to another PR](https://github.com/esphome/esphome/pull/663#discussion_r302421163), and [possibly the referenced PR itself](https://github.com/esphome/esphome-core/pull/573)).

Xaxix, 2021-03-27:
i will love the support of the 3 color 4.2 display!

martin3000, 2021-07-24:
Is there a way to use the colored version without using the red color? just to make it work with b&w?

oscaropenness, 2021-07-28:
> Is there a way to use the colored version without using the red color? just to make it work with b&w?

The displays **are working in b&w** there is just no option yet to make use of the color.

martin3000, 2021-07-28:
[image]
So you think this should look ok? The problem is not the colro version?

henriklied, 2021-11-30:
> [image]
> So you think this should look ok? The problem is not the color version of the panel?

Same issue here. Hoping that there's a way to address this.

parats15, 2021-12-14:
Hello,
I just finish to create a new waveshare epaper screen with red color management
[waveshare_epaper.zip](https://github.com/esphome/feature-requests/files/7707995/waveshare_epaper.zip)
Just you need to replace the waveshare_epaper folder "esphome\components\waveshare_epaper"

On you yaml file you add:

```
color:
  - id: COLOR_RED
    red: 100%
    green: 0%
    blue: 0%
    white: 0%`

display:
  - platform: waveshare_epaper
    model: 7.50inv2b
    [....]
```

For the colors:
- Black : COLOR_ON
- White : COLOR_OFF
- Red : COLOR_RED

parats15, 2021-12-14:
> [image] So you think this should look ok? The problem is not the color version of the panel?

Morever your epaper sceen is an 800 x 480 px
https://www.waveshare.com/wiki/Pico-ePaper-7.5-B

TheFitzZZ, 2021-12-17:
https://github.com/esphome/feature-requests/issues/239#issuecomment-993076104 @parats15 
First of all thank you for adding the color support! 

I got my 7.5" screen working with `model: 7.50in`, but with `model: 7.50inv2b` it can't update anymore: `[E][waveshare_epaper:107]: Timeout while displaying image!`

Is the model just wrong in my case and for mine it's not supported yet?

I got this one (driven by a ESP32 driver board):
[image]

TheFitzZZ, 2021-12-17:
Ah, I got the v1 - which is the lower version. Would you be able to provide the color support for that one as well by any chance @parats15?

parats15, 2021-12-18:
@TheFitzZZ 
I updated my branch, now you can use the red color on `model: 7.50in`
But I haven't this screen so I don't know if this work correctly.
Keep me posted

Also if you have recently adding the external component, add `refresh: 1min` in `external_components:`
https://esphome.io/components/external_components.html

TheFitzZZ, 2021-12-18:
@parats15 thanks for the swift response (and refresh hint)!

The red does work, yet now I have vertical "interlacing" lines:
[image]

Any chance to fix that? Also please drop me your paypal or so, that I can buy you a drink :-)

parats15, 2021-12-18:
Yes think I found my mistake,
I pushed fix

I updated my bio on my github page, Thank

parats15, 2021-12-18:
I just pushed another little fix ^^

TheFitzZZ, 2021-12-18:
Brilliant! Sent a little "thank you" via PayPal :-) 
[image]

Now to lower the update cycle... but I think I can find that myself hehe

Adrian-at-CrimsonAuzre, 2022-01-06:
So I'm guessing their [7 color displays](https://www.waveshare.com/5.65inch-e-paper-module-f.htm) are out of the question for the moment?

Tokn59, 2022-01-07:
Love to see support for the LOLIN 2.13 3-color model. Created feature request already.

patman15, 2022-01-10:
I have implemented red color [support for the 4.20in version 1 model](https://github.com/parats15/esphome/pull/1). I did a pull request against your version @parats15. Are you considering to do a pull request against the main version?
[image]

parats15, 2022-01-11:
@patman15 
Thank you for adding this screen. It's ok I merged you PR.

parats15, 2022-01-11:
@Adrian-at-CrimsonAuzre 
Yes it's possible to add 7 color control on your screen.
Your screen is 600 × 448 × 3bits color => 806400bits in ram.
Technically it should work with an ESP32 WROOM 32 board.

Just I need to work in my branch to manage your screen and more than 3 color.

patman15, 2022-01-11:
> @patman15 Thank you for adding this screen. It's ok I merged your PR.

@parats15 When implementing I noticed a few things:
1. `COLOR_ON` is used for drawing on the e-paper which appears black. On the other hand it is defined as `const Color COLOR_ON(255, 255, 255, 255);` actually meaning the color white. I looked at the implementation for [Inkplate6](https://esphome.io/components/display/inkplate6.html) which uses `COLOR_OFF` (which seems logic to me as the display draws in black). For me it makes issues when moving code from one display to the other as the behavior is exactly opposite. Also I think it could start making issues when rendering pictures for displays with more colors. Easy to change, not sure what you think.
2. I had to do a separate class for the B/W/R display although it is practically identically to the B/W display except that it uses the second memory page for the red color. I would have needed a function to query which (not how many) colors exist to just merge it to the class that already does the B/W display. Probably this is an issue more towards the upstream?
3. The code currently uses a color identity comparison for rendering. I looked at the [waveshare documentation](https://www.waveshare.com/w/upload/a/a7/E-Paper_ESP8266_Driver_Board_EN.pdf) on page 7 you can find hints on mappings that would make it a more generic implementation (independent of the supported colors and levels). Nevertheless, it would need a map of colors that a specific display supports.

Let me know if you want me to support you with any of these topics in case you see it the same way. We could also ask the upstream maintainer on how it is intended.

evadim, 2022-01-11:
I have an 1.54 200 x 200 B/W/R module (from MH-ET Live) and can test it if someone implement support of it.

davideZavattero, 2022-02-06:
> Hello, I just finish to create a new waveshare epaper screen with red color management On you yaml file you add:
> 
> ```
> external_components:
>   - source:
>       type: git
>       url: https://github.com/parats15/esphome.git
>       ref: waveshare-color
>     components: [ waveshare_epaper ]
> 
> color:
>   - id: COLOR_RED
>     red: 100%
>     green: 0%
>     blue: 0%
>     white: 0%`
> 
> display:
>   - platform: waveshare_epaper
>     model: 7.50inv2b
>     [....]
> ```
> 
> For the colors:
> 
> * Black : COLOR_ON
> * White : COLOR_OFF
> * Red : COLOR_RED

Hi, 
I implemented this for my waveshare 2.9in b/w/r, but i'm missing something. it doesn't work.

light-red background should be white and the number 8 red.

[image]

here my code.

```yaml
[...]
    
external_components:
  - source:
      type: git
      url: https://github.com/parats15/esphome
      ref: waveshare-color
      #ref: dev
    components: [ waveshare_epaper ]

    
spi:
  clk_pin: GPIO17
  mosi_pin: GPIO16

color:
  - id: COLOR_RED
    red: 100%
    green: 0%
    blue: 0%
    white: 0%

display:
  - id: "schermo_latte"
    platform: waveshare_epaper
    cs_pin: GPIO5
    dc_pin: GPIO18
    busy_pin: GPIO21
    reset_pin: GPIO19
    model: 2.90in-b
    update_interval: never
    lambda: |-
    
      float livello= id(livello_latte).state;
      int livelloPrint= floor(livello);
      float perc= (livello /1000)*100;
      int tazze= floor(livello / 140);
      
      int levels[]={100,75,50,25,10,0};
      float diff=100;
      int currLevel=100;
      
      for (int i = 0; i < 6; i++){
        if(abs(levels[i] - perc) < diff){
          diff= abs(levels[i] - perc);
          currLevel=levels[i];
        }
        ESP_LOGD("main", "diff: %f, levels[i]: %d, prec: %f, currLevel: %d", diff, levels[i], perc, currLevel);
      }
      
      it.fill(COLOR_ON);
      
      if(currLevel == 100){
        it.image(27,30, id(milk_100), COLOR_OFF, COLOR_ON); 
      }
      
      if(currLevel == 75){
        it.image(27,30, id(milk_75), COLOR_OFF, COLOR_ON); 
      }
      
      if(currLevel == 50){
        it.image(27,30, id(milk_50), COLOR_OFF, COLOR_ON); 
      }
      
      if(currLevel == 25){
        it.image(27,30, id(milk_empty), COLOR_OFF, COLOR_ON); 
        it.image(32,150, id(level_25), COLOR_OFF, COLOR_ON);
      }
      
      if(currLevel <= 10){
        it.image(27,30, id(milk_empty), COLOR_OFF, COLOR_ON); 
        it.image(31,174, id(level_10), COLOR_RED, COLOR_ON);
      }
      
      
      if(id(frigo_temp).state){
        if(id(frigo_temp).state > 4){
          it.filled_rectangle(0, 0, 128, 24, COLOR_RED);
          it.printf(119, 0, id(roboto_black_20), COLOR_ON, TextAlign::TOP_RIGHT, "%.1f",id(frigo_temp).state);
          it.print(119, 5, id(roboto_medium_8), COLOR_ON, TextAlign::TOP_LEFT, "°C");
        }
        else{
          it.printf(119, 0, id(roboto_black_20), COLOR_OFF, TextAlign::TOP_RIGHT, "%.1f",id(frigo_temp).state);
          it.printf(119, 5, id(roboto_medium_8), COLOR_OFF, TextAlign::TOP_LEFT, "°C");
        }
      }
      
      if(currLevel <= 25){
        it.filled_rectangle(0, 198, 128, 98, COLOR_RED);
        
        
        it.print(64,222, id(iconfont_23), COLOR_ON, TextAlign::BOTTOM_CENTER, "\ue900");
        it.printf(64, 221, id(roboto_black_44), COLOR_ON, TextAlign::TOP_CENTER, "%d",livelloPrint);
        
        it.print(40,271, id(iconfont_23), COLOR_ON, TextAlign::TOP_CENTER, "\ue901");
        it.printf(73, 269, id(roboto_black_23), COLOR_ON, TextAlign::TOP_CENTER, "%d", tazze);
      }
      else{
        it.print(64,220, id(iconfont_23), COLOR_OFF, TextAlign::BOTTOM_CENTER, "\ue900");
        it.printf(64, 221, id(roboto_black_44), COLOR_OFF, TextAlign::TOP_CENTER, "%d", livelloPrint);
        
        it.print(40,271, id(iconfont_23), COLOR_RED, TextAlign::TOP_CENTER, "\ue901");
        it.printf(73, 269, id(roboto_black_23), COLOR_RED, TextAlign::TOP_CENTER, "%d", tazze);
        
      }
      
      ESP_LOGD("main", "DISPLAY livello: %f, tazze: %d, currLevel: %d",livello, tazze, currLevel);
```

BluetriX, 2022-04-05:
Hello, 
today i tested the code from @parats15 on my 4.2bv2 display. I had to adjust some code here and there (in the fork of @parats15 the 4.2bv2 does not exist yet).

The code from @parats15 works so far I think. Can we finish developing this and merge it (pull request)? It would be a pity if the functionality is not in esphome.

Greetings,

Erik

Eriner, 2022-05-14:
I added support for the 7.5in-bv2 & 7.5in-bv3 (black and white only) in https://github.com/esphome/esphome/pull/3121, but would like to add color support, but I'm unsure if other (esphome) API changes will be necessary for interacting with the displays; I haven't invested the time to answer that. Perhaps someone more familiar with esphome can help me out in getting started here?

Chief-T1, 2022-05-15:
Thanks for your work on this. It will be an amazing thing to have. 

I'm looking to try and implement this on my set up but am a bit confused about what I put in the lambda if I just want to use a single image. 

Does the colour processing happen automatically if I just put this single line? 
`it.image(0, 0, id(my_image));`

Or do I need a script where it will define when a pixel is red, white or black? 

Thanks

bnnhpt, 2022-06-17:
> Hello, today i tested the code from @parats15 on my 4.2bv2 display. I had to adjust some code here and there (in the fork of @parats15 the 4.2bv2 does not exist yet).
>
> The code from @parats15 works [...]

Hi Erik,
I'm stuggling with a Waveshare 4.2 red-black-display. Is it the same display (4.2bv2) that worked with you? Can you share your adjusted code and maybe the esphome-yaml with me?
My display doesn't refresh proper and it takes minutes to show up.

Thanks

bzzbzzz, 2022-07-08:
> I added support for the 7.5in-bv2 & 7.5in-bv3 (black and white only) in https://github.com/esphome/esphome/pull/3121, but would like to add color support, but I'm unsure if other (esphome) API [...]

@Eriner @parats15 
Did one of you two maybe happen to find the motivation and time to make a 3 color version for 7.5in-bv3 possible?
I can't find the older 7.5in displays where I live without paying ca 90 dollar for just the shipping, only v3 available ..

parats15, 2022-07-09:
> > I added support for the 7.5in-bv2 & 7.5in-bv3 (black and white only) in [esphome/esphome#3121](https://github.com/esphome/esphome/pull/3121), but would like to add color support, but I'm unsure if other (esphome) API changes will be necessary for interacting with the displays; I haven't invested the time to answer that. Perhaps someone more familiar with esphome can help me out in getting started here?
> 
> @Eriner @parats15 Did one of you two maybe happen to find the motivation and time to make a 3 color version for 7.5in-bv3 possible? I can't find the older 7.5in displays where I live without paying ca 90 dollar for just the shipping, only v3 available ..

Hi @bzzbzzz, are you tested your V3 epaper screen with the model: 7.50inv2b ? Because when I check the documentation, I don't find difference between the V2 and V3. So I think you can use the 7.50inv2b for the V3 B screen.
Let me know if it works or not.

bzzbzzz, 2022-07-09:
> Hi @bzzbzzz, are you tested your V3 epaper screen with the model: 7.50inv2b ? Because when I check the documentation, I don't find difference between the V2 and V3. So I think you can use the 7.50inv2b for the V3 B screen. Let me know if it works or not.

@parats15 could be that I'm just too stupid to use the external_components, but I get an error validating, saying `[reset_duration] is an invalid option for [display.waveshare_epaper]`
which I guess I need, not because of the V3 display, but because of the [Rev2.2 HAT that came with it](7.50in-bV2https://github.com/parats15/esphome/commit/cec4a81e14322b970134a1ffff87cb4cdf42db85) (the display doesn't work without the external component, usually working config and `reset_duration: 2ms` commented out).
If I add the external component and comment out the reset_duration, the validation passes, but 
[compiling fails](https://www.toptal.com/developers/hastebin/inececogeb.yaml) (this is where I guess I'm [too stupid to use the external component?](https://www.toptal.com/developers/hastebin/ebisemoveh.yaml) ).

parats15, 2022-07-09:
I think I understand the mistake, my fork is based on old version of esphome (Dec 11), so the reset_duration param has been implemented on Dec 12. This param not exist on the color version. For the moment remove `reset_duration: 2ms` line.
I know I should rebase my changes on the last dev version.

atomicmike, 2022-07-09:
@bzzbzzz I have a 7.5in v3 panel, which I wasn't able to get working without having reset_duration set to 2ms. There's also a breaking change in ESPHome 2022.6 that prevents the current fork from @parats15 from compiling (2022.6 adds a DisplayType property to the display classes). I made a fork that has the necessary changes (but only for the 7.5in display). You're welcome to try it out and see if it works for you...

```
external_components:
  - source:
      type: git
      url: https://github.com/atomicmike/esphome.git
      ref: waveshare-color-2022.6
    components: [ waveshare_epaper ]
```

bzzbzzz, 2022-07-11:
> external_components:
>   - source:
>       type: git
>       url: https://github.com/atomicmike/esphome.git
>       ref: waveshare-color-2022.6
>     components: [ waveshare_epaper ]
> ```
it works, thanks! the only two things that are a bit strange are, that I now have an additional 1mm thick red border (deep red) and the colors of the rest of the screen are a bit brighter / more washed out than the red border (directly after updating, not getting lighter over time)

lawleo, 2022-08-17:
it works for my 7.5in red-black-white too, thanks @atomicmike

turbatus, 2022-08-21:
> Hello, today i tested the code from @parats15 on my 4.2bv2 display. I had to adjust some code here and there (in the fork of @parats15 the 4.2bv2 does not exist yet).
>
> The code from @parats15 works [...]
@BluetriX 
I know it's been a while, but can you post your config? I am struggling to get my 4.2 e-paper to work. How did you managed to make it work? I get compilation errors about some abstract class missing or something similar. Thank you

guardmedia, 2022-09-21:
Thank you for all the work @parats15 and @atomicmike for getting this working! I managed to get my 7.5b V3 display working with color support with your help!

In hopes that this gets merged soon, I created a simplified re-based branch based on current latest from esphome\dev which I hope makes it easier to get merged. I only have a 7.5b V3 display, hoping it works on other displays too.

Hoping we are close to getting a pull request submitted for this

Threnklyn, 2022-11-21:
I have the `WaveshareEPaper7P5InHDB` Screen and can't get it to work.
Tried to implement it like the other screens but I get the error `Could not allocate buffer for display!`
With a 880x528 screen you need a buffer of `(880 x 528 x 2) / 8 = 116160 bytes`
Any idea where I should look next for the error? 116kb should fit into an esp32

parats15, 2022-11-25:
Hello @Threnklyn 
"116kb should fit into an esp32" technically yes, but in the esphome the wifi, ota, homeassistant feature take too much memory also. And the buffer of 116160 bytes it is a malloc and must be memory aligned in memory it can't split the buffer.
So if you have activated some functionality, try to light your code.

JeroenTuinstra, 2022-11-28:
Hi everyone, 

Just testing a new screen i received. This is the ```7.50in-hd-b``` with red color. Difficult to find as they no-longer make them.
This is the code I am using - unfortunately nothing is printed. The screen refreshes, but that text is not showing in red. 

```
esphome:
  name: overview

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "###############"

ota:
  password: "################"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/guardmedia/esphome.git
      ref: waveshare-color
    components: [ waveshare_epaper ]

color:
  - id: COLOR_RED
    red: 100%
    green: 0%
    blue: 0%
    white: 0%

font:
  - file: "fonts/AcuminPro-Medium.ttf"
    id: titel
    size: 30

spi:
  clk_pin: 18
  mosi_pin: 23

display:
  - platform: waveshare_epaper
    cs_pin: 22
    dc_pin: 21
    busy_pin: 5
    reset_pin: 13
    model: 7.50in-hd-b
    update_interval: 30s
    reset_duration: 2ms
    rotation: 90°
    id: epaper_display
    lambda: |-
      it.print(200, 200, id(titel), COLOR_RED, "Hello World!");
```
Could anyone help?

Threnklyn, 2022-11-28:
The 7.50in-hd-b is not implemented in @guardmedia branch yet.
I tried it but failed:
> I have the `WaveshareEPaper7P5InHDB` Screen and can't get it to work. Tried to implement it like the other screens but I get the error `Could not allocate buffer for display!` With a 880x528 screen [...]

JeroenTuinstra, 2022-11-29:
I don't get the error @Threnklyn mentions. My simply doesn't show the red. The screen refreshes, so it works, yet red is not shown.

Threnklyn, 2022-11-30:
That's because I implemented the color functionality for this display type.
But it did not work.
Like I said his branch does not support this display yet

1liminal1, 2022-12-03:
Hey all, 

Having issues with getting my display to work correctly. Can someone 7.50in B-type (red-black-white) is supported, just want to rule that out?

Thanks :)

exc2k, 2022-12-13:
> Hello, today i tested the code from @parats15 on my 4.2bv2 display. I had to adjust some code here and there (in the fork of @parats15 the 4.2bv2 does not exist yet).
>
> The code from @parats15 works [...]

@BluetriX, I am havin massive problems with the 4.2bv2 display too (Even when only sending black pixels). Since it looks like your code has not been merged, would you please share it!!! Vielen Dank :)

TK0130, 2023-01-04:
> 7 color

Attached is example code for the 7 color e-ink (7 color, 7.3" epaper hat)

Just thought it might be useful as I can get that working, I just don't have the skill to make it work in the memory of an ESP32 along with ESPHOME etc.
[epd7in3f.zip](https://github.com/esphome/feature-requests/files/10340750/epd7in3f.zip)

ThomDietrich, 2023-01-31:
Hello everyone,
I am new to this thread. I can see that there are quite a few options for specific displays and their own limitations offered.

Why is there no general solution to this yet? What is missing for the esphome project to bring multiple colors to the waveshare displays?

Threnklyn, 2023-01-31:
more ram for the ESP32 and pull requests i guess.

1liminal1, 2023-02-08:
I am also struggling with this to display correctly, even just white and black :(

Any news on the new code mentioned?

1liminal1, 2023-02-25:
Yeah me too, Even if there was a way to use the three colour without the red??

[image]

It just looks so rubbish, any workarounds?

michaelfuckner, 2023-04-08:
I have the 7.5" RBW Display, too and if you clear the red buffer and draw it, it looks fine for me (currently sample code in pico). Did you rotate the image? Perhaps you used 480x800 instead of 800x480?

bipsendk, 2023-04-19:
Found this thread trying to figure out of the 2.9inch red/black is supported as well ?
I have a couple of [these](https://www.aliexpress.com/item/1005003163275994.html?spm=a2g0o.order_list.order_list_main.5.26be1802NYnaWE), that I would like to use :-)

[149 more replies]