esphome/feature-requests #242: New sensor: Ping

mseefeldt, 2019-05-23:
**Describe the problem you have/What new integration you would like**

Currently ESPHome already supports the Wifi and ethernet network.
A sensor which can regularly ping a host and report back the response time would be great.
(Similar to the HomeAssistant sensor)

**Please describe your use case for this integration and alternatives you've tried:**

I have a remote network to supervise (in a basement) where there is no cellular or wifi
network available (massive construction, tons of steel in the walls). I need to monitor the
hosts in the network and report the data regularly. Power consumption is an issue here since
the monitoring needs to be battery powered to work as well during a power outage. So any ESP is a must, i can't use a raspberry pi with HomeAssistant.

Having the ping sensor would solve half of my problem, only the transmission towards
my host is an another issue. LoRaWan does reach the outer world though but that's another
feature.

**Additional context**

OttoWinter, 2019-05-24:
See status binary sensor - that probably is the solution to the problem.

Also ICMP pings are pretty hard because the ESP SDK does not expose the necessary APIs last time I looked (there are some undocumented ones but they do not work well).

xdss, 2019-07-10:
also interested in something like this within esphome, would implementation be similar to https://github.com/dancol90/ESP8266Ping

bepstein111, 2020-02-20:
Have been looking for this exact same function for weeks. Binary ping sensor which could return a simple true or false for the presence of a device on the network. Would be HUGE, I'm tellin ya!

adrianmihalko, 2020-03-20:
This would be great. Ping a device on the network and cycle relay if ping failed. Combined with fail counter would make it a nice failover device.

randybb, 2020-03-20:
Normally I would disagree as this can be implemented on HA itself and both are "sitting" on the same network, but yes - ESP can act as "watchdog" or remote reset device and then ping/snmp/ssh/telnet/nmap are essential.

magiva, 2020-04-27:
being able to ping HA to see if its alive and then reset it would be a bonus when away from home and HA locks up

bperkins011, 2020-04-30:
I'd also like a ping.  Vacation home has a network with HA, etc.  Between the cable modem, the Unifigateway, the switch, Home Assistant, etc.  Something always dies and there is no network.
If there was a ping that I could put in an IP address (8.8.8.8) and it came back - great.  Otherwise I have it toggle the smartplug outlet - wait 15 minutes for it all to come back up and check again.
A router watchdog reset for the power strip is the goal...

rmrfus, 2020-08-29:
+1 for the feature. ping sensor would be helpful for watchdog type of things.

Fusseldieb, 2020-09-28:
I have a flaky ESP32Cam which streams just fine, but if it's not being "watched"/"used", the connection becomes unstable and when I access it again it needs several seconds to recover. 

With a ping sensor I could let it be "active" the whole time pinging the Home Assistant Interface. I know, this is an awkward workaround, but if it works... :)

Another thing I've considered in my scenario is to prerecord the stream inside HA.

sergeolkhovik, 2020-09-28:
Three is a status sensor, i think it should work for you. I'm checking
ESPHome device availability exactly with that status sensor.

wasyl00, 2021-01-10:
+1 for ping. I would love to have ping sensor to be able to reboot my router when it freezes.

ict-one-nl, 2021-04-23:
Good idea, same use cases relevant to me as already described.

trombik, 2021-07-19:
as I also needed ICMP sensor and wanted to learn the custom sensor component,  just hacked up one today, nothing more than PoC, however.

https://github.com/trombik/esphome-component-ping

if you have more knowledge of `esphome` _and_ `C++` than I do, you should be able to rewrite it as an external component.

trombik, 2021-07-20:
> if you have more knowledge of `esphome` _and_ `C++` than I do, you should be able to rewrite it as an external component.

after painful "banging head on wall" hours, the component is not a custom component anymore.

```yaml
external_components:
  - source:
      type: local
      path: ../components

sensor:
  - platform: ping
    ip_address: 8.8.8.8
    num_attempts: 17
    timeout: 1sec
    loss:
      name: Packet loss
    latency:
      name: Latency
      accuracy_decimals: 3
```

pilotak, 2021-07-20:
@trombik you should make a PR! this is awesome

trombik, 2021-07-20:
I will not create a PR because:

* the code is not `lint`ed
* the code does not follow the code style
* the code is not well-tested in the real world
* the code is not tested on `esp32`
* the code has not been used with other components
* the code is not confirmed not to break other components
* the code is not documented
* the CI does not test anything other than compiler can build the code

if you would like the code to be merged, you are welcome to help fixing the issues above. adding another `+1` will not make it happen.

adrianmihalko, 2021-07-20:
This is awesome, I hope it will be merged at some point.

bepstein111, 2021-07-20:
> I will not create a PR because:

> * the code is not tested on `esp32`
> * the code has not been used with other components
> * the code is not confirmed not to break other components

So, 8266? What _has_ it been tested with?

trombik, 2021-07-22:
now both `esp8266` and `esp32` should work. see the `README.md` in the repository. bug reports, comments, and anything about the component should go to https://github.com/trombik/esphome-component-ping from now on.

Kilowatt-W, 2021-10-30:
Really nice component, hope it will get standard at all :-)

lsaadeh, 2022-02-04:
any update on this one

bepstein111, 2022-02-04:
> any update on this one

On what? The link above works (https://github.com/trombik/esphome-component-ping)

marianomd, 2022-04-12:
I want to check if esp is alive from another esp, then send a telegram alert if it's down. Would this be the correct way? Or is there something integrated that would fulfill this functionality?

closed 2022-07-01 by nagyrobi