esphome/feature-requests #243: Make the status sensor available for the ESP device internally

subzero79, 2019-05-24:
**Describe the problem you have/What new integration you would like**

I want the esp device to know when is connected/disconnected from the hass api
**Please describe your use case for this integration and alternatives you've tried:**

I thought the status binary sensor was reported internally, i actually never tested it. But i have it as a condition of automation execution, just to realise two days ago this only works on the ha side. This is very important when the relays with switch input like shelly drive smart lights (zigbee or wifi smart bulbs). In this case i have to make sure two conditions are met in the esp automation:
1) ESP is connected to home assistant(api available). So the esp can make service calls
2) The zigbee controller is available, which is a sensor that reports to esphome via ha.

Any of this conditions fails the esp should toggle the relay internally as a failover.

This is an example i use:

```yaml
substitutions:
  devicename: shelly2_kitchen_2
  upper_devicename: "Shelly2 Kitchen 2"
  relay1_name: "Kitchen Fan"
  relay2_name: "Patio Control"
  ip_address: 10.10.4.61
  sensor_prefix: esp
  
esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  manual_ip:
    static_ip: ${ip_address}
    gateway: 10.10.4.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  # reboot_timeout: 0s

web_server:
  port: 80

ota:
  safe_mode: True
switch:
  - platform: gpio
    pin: 4
    name: ${relay1_name}
    id: relay1
    restore_mode: RESTORE_DEFAULT_ON
  - platform: gpio
    pin: 5
    name: ${relay2_name}
    id: relay2
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: restart
    name: ${upper_devicename} Restart"
    
binary_sensor:
  - platform: gpio
    pin:
      number: 12
    name: "kitchen sw1"
    on_state:
      then:
        - logger.log: "SW1 single clicked, toggling kitchen Fan"
        - switch.toggle: relay1

  - platform: gpio
    pin:
      number: 14
    name: "kitchen sw2"
    on_multi_click:
    - timing:
        - ON for 1s to 4s
        - OFF for at least 0.3s
      then:
        - logger.log: "SW2 long click"
    - timing:
        - OFF for 1s to 4s
        - ON for at least 0.3s
      then:
        - logger.log: "SW2 long click"
    - timing:
        - OFF for at most 0.4s
        - ON for at most 0.4s
        - OFF for at most 0.4s
        - ON for at least 0.2s
      then:
        - logger.log: "SW2 double clicked"
        - switch.toggle: relay2
    - timing:
        - ON for at most 0.4s
        - OFF for at most 0.4s
        - ON for at most 0.4s
        - OFF for at least 0.2s
      then:
        - logger.log: "SW2 double clicked"
        - switch.toggle: relay2
    - timing:
      - ON for at least 0.5s
      then:
        - script.execute: patio_light
    - timing:
      - OFF for at least 0.5s
      then:
        - script.execute: patio_light

  - platform: status
    name: ${sensor_prefix} ${upper_devicename} Status
    id: api_available

sensor:
  - platform: uptime
    name: ${upper_devicename} Uptime
  - platform: wifi_signal
    name: ${upper_devicename} Signal
    update_interval: 300s

script:
  id: patio_light
  then:
    - logger.log: "SW2 single clicked"
    - if:
        condition:
          - lambda: 'return id(api_available).state;'
        then:
          - logger.log: 'Hass API available'
          - if:
              condition:
                - switch.is_off: relay2
              then:
                - logger.log: "SW2 single click. ${relay2_name} is off, turning back on"
                - switch.turn_on: relay2
                - delay: 30s
                - logger.log: "Turning fish pond off"
                - homeassistant.service:
                    service: switch.turn_off
                    data: 
                      entity_id: switch.fish_pond
              else:
                - logger.log: "SW2 single click. ${relay2_name} is on, executing hass light call service for light.patio"
                - homeassistant.service:
                    service: light.toggle
                    data: 
                      entity_id: light.patio
        else:
          - logger.log: "Hass API not availabe"
          - logger.log: "SW2 single click, toggle relay2"
          - switch.toggle: relay2
```

**Additional context**

Scrolling through logs of discord from last year, someone requested this and @OttoWinter mentioned it was gonna submit a patch to make it work internally, looks like this never happened.

Thanks

subzero79, 2019-05-24:
If there is another way i have not seen in the docs to make it work like I intent please let me know

OttoWinter, 2019-05-24:
The status binary sensor should do exactly that - report the connection state internally. With some quick testing it does appear to work for me too. Please also see the `api.connected` etc conditions (https://next.esphome.io)

subzero79, 2019-05-24:
I added this should be correct

```yaml

  - platform: status
    name: ${sensor_prefix} ${upper_devicename} Status
    id: api_available
    on_state:
      then:
        - if:
            condition:
              lambda: 'return id(api_available).state;'
            then:
              - logger.log: "API connection available"
            else:
              - logger.log: "API connection NOT available"
```

Then the logs when HA stops

```
[08:10:51][C][web_server:126]: Web Server:
[08:10:51][C][web_server:127]:   Address: 10.10.4.61:80
[08:10:51][C][api:101]: API Server:
[08:10:51][C][api:102]:   Address: 10.10.4.61:6053
[08:10:51][C][ota:127]: Over-The-Air Updates:
[08:10:51][C][ota:128]:   Address: 10.10.4.61:8266
[08:10:53][D][api:546]: Client 'Home Assistant 0.92.2 (10.10.0.152)' connected successfully!
[08:11:50][D][switch:031]: 'Patio Control' Turning OFF.
[08:11:50][D][switch:051]: 'Patio Control': Sending state OFF
[08:11:51][D][switch:027]: 'Patio Control' Turning ON.
[08:11:51][D][switch:051]: 'Patio Control': Sending state ON
[08:11:52][D][switch:031]: 'Patio Control' Turning OFF.
[08:11:52][D][switch:051]: 'Patio Control': Sending state OFF
[08:11:53][D][switch:027]: 'Patio Control' Turning ON.
[08:11:53][D][switch:051]: 'Patio Control': Sending state ON
[08:12:47][I][ota:152]: Boot seems successful, resetting boot loop counter.
[08:15:27][D][api:073]: Disconnecting Home Assistant 0.92.2 (10.10.0.152)
```

No further log, also check the with the webserver the status of the sensor

[image]

The api.connected condition is on dev image?

subzero79, 2019-05-25:
According to changelog api.connected should be available in 1.12.2 (which i use) i can see it in the src, but not currently in docs, except for next.esphome.io.

Gave it a try but still same results.

OttoWinter, 2019-05-25:
How are you watching the debug logs? If it's remotely, then that's also using the native API - and also counts as "connected".

You can try to watch the logs over the web server interface - that doesn't use the native API and won't count as connected.

subzero79, 2019-05-25:
Ah ok, yes I am watching the logs through the dashboard. I’ll disconnect the log viewer then to the test.

subzero79, 2019-05-25:
Ok, seems to be working now not using the dashboard, and as expected using the api.connected condition. The status sensor keeps in "ON" state viewing it in the web server, no matter if ha is down and not opening the logs through the dashboard.
Point apart have you considered a component to push the logs to a remote log server (rsyslog)?

OttoWinter, 2019-05-25:
What ESPHome version is this? dev?

subzero79, 2019-05-25:
This is 1.12.2. I attempted yesterday to use dev but i had compilation errors.

OttoWinter, 2019-05-27:
I just checked again and it does work for me in dev - so due to me not being able to replicate this I can't work on this.

subzero79, 2019-05-27:
The api.connected and wifi.connected conditions seem to work for me now. I didn't know them, because there are not in the 1.12.2 docs yet, don't know why. They are in the changelog however.

Guess we can close it

closed 2019-05-27 by OttoWinter