esphome/feature-requests #244: Pulse interval counter

savikko, 2019-05-24:
**Describe the problem you have/What new integration you would like**

I have photoresistor counting blinks on energy meter. Update interval is 60 seconds. LED blinks 1000 imp/kWh.
I would like to count intervals between the blinks and get more accurate power meter using the photoresistor.

**Please describe your use case for this integration and alternatives you've tried:**

Using `pulse_counter` I can get resolution of 60W. `pulse_width` ( #149 ) seems to calculate width of the _pulse_ not interval of the pulses.

**Additional context**

If we have energy meter blinking 1000imp/kWh then if current power is 1000Watts interval is 3.6 seconds. If we take 3 consecutive blinks (lets say [blink] [2.5s] [blink] [3.8s] [blink] [3.0s]) we could calculate (rolling) average of the intervals (3.1s) and then convert it to Watts (60/3.1*60=1161W). Therefore we could get pretty accurate reading faster than with "usual" counter.

OttoWinter, 2019-05-24:
Assuming the pulses are very short (or of constant width), just use pulse_width and use the pin schema to invert the input. Done.

savikko, 2019-05-24:
@OttoWinter : Ah, of course. Thanks. Will try.

savikko, 2019-05-24:
I actually don't know what I am doing wrong:
```sensor:
  - platform: pulse_counter
    internal_filter: 10us
    pin: GPIO2
    unit_of_measurement: 'W'
    name: 'Powermeter'
    accuracy_decimals: 0
    update_interval: 10s
    filters:
      - multiply: 60
```
Outputs:

```[00:28:26][C][pulse_counter:147]: Pulse Counter 'Powermeter'
[00:28:26][C][pulse_counter:147]:   Unit of Measurement: 'W'
[00:28:26](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[00:28:26][C][pulse_counter:147]:   Icon: 'mdi:pulse'
[00:28:26][C][pulse_counter:148]:   Pin: GPIO2 (Mode: INPUT)
[00:28:26][C][pulse_counter:149]:   Rising Edge: INCREMENT
[00:28:26][C][pulse_counter:150]:   Falling Edge: DISABLE
[00:28:26][C][pulse_counter:151]:   Filtering pulses shorter than 10 µs
[00:28:29][D][api:572]: Client 'Home Assistant 0.91.3 (192.168.20.40)' connected successfully!
[00:28:35][D][pulse_counter:158]: 'Powermeter': Retrieved counter: 42.00 pulses/min
[00:28:36][D][sensor:092]: 'Powermeter': Sending state 2520.00000 W with 0 decimals of accuracy
[00:28:45][D][pulse_counter:158]: 'Powermeter': Retrieved counter: 36.00 pulses/min
[00:28:46][D][sensor:092]: 'Powermeter': Sending state 2160.00000 W with 0 decimals of accuracy
```

So seems to be right.

But when I use pulse_width:
```sensor:
  - platform: pulse_width
    pin: 
      number: GPIO2
      inverted: True
    unit_of_measurement: 's'
    name: 'Powermeter interval'
    accuracy_decimals: 3
    update_interval: 1s
```

I get:

```[00:31:05][C][pulse_width:020]: Pulse Width 'Powermeter interval'
[00:31:05][C][pulse_width:020]:   Unit of Measurement: 's'
[00:31:05](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[00:31:05][C][pulse_width:020]:   Icon: 'mdi:timer'
[00:31:05][C][pulse_width:021]:   Update Interval: 1.0s
[00:31:05][C][pulse_width:022]:   Pin: GPIO2 (Mode: INPUT_PULLUP, INVERTED)
[00:31:06][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:06][D][sensor:092]: 'Powermeter interval': Sending state 0.00113 s with 3 decimals of accuracy
[00:31:07][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:07][D][sensor:092]: 'Powermeter interval': Sending state 0.00108 s with 3 decimals of accuracy
[00:31:08][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:08][D][sensor:092]: 'Powermeter interval': Sending state 0.00108 s with 3 decimals of accuracy
[00:31:09][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:09][D][sensor:092]: 'Powermeter interval': Sending state 0.00108 s with 3 decimals of accuracy
[00:31:10][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:10][D][sensor:092]: 'Powermeter interval': Sending state 0.00110 s with 3 decimals of accuracy
[00:31:11][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:11][D][sensor:092]: 'Powermeter interval': Sending state 0.00110 s with 3 decimals of accuracy
[00:31:12][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:12][D][sensor:092]: 'Powermeter interval': Sending state 0.00110 s with 3 decimals of accuracy
[00:31:12][D][api:572]: Client 'Home Assistant 0.91.3 (192.168.20.40)' connected successfully!
[00:31:13][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:13][D][sensor:092]: 'Powermeter interval': Sending state 0.00126 s with 3 decimals of accuracy
[00:31:14][C][pulse_width:026]: 'Powermeter interval' - Got pulse width 0.001 s
[00:31:14][D][sensor:092]: 'Powermeter interval': Sending state 0.00126 s with 3 decimals of accuracy
```

So it seems to be little bit off, seconds should be larger than 0.001.

If you have any clues what could be going on here, I'll be more than grateful to hear that.

OttoWinter, 2019-05-25:
@savikko Depends on the hardware you have. Some photodiodes work with inverted signal, so HIGH means "no light" and LOW mean "there is light". If that is the case you would just need to remove the inverted part again.

savikko, 2019-05-25:
Actually I tried it but seems that Inverted -boolean has no effect.

savikko, 2019-05-25:
And, I came to this conclusion when testing with binary sensor. State was longer ON first and applying inverter: true OFF was longer.

savikko, 2019-05-25:
Just to clarify, when using this configuration:
```binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO2
      inverted: False
      mode: INPUT
    name: "Non-inverted"
```

Output is:
```
[20:14:24][D][binary_sensor:033]: 'Non-inverted': Sending state ON
[20:14:45][D][binary_sensor:033]: 'Non-inverted': Sending state OFF
[20:14:45][D][binary_sensor:033]: 'Non-inverted': Sending state ON
[20:15:32][D][binary_sensor:033]: 'Non-inverted': Sending state OFF
[20:15:32][D][binary_sensor:033]: 'Non-inverted': Sending state ON
[20:15:40][D][binary_sensor:033]: 'Non-inverted': Sending state OFF
[20:15:40][D][binary_sensor:033]: 'Non-inverted': Sending state ON
[20:15:46][D][binary_sensor:033]: 'Non-inverted': Sending state OFF
[20:15:46][D][binary_sensor:033]: 'Non-inverted': Sending state ON
```
Meaning that ON is dominant there.

And when using this setup:
```binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO2
      inverted: True
      mode: INPUT
    name: "Inverted"
```

Then the output is: 
```[20:23:51][D][binary_sensor:033]: 'Inverted': Sending state OFF
[20:23:56][D][binary_sensor:033]: 'Inverted': Sending state ON
[20:23:56][D][binary_sensor:033]: 'Inverted': Sending state OFF
[20:24:33][D][binary_sensor:033]: 'Inverted': Sending state ON
[20:24:33][D][binary_sensor:033]: 'Inverted': Sending state OFF
[20:24:39][D][binary_sensor:033]: 'Inverted': Sending state ON
[20:24:39][D][binary_sensor:033]: 'Inverted': Sending state OFF
[20:26:45][D][binary_sensor:033]: 'Inverted': Sending state ON
[20:26:45][D][binary_sensor:033]: 'Inverted': Sending state OFF
[20:28:38][D][binary_sensor:033]: 'Inverted': Sending state ON
[20:28:39][D][binary_sensor:033]: 'Inverted': Sending state OFF
```

And OFF is dominating now.

When testing with `pulse_width` it seems that there is no effect when trying both `inverted: False` and `inverted: True`.

OttoWinter, 2019-05-27:
> When testing with pulse_width it seems that there is no effect when trying both inverted: False and inverted: True.

That is interesting, it should make a difference - from some quick testing it looks like it does work correctly for me at least.

Do you have some way of inspecting the signal further with an oscilloscope or something like that? It could be that there are some short additional pulses generated by the diode when it is switching off, that would explain the behavior.

If that's the case, an interval counter would also not really help (since it would measure the same thing).

I also don't think pulse interval counter is different from pulse width - if the off pulse has a fixed length or fixed percentage of the width, you can easily calculate it back. And if that assumption is not true, the pulse interval is probably not helpful anyway.

I'm going to close this issue as I see pulse interval as unnecessary, but further discussion about why your setup is failing can continue here.

closed 2019-05-27 by OttoWinter