esphome/feature-requests #246: Request for integration Lessar HVAC

erudebu, 2019-05-27:
**Describe the problem you have/What new integration you would like**

Please add 2 parameters to custom protocol of remote_transmitter device, or add new integration of the Lessar HVAC

**Please describe your use case for this integration and alternatives you've tried:**

Please add 2 parameters to the custom protocol component of <b>remote_transmitter</b> device for the ability to transfer data like "L 10110010 01001101 1011111 1111101 01110000 00001110 SL 10110010 01001101 1011111 1111101 01110000 00001110 T", (where "L" is the initialization code, "S" is the separating code ". T "- terminating code) for the operation of an infrared transmitter to control the air conditioner (HVAC) of the manufacturer's company "LESSAR"
sensor:

**Additional context**

A document from the manufacturer is available with a full description of the infrared remote control protocol (document in Chinese)
[LZ-KBP_description of the remote protocol.docx](https://github.com/esphome/feature-requests/files/3223283/LZ-KBP_description.of.the.remote.protocol.docx)

OttoWinter, 2019-05-27:
Technically this is already possible, see raw remote codes.

I'm going to merge this into https://github.com/esphome/feature-requests/issues/38 and https://github.com/esphome/feature-requests/issues/24

closed 2019-05-27 by OttoWinter

erudebu, 2019-05-27:
I'm cannot understand, how can I receive a list of data values (not only
one value) on to esphome device from homeassistant for it.

OttoWinter, 2019-05-27:
I'm not sure what you mean - but receiving raw data is also possible, see https://next.esphome.io

erudebu, 2019-05-28:
I'm mean about receiving list of code data from homeassistant api, not from remote_receiver.
(example code list: [-4400, 4400, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 540, -540, 540, -540, 1620, -540, 540, -540, 540, -540, 1620, -540, 540, -540, 540, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 5220, -4400, 4400, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 540, -540, 540, -540, 1620, -540, 540, -540, 540, -540, 1620, -540, 540, -540, 540, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 540, -540, 1620, -540, 1620, -540, 1620, -540])

OttoWinter, 2019-05-28:
That is also possible via the raw transmit action, see the link I posted before