esphome/feature-requests #250: Support partial update on more Waveshare EPaper sizes

TheDorje, 2019-05-28:
**Describe the problem you have/What new integration you would like**
Looking here: [https://esphome.io/components/display/waveshare_epaper.html](url), I can see that partial refresh is only possible on 1.54in, 2.13in and 2.90in. 
I just got a 7.5 3 black/red/white screen. I know three colors are not yet possible (there's a request for that already here), but the partial refresh would be awesome to get before the color thing is solved. 

**Please describe your use case for this integration and alternatives you've tried:**
Simplest use case is a wall clock with the screen. Considering the full update blinks the whole screen over something like 16s, it would take away from the coolness factor of such a clock. Ideally, I'd like to run the full update every 30 minutes, or maybe even once per hour, on the hour. I didn't dig too deep into the code yet, but I think I saw DEEP_SLEEP supported, just so the screen is protected (as Waveshare insists on happening). 

**Additional context**
Nothing to add, other than hope this is an easy fix or that I can somehow help and contribute to testing to make it happen :)

OttoWinter, 2019-05-28:
The code for that is entirely based on this library: https://github.com/soonuse/epd-library-arduino

The way the update parameters (and thus partial update parameters) are passed in is with an array with some magic values - what format they are in is difficult to reverse-engineer, and testing with custom ones is dangerous as that can quickly damage the screen when wrong ones are chosen. So unless the manufacturer provides a list of these values for partial updates too, there's little hope of getting that supported.

TheDorje, 2019-05-30:
Right. I'll should have some more time during the weekend to dig through the code WaveShare provide with their samples. It seems like their lib does support that. 

Will update. 

Thanks ;)

TheDorje, 2019-06-04:
I dug through the docs and the code and yeah. These don't seem to support partial refresh or the manufacturer doesn't want to share how it's done. Shame. Closing the issue.

closed 2019-06-04 by TheDorje