esphome/feature-requests #251: Support for e-ink Xiaomi BLE Thermometer LYWSD02 Model

sermayoral, 2019-05-29:
**Describe the problem you have/What new integration you would like**

Actually ESPHome does not support Xiaomi BLE Thermometer LYWSD02 Model

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

There is a custom_components that works for this model: https://github.com/h4/LYWSD02-home-assistant. It would be great if there is official support for this model

OttoWinter, 2019-05-29:
Please enable VERY_VERBOSE mode in the logger and print out some of the debug logs you see when an LYWSD02 is in range.

intermittech, 2019-05-29:
Oh, I'll join in on this one too, I have both of the e-ink versions (bought them to test).

intermittech, 2019-05-29:
Ok, I've generated some logging with a MH-ET-Live ESP32. I have attached the file.

This was in my garage testing room where one "normal" Maija was present and both of the new e-ink models. My phone was also there so that might have spewed some bluetooth too.
[xiaomibluetooth_2019-05-29.txt](https://github.com/esphome/feature-requests/files/3234409/xiaomibluetooth_2019-05-29.txt)

intermittech, 2019-05-29:
Ok, I've generated some logging with a MH-ET-Live ESP32. I have attached the file.

This was in my garage testing room where one "normal" Maija was present and both of the new e-ink models. My phone was also there so that might have spewed some bluetooth too.
[xiaomibluetooth_2019-05-29.txt](https://github.com/esphome/feature-requests/files/3234419/xiaomibluetooth_2019-05-29.txt)

sermayoral, 2019-05-29:
Thanks @intermittech. I hope it can help :-)

Kvasenok, 2019-05-31:
There is also an [article](https://habr.com/ru/post/452558/) in Russian from the developer of the custom component where he explains how he parsed the protocol. Google Translate handles the translation quite well. Might also be helpful.

kastro123, 2019-06-03:
This E-ink sensor looks the same and there is also some Node.js script.
https://community.home-assistant.io/t/xiaomi-mijia-cleargrass-qingping/116819/9

james-fry, 2019-06-05:
Would be great to see this added in esphome :)

Kvasenok, 2019-06-09:
Also, this BLE sensor got integrated into OpenMQTTGateway, so I guess there is no problem with obtaining and decoding the data from the sensor.

xenio, 2019-06-16:
I have this clock too. I was using it with OpenMQTTGateway but I am now on EspHome. I hope it will be possible to support.

h4, 2019-06-21:
Hi. I'm going to continue work with integration. Some code was moved into separate repo an published on PyPI: https://github.com/h4/lywsd02

alexeyzel, 2019-06-26:
I am considering buying a Xiaomi Bluetooth Thermometer. Is there a chance that the new version (Xiaomi ClearGrass Bluetooth Thermometer) will be supported soon? Or should I get the old one?

G1K, 2019-06-30:
@OttoWinter , OMG implemented it
https://github.com/1technophile/OpenMQTTGateway/commit/4ba27481b5b8710c0a075329768ee3b72a52ed78

https://1technophile.blogspot.com/2019/06/integration-of-mijia-clock-temperature.html

intermittech, 2019-06-30:
> @OttoWinter , OMG implemented it
> [1technophile/OpenMQTTGateway@4ba2748](https://github.com/1technophile/OpenMQTTGateway/commit/4ba27481b5b8710c0a075329768ee3b72a52ed78)
> 
> https://1technophile.blogspot.com/2019/06/integration-of-mijia-clock-temperature.html

Lol, your message was quite confusing, I thought you stated it was implemented so I started searching for it but couldn't find it. But OMG stands for "Open MQTT Gateway". DOH!

sermayoral, 2019-09-09:
I've closed this issue because it has been merged in [#664](https://github.com/esphome/esphome/pull/664)

closed 2019-09-09 by sermayoral