esphome/feature-requests #255: Invert Logic for softwareserial.

swampyclark, 2019-05-30:
The base arduino swserial has the invert logic.  

class SoftwareSerial : public Stream {
public:
  SoftwareSerial(int receivePin, int transmitPin, bool inverse_logic = false, unsigned int buffSize = 64);
  virtual ~SoftwareSerial();

**Describe the problem you have/What new integration you would like**

add the inverted flag to the constructor and use it to flip flop as needed during the methods. 
**Please describe your use case for this integration and alternatives you've tried:**

It would be nice addon if that option existed in esphome.  All the honeywell alarm panels work in inverse.  can't read and write from the alarm panel without using it this way.
**Additional context**

tbrasser, 2020-01-17:
This should also work with HardwareSerial on ESP32 and also on ESP8266 on UART0.

svedaker, 2021-02-08:
This could also be useful in this project reading the p1-port on powermeters.
https://github.com/psvanstrom/esphome-p1reader

endor-force, 2021-11-22:
Yes, another reference is this power meter hw built by Guy:
http://weigu.lu/microcontroller/smartyReader_P1/index.html

I am looking in to setting up his HW with esphome but i am missing the invert option on the serial config as of now. :)

https://github.com/weigu1/SmartyReader/blob/main/Arduino/SmartyReader_P1_WiFi/SmartyReader_v1_5/SmartyReader_v1_5.ino
ESP8266 - See line 347 for reference.

ESP32 invert option seems like it was added from this issue: https://github.com/esphome/feature-requests/issues/1145

endor-force, 2021-11-29:
Seems like there is an ongoing Pull request for fixing this feature:
https://github.com/esphome/esphome/pull/1727

endor-force, 2021-12-13:
This now in beta for ESP8266, i have tried it out with smartyreader on wemos d1 mini pro with inverted rx.
Works great.