esphome/feature-requests #256: Support "--version"

Joshfindit, 2019-05-30:
**Describe the problem you have/What new integration you would like**

Just looking for an easy way to check which version I'm currently running while in the shell

**Please describe your use case for this integration and alternatives you've tried:**

"Standard" methods are typically:
- `--version`
- `-v` (Not asking for this as `-v` is already taken)
- Outputting the version number as part of the `usage:` message

Personally, I'll take any or whichever alternate method you decide is best

**Additional context**

Cheers

OttoWinter, 2019-05-31:
Already exists: `esphome dummy.yaml version` (dummy is required because the arg parser has that syntax). `-v` is already used for verbose ESPHome mode.

closed 2019-05-31 by OttoWinter

Joshfindit, 2019-05-31:
That syntax seems to me intuitively to be asking for the version of the YAML file as opposed to esphome itself.

But it works, thank you