esphome/feature-requests #257: NTC self-heating

escoand, 2019-05-31:
Thanks for the new NTC sensor. This allows me to throw some filter juggling away. But I have a question to the current implementation.

**Describe the problem you have/What new integration you would like**
How is self-heating of the NTC avoided? My solution was until now to use VCC from a GPIO which is turned on before the reading and off afterwards. In the documentation there is no hint that this is done in the code as well.
Your linked product has a description of this problem: https://learn.adafruit.com/thermistor/using-a-thermistor#self-heating-3-22

**Please describe your use case for this integration and alternatives you've tried:**
My current alternative:
```yaml
  - platform: adc
    pin: A0
    id: temperature
    name: Temperatur
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    update_interval: 30min
    icon: mdi:thermometer
    filters:
      - lambda: !lambda >-
          static int isMeasuring = 0;
          static double uMax = 1.0;
          static double tStatic = 25 + 273.15;
          static double beta = 4090;
          static double rStatic = 10000;
          if(!isMeasuring) {
            isMeasuring = 1;
            id(thermistor_up).turn_on();
            id(temperature).update();
            id(thermistor_up).turn_off();
            isMeasuring = 0;
            return {};
          } else {
            double rThermistor = rStatic * (uMax / x - 1);
            double tKelvin = 1.0 / ( (1.0/tStatic) + (1.0/beta) * log(rThermistor/rStatic) );
            return tKelvin - 273.15;
          }
output:
  - platform: gpio
    pin: D2
    id: thermistor_up
```

**Additional context**

OttoWinter, 2019-05-31:
Simple:

```yaml
sensor:
- platform: adc
  update_interval: never
  id: adc_sens
  pin: A0
  # ...

switch:
- platform: gpio
  pin: D0
  id: adc_vcc

interval:
- interval: 60s
  then:
    - switch.turn_on: adc_vcc
    - component.update: adc_sens
    - switch.turn_off: adc_vcc
```

(Feel free to update the NTC docs about this, I'll close this feature request as already possible).

closed 2019-05-31 by OttoWinter

escoand, 2019-05-31:
Thanks for the fast answer. If it's possible was not my indented question. I wondered if something like this is already implemented in the NTC sensor code. I would say without this switch on and off the sensor will report inaccurate values in any case. So for me this is mandatory.

OttoWinter, 2019-05-31:
I'm not sure I understand the question.

It's not implemented in the NTC sensor code because the NTC code doesn't need to know about it. The requested behavior can be done in a few lines of yaml (and the docs can be updated for that)