esphome/feature-requests #259: Support for Code Mercenaries LED-Warrior

lblokland, 2019-05-31:
**Describe the problem you have/What new integration you would like**

I'm trying to create a setup with an ESP2866 board to control my DALI lighting system, which is currently controlled by a 'dumb' controller. I set the timers so the lighting is fading in at a certain time, and later fade out to simulate longer daylight in the stables.
I've purchased some boards from the Code Mercenaries (LED-Warrior) to connect the Dali interface to I2C (as a DALI master)
I'm not much as a programmer, but with the available documentation with all the I2C addresses it should be possible.
The lighting 'm using are fluorecent bars already fixed to the ceiling and the current contrller,, but for testing I also purchaed a LED Warrior board that is a DALI slave and able to control a LED as a driver.
So, the short version is to create a DALI interface for ESP Home.
**Please describe your use case for this integration and alternatives you've tried:**

The alternative is the 'dumb' controller with manual timers to start the fading sequences. I can try to use an 4CH Sonoff pro so alter the inputs of the dumb controller, so I can automate the timing but it would be much sweeter to control the DALI interface directly as it is able to report back info regarding the lighting and seperately controle the bars.
**Additional context**

Some URLS to the products:
https://www.codemercs.com/en/dali-interface
https://shop.codemercs.com/en/led-warrior09-02-module.html
https://www.codemercs.com/downloads/ledwarrior/lw_arduino_sdk.zip

brandond, 2019-05-31:
Looks like the i2c interface for this board is documented here:
https://www.codemercs.com/downloads/ledwarrior/LW14_Datasheet.pdf

They defer to the actual DALI spec for commands sent to the bus:
https://www.energy.ca.gov/2005publications/CEC-500-2005-141/CEC-500-2005-141-A23.PDF

lblokland, 2019-06-01:
Hey Brandon,

that's right...but how to implement it as a device, or light component...that's the question (for me it is).

Juice10, 2022-07-24:
I'm in the same boat, trying to get DALI working with esphome.
In my case the hardware is slightly different.
I'm using the MikroE Click DALI2 module: https://www.mikroe.com/dali-2-click

Tried to get it working with [the code found here](https://create.arduino.cc/projecthub/NabiyevTR/simple-dali-controller-506e44) but no luck as of yet. Between the ESP8266 being different than what they are using and the DALI module I'm using being different and the fact that I'm more of a software guy than a hardware guy I haven't been able to crack it yet.

OdinReiter, 2023-01-30:
hi guys,

i've also have some luminaires with a DALI LED-Driver in my SmartHome and i've build my own really simple interface (an optocoupler with an aditional transistor).
And i've got my implementation up and running in esphome.

here's an github link to my repo:
https://github.com/OdinReiter/ESPHome_DALI-Interface

i didnt have the time for the dokumentation but if someone need that, i will try to update that.

varbeti, 2023-02-11:
Hello @OdinReiter. I've seen your repo but I don't know what to do with it (sorry, I'm really new with ESPHome). Could you prepare same documentation, please? Thanks!

OdinReiter, 2023-02-11:
Hello @varbeti,
yeah my repo is a bit empty and doesnt have much information. sorry but i had no time to fill it up.
im also new withesphome (like 2-3 month) and totally new in github.

if you are using the i2c DALI controller (like the LED-Warrior board) my component wont work.
i've created my own bus coupling unit which can send commands via an gpio pin.
and my components can send just brightness commands to one adress  (or broadcast)

if you concretize your questions, i will try to help you!
and i will try to document my github page a bit more in the near future

sanderloones, 2023-04-22:
I would like to use your repository. 
I want to use your project for wall switches with up to 6 buttons. Can this work?

Also, I'm a bit lost on the hardware side. 
I cant find B140C1500 on Aliexpress, is there an alternative?
Could you also include a picture of how you soldered it?

OdinReiter, 2023-04-23:
Hi @sanderloones 

sadly my component and this easy bus controller wont read any values from DALI input devices.  
So it wont work with wall switches. And input devices are a bit tricky because many manufacturer have some features which are not supported in the DALI standard. 
maybe you will find a better solution for that.

Of course there are alternatives for this full bridge rectifier. Just take one with a nominal voltage above 25V and a nominal current above your DALI power supply.
Here are some examples:
- https://de.aliexpress.com/item/1005004729127520.html
- https://de.aliexpress.com/item/32923248227.html

i can upload a picture of my pcb from an old prototyp on my github page, when i found it.

sanderloones, 2023-04-23:
The idea was to use the native input from esphome and use this directly to Dali led drivers.

Would this work?

OdinReiter, 2023-04-23:
ok i misunderstood your first message. 
If you want to get the input from the GPIO pins and send some brightness value via DALI to the LED Driver than this component can work for you.

found my first prototyp for that project. It has also a reading part but that isnt implemented at the moment.
i've attached a picture of it in this message (hopefully you can receive it)

[image]

sanderloones, 2023-05-23:
Thanks for all your work!
I would also like to get the status of my lights in Home Assistant. How
would I do that?

OdinReiter, 2023-05-25:
Hi @sanderloones,
you're welcome. :)

so if you using lights in esphome it will saving the current state of your lamp.
But if you want to read your DALI device via the bus, because you have some DALI swiches in your setup, this wont be possible atm. And this wont be easy to realize with the collision detection, reading of the backward frames, other types of DALI interfaces and so on...

im using just this module as an DALI input device to control all my lamps so i wont need to read the Bus.
can you give me a rough overview of your DALI Bus / your DALI devices?

sanderloones, 2023-05-25:
Hi

I'm building a new home. So I have carte blanche to make it, in my eyes
"perfect".
I want a smart lighting system that is redundant. I found Dali interesting
because it is multimaster.

So the idea is that I would have an esphome per switch-point. It would
control RGB-diodes on the switch(es) that specify which lights are
on (wifi).
It would also control sensors and control the lights through the bus(Dali?).

If my wifi and HA breaks, my lights will continue the work because they
work through the bus.

In HA I would want to be able to control those esphome switch-points too.

Could this help?
https://www.adfweb.com/Home/products/DALI_MQTT.asp?frompg=nav28_4

OdinReiter, 2023-06-03:
Hi,

sorry for my late reply.
your idea is not bad. So if you already have a DALI Light and some DALI switches for controlling this bus, then its already independent from HA.
But my lib for esphome dont support this yet. And i dont know, if im able to implement this, cause i'm working on other projects right now. 
Maybe you can implement this? :)

i dont know, if you will be able to control and read you luminaires with this Module. 
But if you can get DALI to MQTT, you could use nodered to set your lights in HA to the current brightness value...