esphome/feature-requests #260: FastLED color temperature / WWA strip support

SebDominguez, 2019-06-01:
**Describe the problem you have/What new integration you would like**

-  Be able to set color temperature instead of RGB values for RGB LED chip

-  Be able to control WWA LED chip (white+ warm white + amber) in the same way the cwww light platform do to control SK6812 chip for instance (basically the same as ws2812) that have the 3 white channel instead of 3 color channel.

**Please describe your use case for this integration and alternatives you've tried:**

-  To control the RGB LED strip with temperatures values I'm using a conversion table + node-red.

-  For the SK6812 I'm using it as a regular RGB LED strip. Just mapping the RGB value to the corresponding temperature :

```
color1800k = {cw: 0, ww: 0, a: 255}
color3000k = {cw: 0, ww: 255, a: 0}
color7000k = {cw: 255, ww: 0, a: 0}
```
_cw = cold white, ww = warm white, a = amber_

**Additional context**

My final goal with those SK6812 is to be able to simulate indoor sunlight.

I'm actually using node-red + Arduino sketch + MQTT to achieve that, but it's really messy and over complicated.

[image]
_SK6812 Timelaps_

I'm checking the actual date / location with a random API on internet to get the exact time of sunrise, sunset, solarnoon, twilight begin / end, then I'm fixing some arbitrary temperatures for those events, and I'm just sending every minute via MQTT a different light temperature, with a nice and soft transition during the day.

But I'm looking for a better way to implement that since again my actual setup is really messy and hard to deploy / maintain.

[image]

maybe in a future release we can have a light platform that automatically light up with an automatically calculated temperature according to a set location and timezone, that would be really great, but until then, being able to just set temperature on RGB LED and control WWA LED would be really helpful.

vmazmaz, 2020-01-24:
I would also like the possibility to address the WWA strip simply by a color temperature.

Regarding :

> light platform that automatically light up with an automatically calculated temperature according to a set location and timezone

If you use home assistant you could look into either [Flux](https://www.home-assistant.io/integrations/flux/) or [Circadian Lighting](https://community.home-assistant.io/t/circadian-lighting-custom-component/61246).

Both can either handle RGB strips or Color Temperature directly. So I feel the integration with WWA strips is very near.

yschroeder, 2020-04-28:
I would also like to see this functionality! A WWA mode for the FastLED module so that the Flux component can control the color temperature of the LED strip.

johanneswilkens, 2021-12-10:
Would love to see this in esphome, have some of this strip laying around and don't know how to properly control it in HA

ashp8i, 2022-01-06:
The strips can be controlled but the colors will be wrong, you have to specify them as regular RGB sk6812 and substitute colors. If these strips are supported we can have the colors mapped correctly. Please support this strip type.

markheloking, 2023-10-09:
I would also like to see support for WWA strips in FastLED/ESPHome so we can use these as wakeup lights in HomeAssistant easily. Unfortunately it seems the [pull request](https://github.com/esphome/esphome/pull/1500) where this was initially worked on by [mhentschke](https://github.com/mhentschke) has died a slow death in "Needs Review" status, straight into closure due to inactivity :(

johanneswilkens, 2023-10-11:
I now had a few projects requiring this feature, so I decided to hack something to make it kinda work. Maybe it's useful for other people as well. Basically I made a hidden neopixel light, which is controlled by an exposed dummy cct light using template outputs and global variables. 
The color temperature can also be seen as a hue value, brightness is just brightness, and saturation should always be max. So actually the color temperature light gives HSB values which can be converted to RGB to control the neopixel/fastled/RMT or whatever.
[For those who are interested, here is the gist](https://gist.github.com/johanneswilkens/91ac8ba489d8ce42112750e51b07ea29)
It's probably not the most efficient way, but after a few tries it works perfectly now.

ashp8i, 2023-10-11:
Thanks for the gist I can make out what it’s doing and seems a very intuitive low code solution using hidden/internal lights. I used a similar technique with https://coolcomponents.co.uk/products/lumini-led-ring-1-inch-20-x-apa102-2020-com-14967<https://coolcomponents.co.uk/products/lumini-led-ring-1-inch-20-x-apa102-2020-com-14967?currency=GBP&variant=13910804791357&utm_medium=cpc&utm_source=google&utm_campaign=Google%20Shopping&stkn=ab4ba3a5261a&gclid=Cj0KCQjwj5mpBhDJARIsAOVjBdrzaMn2yfRuqilJLGM1QR4NpjawCbaiwRpGxSXgeMk6UuLeN_tkvAYaAshEEALw_wcB>

Lumini apa102 ring that was too bright, had to be limited using hardcoding in esphome and used a mosfet to totally cut off power to led strip using hidden lights, that can be used to fix the pixel order if you did not quite get it right.

At least we can use the strip normally in home assistant as esphome is letting home assistant know what the colour temperature extremes are.

ElektronikCZ, 2024-02-27:
Im also interrested in this feature. Any progress on this? Its been open for 4 years.