esphome/feature-requests #261: Trigger BLE Scan on Demand

OkhammahkO, 2019-06-01:
**Describe the problem you have/What new integration you would like**

- Add component.update support for esp32_ble_tracker sensors.

- I want to be able to trigger a scan on demand as an alternative to setting short scan intervals which kills my 2.4GHZ.

I guess this would look like:
```
binary_sensor:
  - platform: ble_presence
    mac_address: C1:DE:8E:XX:YY:58
    id: mibandpresence_1
    name: "Presence MiBand EntryESP"

script:
  - id: update_ble
    then:
      - component.update: mibandpresence_1
      - lambda: 'id(mibandpresence_1).update();'
```

**Please describe your use case for this integration and alternatives you've tried:**

- I would use a xiaomi motion detector positioned outside my apartment and pointing at the elevator door to first detect motion (they do this very fast), and then trigger an "arrival scan" for my BLE devices to confirm it is me approaching the door. Then my ESP32 would unlock my smart lock (already working).

- I tried component.update but I think it failed on compile, so I guess it's not supported?

**Additional context**
ESPHome is awesome. Thanks.

Home assistant Q:
https://community.home-assistant.io/t/esphome-ble-scanner-arrival-presence-detection-trigger-a-scan-on-demand/114911

OttoWinter, 2019-06-02:
> I want to be able to trigger a scan on demand as an alternative to setting short scan intervals which kills my 2.4GHZ.

Have you already tried this or is this just an assumption? ESPHome's BLE scanning is a semi-active BLE scan - but for most purposes it's 100% passive. Only if a device already pushes out an ADV frame ESPHome will sometimes ask for more data via scan req packets.

Plus BLE scanning takes place on the channels farthest away from the WiFi spectrum, so I don't think BLE scanning can influence other device's WiFi connectivity much.

OkhammahkO, 2019-06-03:
I tried 5 sec scans, and it seemed to really slow things down (sometimes halt it).  But it's hard for me to be 100% as I had a lot of other things going on too, so I might have mis-diagnosed. I admit I'm not that knowledgeable on the details.  I guess I can give it another go. I live in a 2.4ghz saturated apartment block...

My diagnoses was based on reading a few threads, especially this awesome one, and this post.
https://community.home-assistant.io/t/monitor-reliable-multi-user-distributed-bluetooth-occupancy-presence-detection/68505/597?u=mahko_mahko

Also, lots of people have reported similar issues with the Home Assistant BLE scanner.

I still think there is merit in my request? Home assistant has this service for traditional Bluetooth.
https://community.home-assistant.io/t/how-to-use-the-device-tracker-bluetooth-tracker-update-service/114927/5?u=mahko_mahko

Thank you.

closed 2022-05-10 by OkhammahkO