esphome/feature-requests #268: m5stick-c

WhistleMaster, 2019-06-03:
**Describe the problem you have/What new integration you would like**

I would like to use ESPHome with the m5stick-c, but I don't see which board should be used.

```
platform: ESP32
board: m5stick-c
```

But I get the following error:

  `Could not find board 'm5stick-c'. Valid boards are alksesp32, bpi-bit, d-duino-32, esp-wrover-kit, esp32-evb, esp32-gateway, esp32-poe, esp320, esp32dev, esp32doit-devkit-v1, esp32thing, esp32vn-iot-uno, espea32, espectro32, espino32, featheresp32, firebeetle32, fm-devkit, heltec_wifi_kit_32, heltec_wifi_lora_32, hornbill32dev, hornbill32minima, intorobot, lolin32, lolin_d32, lolin_d32_pro, lopy, lopy4, m5stack-core-esp32, m5stack-fire, mhetesp32devkit, mhetesp32minikit, microduino-core-esp32, nano32, nina_w10, node32s, nodemcu-32s, odroid_esp32, onehorse32dev, oroca_edubot, pico32, pocket_32, quantum, ttgo-lora32-v1, ttgo-t-beam, turta_iot_node, wemosbat, wesp32, widora-air, xinabox_cw02.`
  

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-06-03:
See what the wizard writes - If unsure use the generic option (`esp-wrover-kit`)

closed 2019-06-03 by OttoWinter

WhistleMaster, 2019-06-03:
Thanks, got it working with:

```
platform: ESP32
board: esp-wrover-kit
```