esphome/feature-requests #269: PMSX0003 Update interval

jblb, 2019-06-03:
Message from https://community.home-assistant.io/t/pmsx0003-update-interval/112325

**Describe the problem you have/What new integration you would like**

I bought a sensor PMS7003, used config from: https://esphome.io/components/sensor/pmsx003.html?highlight=particulate 7 and it is working perfectly but I would like to extend a lifetime of this sensor by extend interval time. Now it is a few seconds. I just want to use it as: https://esphome.io/components/sensor/sds011.html 4 there is a update interval option. Is it possible to add soon?

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-06-03:
The [datasheet](http://www.aqmd.gov/docs/default-source/aq-spec/resources-page/plantower-pms5003-manual_v2-3.pdf) doesn't give any warning about lifetime with the default setup. I think the update_interval thing is probably just a thing with the SDS011. If you have concrete data that points to the PMS having the same problem then post it here - otherwise I'll assume it's a SDS-only thing.

Second, the PMS adjusts its update interval automatically depending on the number of particles in the air (see page 4), this might have that built-in itself.

Next, the PMS does not have an update interval register like the SDS has. So software can not configure the PMS for a certain update interval. The only thing that can be done is to force the sensor to sleep manually with the SET pin (page 4 again).

For using the SET pin, please see the NTC sensor page (section self-heating) for turning the sensor on only during measurement.

Until someone can show this is really required for this sensor, I will close this feature request.

https://download.kamami.com/p564008-p564008-PMS7003%20series%20data%20manua_English_V2.5.pdf

closed 2019-06-03 by OttoWinter

ozzi91, 2019-06-05:
Hello @OttoWinter 

Look at this: https://www.letscontrolit.com/wiki/index.php/PMSx003 (part lifetime helper). They mentioned "The laser diode inside the PMSx003 has a lifetime of about 8000 h, nearly one year"

OttoWinter, 2019-06-06:
@ozzi91 Thanks for sharing, that does confirm the issue then.

But, ESPHome does already support this SET pin too - see NTC sensor page as an example. The docs for that could then be copied to the PMS page too.

ozzi91, 2019-06-06:
I am writing this just to be sure how make it right for me and other people :)

I have 4 pins on PMS7003
1. VCC
2. GND
3. RX
4. TX

Pin GND, RX, TX connect normally. VCC connect to A0 pin. Am i right?

OttoWinter, 2019-06-06:
@ozzi91 No no, not quite exactly the same as the NTC page - just the general principle of toggling a pin ON/OFF for intervals.

Connections:
 - VCC/GND/RX Same as before. (TX not required)
 - Wire up SET to a free GPIO pin, for example GPIO0

YAML:
- PMS section same as before.
- Add:

```yaml

switch:
  - platform: gpio
    pin: 
      number: GPIO0
	  inverted: yes
    id: pms_set

interval:
  - interval: 60s
    then:
      - switch.turn_on: pms_set
      - delay: 5s
      - switch.turn_off: pms_set
```

ozzi91, 2019-06-06:
@OttoWinter 
Seems I am too stupid to understand "Wire up SET to a free GPIO pin, for example GPIO0"

GPIO0 u mean connect a swtich and when it will be manually triggered it will launch reading sensor data in 60s intervals?

Could u use simpler words? I am not perfect in english :/

nickrout, 2019-06-09:
Read the datasheet, there is a pin "SET" on the device, wire it up to the esp device as Otto says. it does this

```

> PIN10 SET Set pin /TTL level@3.3V，high level or
> suspending is normal working status, while
> low level is sleeping mode.

```

ozzi91, 2019-06-09:
Ok now i got this!
But it seems it is not working anyway. Look at this log: 

https://pastebin.com/BtE25P9C

I tried to set up it by my way (one sensor read in one hour) but it is not working. So i tried your config and i think this is broken anyway.

OttoWinter, 2019-06-10:
@ozzi91 You probably didn't invert the pin as in my config - as the pastebin logs show, when the pin is in the low state, the sensor correctly does not receive any data.

ozzi91, 2019-06-10:
@OttoWinter 

This is my config :)

https://pastebin.com/UYUXciq4

OttoWinter, 2019-06-10:
@ozzi91 Ok try inverted: no

ozzi91, 2019-06-10:
@OttoWinter Now is working like a charm, Thank you very much!