esphome/feature-requests #276: Add new board: Heltec WiFi LoRa 32 (V2)

mseefeldt, 2019-06-07:
**Describe the problem you have/What new integration you would like**

I would like to request the introduction of the ESP32 board **Heltec WiFi LoRa 32 (V2).**
(There is already a closed issue about this board but i wanted to retry with more information
provided.)

**Please describe your use case for this integration and alternatives you've tried:**

The new board is already listed on the platform.io page under the board name [heltec_wifi_lora_32_V2](http://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_lora_32_V2.html#board-espressif32-heltec-wifi-lora-32-v2)

The flash size of the module in 8MB. Is that defined anywhere in esphome?
I didn't find a reference.

**Additional context**

The pins_arduino.h file for the already included version can be found here: [pins_arduino.h V1](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/variants/wifi_lora_32/pins_arduino.h)

The pins_arduino.h file for the new V2 version can be found here: [pins_arduino.h V2](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series/blob/master/esp32/variants/wifi_lora_32_V2/pins_arduino.h)

The meaningful diff between the two files yield the following result (i removed one diff resulting from white space differences):

` 6c6`
` < #define WIFI_LoRa_32 true`
` ---`
` > #define WIFI_LoRa_32_V2 true`
` 72,73c72,73`
` < static const uint8_t DIO1 = 33;`
` < static const uint8_t DIO2 = 32;`
` ---`
` > static const uint8_t DIO1 = 35;`
` > static const uint8_t DIO2 = 34;`

The difference in the provisioned pins is not referenced in the pin layout used by esphome.

In _esphome/esphome/pins.py_ the old board is included as

`157 'heltec_wifi_lora_32': {'LED': 25, 'BUTTON': 0, 'SDA': 4, 'SCL': 15, 'SS': 18, 'MOSI': 27,   `
`158                        'SCK': 5, 'A1': 37, 'A2': 38, 'OLED_SCL': 15, 'OLED_SDA': 4,   `
`159                        'OLED_RST': 16, 'LORA_SCK': 5, 'LORA_MOSI': 27, 'LORA_MISO': 19,   `
`160                        'LORA_CS': 18, 'LORA_RST': 14, 'LORA_IRQ': 26},`

In _esphome/dashboard/templates/index.html_ the old board is included as

`
285  <option value="heltec_wifi_lora_32">Heltec WIFI LoRa 32</option>
`

**I propose the following changes:**

* Add one line in _esphome/dashboard/templates/index.html_ after line 285:

`new  <option value="heltec_wifi_lora_32_V2">Heltec WIFI LoRa 32 V2</option>`

* Add four new lines in _esphome/esphome/pins.py_ after line 160:

`new 'heltec_wifi_lora_32_V2': {'LED': 25, 'BUTTON': 0, 'SDA': 4, 'SCL': 15, 'SS': 18, 'MOSI': 27,`
`new                        'SCK': 5, 'A1': 37, 'A2': 38, 'OLED_SCL': 15, 'OLED_SDA': 4,`
`new                        'OLED_RST': 16, 'LORA_SCK': 5, 'LORA_MOSI': 27, 'LORA_MISO': 19,`
`new                        'LORA_CS': 18, 'LORA_RST': 14, 'LORA_IRQ': 26},`

Sorry that i don't provide a pull request but i am not yet used to the github
handling. I only use git on a local basis at work.

Further references:

[Hardware specific information](https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series)

[Software specific information](https://github.com/HelTecAutomation/Heltec_ESP32)

OttoWinter, 2019-06-07:
> The flash size of the module in 8MB. Is that defined anywhere in esphome?

It's not - currently all ESP32 boards use 4MB of flash in ESPHome. But that also doesn't really matter because it would be very hard to cross 4MB with ESPHome I think (for example tests/test1.yaml that includes a whole bunch of stuff doesn't cross that threshold).

The problem with adding this board is that ESPHome doesn't use the 1.8.0 version of the platformio espressif32 package yet - so including it will result in a compile error.

Plus that would only change the pin aliases - if you use the GPIO pin names everything will be fine.

magnusoverli, 2020-06-13:
@OttoWinter What is the status of ESPhome use of platformio espressif32 package version now? Any chance of getting this board added now?

lerra, 2020-12-21:
It would be great if this board could be added and get data from sensors where wifi does not cover

joerocklin, 2020-12-29:
This seems to have been updated but not yet released. I was just able to flash (USB, then OTA) a heltec lora v2. But I had to install ESPHome with `pip install git+https://github.com/esphome/esphome@dev`.

closed 2022-07-01 by nagyrobi