esphome/feature-requests #278: AC Dimmer using zero crossing

rmesnard, 2019-06-10:
**Describe the problem you have/What new integration you would like**

I want to drive an AC Dimmer  of this type : [Robotdyn Ac Dimmer](https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.htmlhttps://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html
) to control the intensity of a light.

**Please describe your use case for this integration and alternatives you've tried:**

I made a prototype withtout esphome using Node MCU and this [library ](https://github.com/RobotDynOfficial/RBDDimmer)

**Additional context**

My goal is to use together Sonoff basic with ESPHome + Robotdyn Ac Dimmer to drive some ligth in AC

Tuckie, 2019-06-11:
FYI, you have an error in your first link

rmesnard, 2019-06-11:
thanks , i fix it.

HFsi, 2019-06-12:
I´d like to have support for these modules, too

kajhoej, 2019-06-12:
This would be awesome!

OttoWinter, 2019-06-12:
@HFsi @kajhoej Please use the upvote feature to show support (comments are spammy and upvotes make it easier to compare requests)

igalane, 2019-07-24:
Hi Otto, Do you think we can see this integration in ESPHome in the near future. I tried using it with ESP32 ledc but it did not work.

HFsi, 2019-08-30:
I'm making a custom component to drive this dimmer, the first thing I need to do is detect the ZC pulse. All the examples I saw uses the attachInterrupt command to define the pin so it can detect the ZC, but when I compile the code, I get:

`In file included from src/main.cpp:9:0:
src/tsadimmer.h: In member function 'virtual void TSADimmer::setup()':
src/tsadimmer.h:8:42: error: cannot convert 'TSADimmer::zero_cross' from type 'void (TSADimmer::)()' to type 'void (*)()'
attachInterrupt(4, zero_cross, RISING);
^
*** [/data/rfid/.pioenvs/rfid/src/main.cpp.o] Error 1`

I'd be much appreciated some help here

glmnet, 2019-08-30:
post your code too

HFsi, 2019-08-30:
`
#include "esphome.h"

class TSADimmer : public Component {
 public:
  void setup() override {
    attachInterrupt(4, zero_cross, RISING);
    pinMode(14, OUTPUT); //nodemcu D5 PWM
    //pinMode(4, INPUT); //nodemcu D2 ZC
  }

  void loop() override {
    digitalWrite(14, HIGH);
    delay(1);
    }

  void zero_cross() {
    digitalWrite(14, LOW);
  }
};
`

Is just for testing the interrupt

glmnet, 2019-08-30:
Ahhh you cannot pass a class member to attachInterrupt, the following should compile:

```
#include "esphome.h"

void zero_cross() {
  digitalWrite(14, LOW);
}

class TSADimmer : public Component {
 public:
  void setup() override {
    attachInterrupt(4, &zero_cross, RISING);
    pinMode(14, OUTPUT); //nodemcu D5 PWM
    //pinMode(4, INPUT); //nodemcu D2 ZC
  }

  void loop() override {
    digitalWrite(14, HIGH);
    delay(1);
    }

};
```

you cannot access class members from zero cross, anyway is just a custom component so big design might not be necessary can you can be a bit messy

OttoWinter, 2019-08-31:
@glmnet Is right, you need to use top-level functions here (or ESPHome's GPIOPin class, but that's a bit more complex).

Was about to implement this feature because it *is* kind of handy, but quickly was reminded again why this would not be easy: The zero-cross detection is one part (solved by pin interrupt), but to get accurate timing of the actuator you have to use a hw timer. Unfortunately, ESPHome doesn't have a platform-independent abstraction of HW Timers yet, so that would have to be added first (and then also deal with all limitations, like limited number of HW timers available).

HFsi, 2019-09-09:
It´s super handy, not only for dimming but for power factor measurement, for example.
So I can´t call a HW timer function inside a custom component?

Expaso, 2019-09-23:
Owh!! I would love this! And when put on an ESP32, it would be magic and awesomeness :)
Can I do it the messy way btw?

Expaso, 2019-09-24:
Allright then..

> you have to use a hw timer. Unfortunately, ESPHome doesn't have a platform-independent abstraction of HW Timers yet, so that would have to be added first (and then also deal with all limitations, like limited number of HW timers available).

Hmm.. I think we can work this out the elegant way..

The ESP has LEDC hardware. This hardware generates PWM signals of your choice on the pins of your choice, and uses hardware-timers under the hoods. This functionality is already supported on ESPHome by means of the LEDC component: [https://esphome.io/components/output/ledc.html?highlight=ledc](url)

Now, when using PWM hardware, you have the frequency-component, and the duty-cycle.
the duty-cycle can be used to set the brightness of the dimmer using leading-edge dimming or trailing-edge dimming. See [https://www.ecmweb.com/sites/ecmweb.com/files/uploads/2015/01/Leading-Edge.jpg](url)

The frequency has to be fixed, and set to the frequency of your mains * 2. 
Here in the Netherlands this is 50Hz (at 230V), so the PWM frequency needs to be set to 100Hz.

To prevent flickering of the lights due to frequency drifts, the PWM frequency needs to be synchronized with the mains frequency.
This is where the zero-crossing-detection-pin comes into play.

When you tie this pin to an interrupt, it will trigger 100 times a second, 2 times per full period.
**On each trigger, we only have to reset the PWM's internal timer to keep it perfectly in sync**.
We can use the ledc_timer_rst() function for this.

See also: [https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/ledc.html](url)

So.. If I can find some time for this, maybe I can propose a PR for this addition to the ESPHome's LEDC Component (or just create a new one).

smigielski, 2019-10-09:
Hi,
Would it be possible also to consider 4ch version as well. Like the one here: https://www.tindie.com/products/bugrovs2012/4ch-ac-led-light-dimmer-v2-module-controller-board/

I am not sure how much more complexity it is but 4ch is more compacted for some use-cases.

lpt2007, 2019-10-28:
Can I use [ac dimmer](https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html) in my esphome device? how?

vandreicerqueira, 2019-10-28:
> Can I use [ac dimmer](https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html) in my esphome device? how?

Unfortunately, not yet.

lpt2007, 2019-10-29:
Is there a similar product that I can use to control ac fan?

lpt2007, 2019-10-29:
Thanks for suggestion I wll stay on electronic components only:)

glmnet, 2019-10-29:
Now I don’t have a robotdyn module to do experiments but I do have a logic analyzer. 

I have been looking at how the esp8266 pwm was re-written sometime this year and I believe it might be possible to do this component. I have to do some tests first, but it will not work for esp32. 

analogWrite, Servo, etc is using this...
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_waveform.h

I’ve just starting with this but to my understanding the timing should be something like this: 

1. Wait for pulse zero crossing on an interrupt
2. Call startwaveform but it should start with the output low, I don’t know if this is “configurable” or it starts high by default and in that case I’d need to hack the waveform library

Seems like the startwaveform function allows to set the hightime, low time and then how long to run so it will auto stop, which is great as we just need a pulse to trigger the triac 

So the first delay should be something between half wavelength and 0, mapped to 0% to 100% and the “pulse” should be really short (i don’t know how long) as the triac latch itself until the next zero cross. 

Can anybody tell me if I’m right?

vandreicerqueira, 2019-10-29:
> Now I don’t have a robotdyn module to do experiments but I do have a logic analyzer.
>
> I have been looking at how the esp8266 pwm was re-written sometime this year and I believe it might be possible [...]

Regarding #1, I do have a suggestion: you might not need to call "startwaveform" on every zero crossing; those clocks (both AC mains and ESP8266) are pretty accurate. If I were you I'd call waveform only if duty cycle is changed or after 10-20 zero-crossings. So, it would reduce "CPU"  load.

In other words, your zero crossing interrupt only increments a counter; changes in waveform generation should happen only after counter goes above 20 or if you have to change light brightness (or fan speed).

OttoWinter, 2019-10-30:
@glmnet Yeah so seeing how much requested this feature is I think it's ok to make this ESP8266-only for now. As I mentioned before, making this compatible with ESP32 too would require a bunch more work.

Regarding the implementation: I don't think using the `xWaveform` functions will work well. Yes theoretically you can use them with the last param for `startWaveform`, but it won't work well for differences in setup.

AFAIK, yes the waveform you described is correct: When detecting a zero-cross, set LOW and after a time pulse HIGH (or maybe also hold HIGH until next zc). Though I think there are also different triacs (for example ones where the mapping is inverted).

I would recommend just doing the following:

- In setup: setup trigger on ZC pin, and setup HW timer1
- In ZC interrupt routine: Reset timer1 and schedule for the appropriate delta time with `timer1_write`.
- In timer1 interrupt routine, pulse triac pin.

@vandreicerqueira "So, it would reduce "CPU" load." That's irrelevant because the waveform functions are already interrupts. It would not change anything. Besides, the timer1 interrupt can be done in very few clock cycles (probably around 4-10, minus memory paging latency).

lpt2007, 2019-10-31:
There are two videos about using robotdyn ac dimmer in arduiono:

[video1](https://www.youtube.com/watch?v=eFeOSiL-IBQ&t=331s)
[video2](https://www.youtube.com/watch?v=uC9vyMdiJKQ&t=413s)

HFsi, 2019-10-31:
Not in Arduino, but in ESP8266.
But yes, that´s how I managed to make the Dimmer work with Home Assistant

CSnipper, 2019-11-08:
It would be great to have this integration.

OttoWinter, 2019-11-22:
So, seeing as this is the most requested feature in this request tracker, I thought I'd take a stab at this.

I just pushed a PR here with some (for now untested) code: https://github.com/esphome/esphome/pull/880

Expaso, 2019-12-01:
I finally had some time to write a custom integration for the ESP32, using the MCPWM modules.
(for driving the robotdyn modules)

This means there are no interrupts involved, and it purely uses the hardware to sync the dimmer signal to the grid-phase. This leaves all the CPU resources free for other tasks.

Please see attached integration for documentation, details and implementation.
It works rock-solid for me without any glitches or flickering, but do read the docs in the header for details.

Please tell me what you think or where to improve.

[zero_crossing_dimmer_output.zip](https://github.com/esphome/feature-requests/files/3908671/zero_crossing_dimmer_output.zip)

Expaso, 2019-12-01:
> Hi,
> Would it be possible also to consider 4ch version as well. Like the one here: https://www.tindie.com/products/bugrovs2012/4ch-ac-led-light-dimmer-v2-module-controller-board/
> 
> I am not sure how much more complexity it is but 4ch is more compacted for some use-cases.

See my last post for a custom integration. It will work with this board, but it may be necessary to use schmitt-trigger buffers. See the zip-file for details.

OttoWinter, 2019-12-03:
@Expaso Ah thanks!

I haven't thought of using MCPWM for ESP32 dimmer driving before, but it's the perfect peripheral to use for that task on ESP32s now that I think of it. However, if I understand correctly the MCPWM implementation would have the downside that it can only have max. 3 dimmers because there are only that many MCPWM units, right?

I'll see if I can get the implementation using timer interrupts to be more stable, but if not then MCPWM sounds like a good choice too.

glmnet, 2019-12-03:
@Expaso is that plot the red channel is the output from the ESP chip or is it the ZC signal from the robotdyn?
If it is the ESP output, then the biggest difference I see is that @OttoWinter code is sending a short pulse to trigger the triac whereas other implementations are sending a long pulse that goes low with the ZC detection. If I understand triacs properly this shouldn't make a difference.

Expaso, 2019-12-03:
> @Expaso Ah thanks!
>
> I haven't thought of using MCPWM for ESP32 dimmer driving before, but it's the perfect peripheral to use for that task on ESP32s now that I think of it. However, if I understand [...]

The ESP32 has 2 MCPWM units, with 3 independent AB-channels each, so that gives a max of 6 dimmers.
But... I only use the A-sides of the 3 channels. 
From the datasheet, it looks like I can also set the the PWM on the B-Sides to different values, so maybe I can stretch it to a max of 12 dimmers. (using a maximum of 6 different Zero-detection signals).

Expaso, 2019-12-03:
> @Expaso is that plot the red channel is the output from the ESP chip or is it the ZC signal from the robotdyn?
> If it is the ESP output, then the biggest difference I see is that @OttoWinter code is [...]

The red line is the PWM-output going to the TRIAC. 
You can see that the signal pulls low just before the zero-crossing of the AC-Sine, to let the TRIAC go OFF (so a new pulse can fire it up again).

HFsi, 2019-12-03:
> @Expaso From the datasheet, it looks like I can also set the the PWM on the B-Sides to different values, so maybe I can stretch it to a max of 12 dimmers. (using a maximum of 6 different Zero-detection signals).

 Why would be necessary to have more than one ZC detection per phase?

Expaso, 2019-12-03:
> > @Expaso From the datasheet, it looks like I can also set the the PWM on the B-Sides to different values, so maybe I can stretch it to a max of 12 dimmers. (using a maximum of 6 different Zero-detection signals).
> 
> Why would be necessary to have more than one ZC detection per phase?

The dimmer funtionality using MCPWM is a pure hardware solution. It works because the MCPWM units have internal hardware timers, which can be hardware synced with an IO pin, and PWM operators. 
There are 2 units. So, to sync both units, you need 2 IO pins. These can be tied to the same zerocross detection pin if you like. 

Having more then 1 zerocross detection can be handy if you want to drive dimmers on different AC phases at the same time. 
Or maybe you are sharing a unit with a DC motor.. 
But its purely optional

Expaso, 2019-12-03:
[image]

Tested and confirmed -> B channels can also be used.
So indeed we can drive max **12 dimmers simultaneously** without using any CPU resources!

@OttoWinter I can adapt the code to support this if you think it's useful to ESPHome.

HFsi, 2019-12-03:
@Expaso Thanks for the info

> So indeed we can drive max 12 dimmers simultaneously without using any CPU resources!

In my wildests dreams, I think of 4....12 is like the nerd heaven!

satreddys, 2019-12-08:
So, theoretically, we can use it for ceiling fans as well right? I heard about expecting some humming noises from fan.

Expaso, 2019-12-08:
I would not reccomend to use this or any dimmer on a fan or motor. You need a fan or speed controller for that.

AndresL71, 2019-12-11:
@Expaso :+1: 
can you point a Yaml config file for usinr two dimmers, one ZeroCross?

I've tried 

```
output:
    - platform: custom
      type: float
      lambda: |-
        auto detectorPin = new GPIOPin(5, INPUT, false);
        auto dimmer = new ZeroCrossingDimmerOutput(outputPin, detectorPin);
        //dimmer->tronic = true; 	//Use a tronic (MOSFET dimmer)
        //dimmer->unit = 1; 		//Select other MCPWM unit (0 or 1)
        //dimmer->channel = 0;		//Select other channel witin the unit (0, 1 or 2)
        App.register_component(dimmer);
        return {dimmer};
      outputs:
        id: light_0_out
  - platform: custom
      type: float
      lambda: |-
        auto outputPin1 = new GPIOPin(21, OUTPUT, false);
        auto detectorPin1 = new GPIOPin(5, INPUT, false);
        auto dimmer1 = new ZeroCrossingDimmerOutput(outputPin1, detectorPin1);
        //dimmer1->tronic = true; 	//Use a tronic (MOSFET dimmer)
        //dimmer1->unit = 1; 		//Select other MCPWM unit (0 or 1)
        //dimmer1->channel = 1;		//Select other channel witin the unit (0, 1 or 2)
        App.register_component(dimmer1);
        return {dimmer1};
      outputs:
        id: light_1_out
```
unable to compile ...

oOSephirothOo, 2019-12-14:
First of all thanks to @OttoWinter and @Expaso for the code.

@Expaso, I put the extracted zip into the esphome data folder and pasted the code from the leeme to the yaml file.

I get the following after try to compile...

`Compiling /data/ac_dimmer/.pioenvs/ac_dimmer/src/main.cpp.o
src/main.cpp:147:3: warning: multi-line comment [-Wcomment]
   //   lambda: !lambda "auto outputPin = new GPIOPin(16, OUTPUT, false);\nauto detectorPin\
   ^
Linking /data/ac_dimmer/.pioenvs/ac_dimmer/firmware.elf
/data/ac_dimmer/.pioenvs/ac_dimmer/src/main.cpp.o:(.literal._ZNSt17_Function_handlerIFSt6vectorIPN7esphome6output11FloatOutputESaIS4_EEvEZ5setupvEUlvE_E9_M_invokeERKSt9_Any_data+0x0): undefined reference to `vtable for ZeroCrossingDimmerOutput'
/data/ac_dimmer/.pioenvs/ac_dimmer/src/main.cpp.o:(.literal._ZNSt17_Function_handlerIFSt6vectorIPN7esphome6output11FloatOutputESaIS4_EEvEZ5setupvEUlvE_E9_M_invokeERKSt9_Any_data+0x4): undefined reference to `vtable for ZeroCrossingDimmerOutput'
collect2: error: ld returned 1 exit status
*** [/data/ac_dimmer/.pioenvs/ac_dimmer/firmware.elf] Error 1`

What I am missing?

Expaso, 2019-12-15:
@oOSephirothOo and @AndresL71:

Please bear in mind:
- This code only works for ESP32 (the MCPWM modules do not exist on an ESP8266)
- You have to include the custom component (see this YAML example)
   make sure you place the .cpp and .h file in this folder:
   <home assistant config dir>/esphome/custom_components/zero_crossing_dimmer/

```
esphome:
  name: $hostname
  platform: ESP32
  board: esp32dev
  includes:
    - custom_components/zero_crossing_dimmer

output:
   - platform: custom
     type: float
     lambda: |-
       auto outputPin = new GPIOPin(16, OUTPUT, false);
       auto detectorPin = new GPIOPin(2, INPUT, false);
       auto dimmer = new ZeroCrossingDimmerOutput(outputPin, detectorPin);
       App.register_component(dimmer);
       return {dimmer};
     outputs:
       id: dimmer_output

# Example usage in a light
light:
  - platform: monochromatic
    output: dimmer_output
    name: "AC Dimming Light"
    gamma_correct: 0
    id: ACDimmerLight

```

Expaso, 2019-12-15:
> can you point a Yaml config file for using two dimmers, one ZeroCross?

It looks like your yaml indentation is wrong. The two platform blocks aren't on the same level.

AndresL71, 2019-12-15:
corrected indentation 
This is the complete yaml and compiling errors...
Thanks

```
esphome:
  name: dorm_ppal
  platform: ESP32
  board: esp32doit-devkit-v1
  includes:
    - zero_crossing_dimmer_output.h

wifi:
  ssid: "router"
  password: "+++++++"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dorm Ppal Fallback Hotspot"
    password: "+++++++"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "++++++++"

ota:
  password: "++++++++"

light:
  - platform: monochromatic
    name: "Luz Dormitorio tele"
    id: light_0
    output: light_0_out

  - platform: monochromatic
    name: "Luz Dormitorio cama"
    id: light_1
    output: light_1_out

# Example output entry
# output:
#  - platform: custom
#    id: light_0_out
#    pin: D19
#    inverted: true

output:
  - platform: custom
    type: float
    lambda: |-
      auto outputPin = new GPIOPin(19, OUTPUT, false);
      auto detectorPin = new GPIOPin(5, INPUT, false);
      auto dimmer = new ZeroCrossingDimmerOutput(outputPin, detectorPin);
      //dimmer->tronic = true; 	//Use a tronic (MOSFET dimmer)
      //dimmer->unit = 1; 		//Select other MCPWM unit (0 or 1)
      //dimmer->channel = 0;		//Select other channel witin the unit (0, 1 or 2)
      App.register_component(dimmer);
      return {dimmer};
    outputs:
        id: light_0_out

#  - platform: custom
#    id: light_1_out
#    pin: D21
#    inverted: true

  - platform: custom
    type: float
    lambda: |-
      auto outputPin1 = new GPIOPin(21, OUTPUT, false);
      auto detectorPin1 = new GPIOPin(5, INPUT, false);
      auto dimmer1 = new ZeroCrossingDimmerOutput(outputPin1, detectorPin1);
      //dimmer1->tronic = true; 	//Use a tronic (MOSFET dimmer)
      //dimmer1->unit = 1; 		//Select other MCPWM unit (0 or 1)
      //dimmer1->channel = 1;		//Select other channel witin the unit (0, 1 or 2)
      App.register_component(dimmer1);
        return {dimmer1};
    outputs:
        id: light_1_out

binary_sensor:
  - platform: gpio
    pin: GPIO23
    id: light_0_touch
    on_click:
      then:
        - if:
            condition:
              light.is_off: light_0
            then:
              light.turn_on: 
                id: light_0
                brightness: 1.0
            else:
              light.turn_off: light_0
    on_press:
      then:
      - if:
          condition: 
            light.is_off: light_0
          then:
          - delay: 0.5s
          - while:
              condition:
                binary_sensor.is_on: light_0_touch
              then:
                - light.dim_relative:
                    id: light_0
                    relative_brightness: 5%
                    transition_length: 0.1s
                - delay: 0.1s
          else:
          - delay: 0.5s
          - while:
              condition:
                and:
                  - binary_sensor.is_on: light_0_touch
                  - light.is_on: light_0
              then:
                - light.dim_relative:
                    id: light_0
                    relative_brightness: -5%
                    transition_length: 0.1s
                - delay: 0.1s
  - platform: gpio
    pin: GPIO22
    id: light_1_touch
    on_click:
      then:
        - if:
            condition:
              light.is_off: light_1
            then:
              light.turn_on: 
                id: light_1
                brightness: 1.0
            else:
              light.turn_off: light_1
    on_press:
      then:
      - if:
          condition: 
            light.is_off: light_1
          then:
          - delay: 0.5s
          - while:
              condition:
                binary_sensor.is_on: light_1_touch
              then:
                - light.dim_relative:
                    id: light_1
                    relative_brightness: 5%
                    transition_length: 0.1s
                - delay: 0.1s
          else:
          - delay: 0.5s
          - while:
              condition:
                and:
                  - binary_sensor.is_on: light_1_touch
                  - light.is_on: light_1
              then:
                - light.dim_relative:
                    id: light_1
                    relative_brightness: -5%
                    transition_length: 0.1s
                - delay: 0.1s
```
```
Processing dorm_ppal (board: esp32doit-devkit-v1; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <WiFi> 1.0
|   |-- <FS> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
Linking .pioenvs/dorm_ppal/firmware.elf
.pioenvs/dorm_ppal/src/main.cpp.o:(.literal._ZNSt17_Function_handlerIFSt6vectorIPN7esphome6output11FloatOutputESaIS4_EEvEZ5setupvEUlvE0_E9_M_invokeERKSt9_Any_data+0x0): undefined reference to `vtable for ZeroCrossingDimmerOutput'
.pioenvs/dorm_ppal/src/main.cpp.o:(.literal._ZNSt17_Function_handlerIFSt6vectorIPN7esphome6output11FloatOutputESaIS4_EEvEZ5setupvEUlvE0_E9_M_invokeERKSt9_Any_data+0x4): undefined reference to `vtable for ZeroCrossingDimmerOutput'
collect2: error: ld returned 1 exit status
*** [.pioenvs/dorm_ppal/firmware.elf] Error 1
========================== [FAILED] Took 4.68 seconds ==========================
```

Expaso, 2019-12-15:
@AndresL71 

You have to adjust your include in the ESPHome section of your yaml to point to the _folder_ of the 2 source files, instead of only the .h file.
Place the .cpp and the .h file in a separate folder, and include that foldername.

With this modificaiton, i could successfully compile your yaml.

AndresL71, 2019-12-15:
Thanks !!!
Stupid mistake.

Expaso, 2019-12-15:
I hope it was just in time to finish your christmas lights 😋

markiv25, 2019-12-16:
hello 
i had worked on dimmer project on esp32 using mcpwm but there was a lot of flickering
heres my code could any one help me with it
//#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "driver/gpio.h"
#include "mcpwm1.h"
#include "string.h"
#include "freertos/queue.h"
#include "esp_attr.h"
#include "soc/rtc.h"
#include "soc/mcpwm_reg.h"
#include "soc/mcpwm_struct.h"
//#include "mcpwm.h"
#define GPIO_PWM0A_OUT 17
#define GPIO_PWM0B_OUT 19
#define GPIO_PWM1A_OUT 21
#define GPIO_PWM1B_OUT 22
#define GPIO_PWM2A_OUT 23
#define GPIO_PWM2B_OUT 12
#define GPIO_SYNC0_IN 18 //Set GPIO 18 as SYNC0
//#define MCPWM_CLK_PRESCL 159 //MCPWM clock prescale------160/160=1mhz
//#define TIMER_CLK_PRESCALE 199 //MCPWM timer prescales----1mhz/(50hz*(39+1))--50hz
mcpwm_pin_config_t pin_config;
mcpwm_config_t pwm_config;
mcpwm_timer_t i;
// MCPWM_TIMER_0;
// MCPWM_TIMER_1;
// MCPWM_TIMER_2;
int k;
mcpwm_dev_t *MCPWM_U[2] = {&MCPWM0, &MCPWM1};
void setup() {
Serial.begin(115200);
pin_config.mcpwm0a_out_num = GPIO_PWM0A_OUT;
pin_config.mcpwm0b_out_num = GPIO_PWM0B_OUT;
pin_config.mcpwm1a_out_num = GPIO_PWM1A_OUT;
pin_config.mcpwm1b_out_num = GPIO_PWM1B_OUT;
pin_config.mcpwm2a_out_num = GPIO_PWM2A_OUT;
pin_config.mcpwm2b_out_num = GPIO_PWM2B_OUT;
pin_config.mcpwm_sync0_in_num = GPIO_SYNC0_IN;
gpio_pulldown_en(GPIO_NUM_18);
for(int num=0;num<3;num++){
if (num==0){ i=MCPWM_TIMER_0;}
else if (num==1){ i=MCPWM_TIMER_1;}
else if (num==2){ i=MCPWM_TIMER_2;}
pwm_config.frequency = 50;
pwm_config.cmpr_a = 0;
pwm_config.cmpr_b = 0;//duty cycle of PWMxb = 50.0%
pwm_config.counter_mode = MCPWM_UP_COUNTER;
pwm_config.duty_mode = MCPWM_DUTY_MODE_1;
mcpwm_init(MCPWM_UNIT_0, i, &pwm_config);

mcpwm_set_pin(MCPWM_UNIT_0, &pin_config);

// mcpwm_sync_enable(MCPWM_UNIT_0, i,MCPWM_SELECT_SYNC0, 1);
}
MCPWM_U[0]->clk_cfg.prescale=255;
MCPWM_U[0]->timer[0].period.period=300; //----10MS TIME
MCPWM_U[0]->timer[0].period.prescale=255;
MCPWM_U[0]->timer[0].period.upmethod = 0;
MCPWM_U[0]->channel[0].cmpr_value[0].cmpr_val = 150;
MCPWM_U[0]->channel[0].cmpr_cfg.a_upmethod = BIT(0);
MCPWM_U[0]->timer[0].sync.timer_phase=10;
MCPWM_U[0]->timer_synci_cfg.t0_in_sel = MCPWM_SELECT_SYNC0;
MCPWM_U[0]->channel[0].gen_cfg0.t0_sel= 3;
MCPWM_U[0]->channel[0].gen_cfg0.upmethod= 1;
MCPWM_U[0]->timer[0].sync.in_en = 1;
MCPWM_U[0]->timer_sel.operator0_sel=0;
MCPWM_U[0]->channel[0].generator[0].ut0=3;
// MCPWM_U[0]->channel[0].generator[0].utez=1;
//MCPWM_U[0]->channel[0].generator[0].utep=2;
}
void loop(){
if (Serial.available()>0)
{ k= Serial.parseInt();
MCPWM_U[0]->timer[0].sync.timer_phase=k;
Serial.println(k);
}
}

Expaso, 2019-12-16:
Just download the source code of the custom component I created. 
There's plenty of comments and information there. 
And make sure you use schmitt-trigger buffers in the zero-detect line.

markiv25, 2019-12-19:
vikram-vp@vikram-vp:~$ esphome room.yaml run
WARNING You're using ESPHome with python 2. Support for python 2 is deprecated and will be removed in 1.15.0. Please reinstall ESPHome with python 3.6 or higher.
INFO Reading configuration room.yaml...
ERROR Error while reading config: Invalid YAML syntax. Please see YAML syntax reference or use an online YAML syntax validator:

mapping values are not allowed here
  in "room.yaml", line 27, column 13:
          output: dimmer_output

markiv25, 2019-12-19:
could you help?

LegendOfSparta, 2020-01-13:
i'm using this circuit https://techiesms.com/iot-projects/esp01-projects/internet-controlled-ac-dimmer-using-esp8266-and-mqtt/ to make a light dimmer, the circuit is fine, what stops me is making a working code for Esphome, i won't be using MQTT, anyone would help or guide me through this ?

glmnet, 2020-01-13:
> i'm using this circuit https://techiesms.com/iot-projects/esp01-projects/internet-controlled-ac-dimmer-using-esp8266-and-mqtt/ to make a light dimmer, the circuit is fine, what stops me is making a working code for Esphome, i won't be using MQTT, anyone would help or guide me through this ?

Did you get this working with another software? Whats the GPIO 1 TX for? I believe it should be connected to GND instead. Usually the drivers have two pins interface, the ZC and the Output trigger. (e.g. TRIAC)

You should follow the instructions from the PR here https://github.com/esphome/esphome/pull/880#issuecomment-557614337

Fall3n-be, 2020-01-15:
Is it already possible to try this out in Home assistant?

LegendOfSparta, 2020-01-15:
@glmnet thanks mate, but no I haven’t tried it... looking to it, yeah it should be grounded!

Anyway the 4 files, where should i put them exactly in my “/config”?

Thanks again for the help

AndresL71, 2020-01-15:
config/custom_components/dimmer/files.*

LegendOfSparta, 2020-01-15:
i already tried that didn't work, should i add "#include dimmer.h" or something ?

glmnet, 2020-01-15:
What error is throwing

LegendOfSparta, 2020-01-16:
It won’t compile and says “output.dimmer can’t be found”(something like that) on the line “ platform: dimmer”

glmnet, 2020-01-16:
> It won’t compile and says “output.dimmer can’t be found”(something like that) on the line “ platform: dimmer”

the *custom_components* folder must be at the same level than your yaml file, then *dimmer* folder in there and the files there, make sure you get the files right, some people downloads the web page instead of the file, you have to right click the *Raw* button and select *Save target as*

Luisiado, 2020-02-19:
Thanks for the component, I am trying to compile and I am getting erros:

> src/main.cpp: In lambda function:
src/main.cpp:126:25: error: expected type-specifier before 'ZeroCrossingDimmerOutput'
       auto dimmer = new ZeroCrossingDimmerOutput(outputPin, detectorPin);
                         ^
src/main.cpp:128:21: error: could not convert '{dimmer}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::output::FloatOutput*>'
       return {dimmer};
                     ^
*** [/data/dimmertriac/.pioenvs/dimmertriac/src/main.cpp.o] Error 1

> 

my yaml
`esphome:
  name: dimmertriac
  platform: ESP32
  board: esp32doit-devkit-v1
  includes:
    - custom_components/zero_crossing_dimmer/dimmer.h
   

# custom component  https://github.com/esphome/esphome/tree/ac-dimmer/esphome/components/dimmer

wifi:
  ssid: 
  password: 

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 

ota:
  password: 

status_led:
  pin: GPIO2
 
output:
  - platform: custom
    type: float
    lambda: |-
      auto outputPin = new GPIOPin(16, OUTPUT, false);
      auto detectorPin = new GPIOPin(2, INPUT, false);
      auto dimmer = new ZeroCrossingDimmerOutput(outputPin, detectorPin);
      App.register_component(dimmer);
      return {dimmer};
    outputs:
        id: dimmer_output

# Example usage in a light
light:
  - platform: monochromatic
    output: dimmer_output
    name: "AC Dimming Light"
    gamma_correct: 0
    id: ACDimmerLight`

and custom components folder is inside of config/esphome/custom_componets....

[8 more replies]

closed 2020-04-10 by glmnet