esphome/feature-requests #280: Allow homeassistant.service to submit arrays

kirichkov, 2019-06-12:
**Describe the problem you have/What new integration you would like**

The homeassistant `device_tracker.see` service allows submission of GPS coordinates. These coordinates are submit as array with the elements - latitude and longitude. Currently when one adds an array to the `data` part of a `homeassistant.service` action gets an error:

```
string value cannot be dictionary or list.
gps: 
  - 00.000000
  - 11.000000
```

**Please describe your use case for this integration and alternatives you've tried:**

I'm firing up the homeassistant service `device_tracker.see` to integrate the ESPHome's BLE presence binary_sensor with the person's component. I've described my approach in https://github.com/esphome/feature-requests/issues/204

**Additional context**

Here's the config I have:
```
binary_sensor:
  - platform: ble_presence
    mac_address: AA:BB:CC:DD:EE:FF
    name: Tracked device
    id: tracked_device
    on_state:
      then:
        - if:
            condition: 
              binary_sensor.is_on: tracked_device
            then:
              - homeassistant.service:
                  service: device_tracker.see
                  data:
                    dev_id: tracked_device
                    location_name: 'home'
                    gps: [00.000000, 11.111111]
                    gps_accuracy: 10
            else:
              - homeassistant.service:
                  service: device_tracker.see
                  data:
                    dev_id: tracked_device
                    location_name: 'not_home'
```

OttoWinter, 2019-06-12:
Not easy to add because of some architectural issues. (and HA's service call schemas is a bit broken anyway)

Probably you're best off by using a script in HA and triggering that from homeassistant.service.

kirichkov, 2019-06-12:
Using a script is a good solution, actually, but it did not occur to me 🤦‍♂  I wanted to mention this in case it was a bug!

closed 2022-07-07 by nagyrobi