esphome/feature-requests #281: Display latest commit in Webinterface

Xeroxxx, 2019-06-12:
**Describe the problem you have/What new integration you would like**

Display the latest commit in ESPHome (Dev) Webinterface. Like "Current version: Commit 697"

I think this would be great to see which version is currently used when using dev repo.

**Please describe your use case for this integration and alternatives you've tried:**

-

**Additional context**

-

Xeroxxx, 2019-06-13:
Maybe something like this.

[image]

Xeroxxx, 2019-06-13:
To clarify the latest commit of the build thats being used, so I know which commit I'm using right now.

brandond, 2019-06-13:
Commits are usually referred to by their git short hash. Not every commit will have a MR# associated with it. You'd probably be best off showing the version, and maybe the commit hash for -dev versions.

OttoWinter, 2019-06-14:
Thought about this some more. And I'm gonna close this feature request.

It looks a lot like @Xeroxxx wants this from a user perspective - for using the dev version of esphome in production devices.

You *should not* do that - even though it might work now, the dev branch *can* always be unstable and could break any number of things. I do not want to add features that promote using dev in any way.

From the developer perspective, this feature doesn't really add any value either because then you already have to use a git client of some sort to check out commits. And a git commit hash really isn't that useful here either (because it's a hash)

closed 2019-06-14 by OttoWinter

Xeroxxx, 2019-06-14:
Okay, I'm good with this.