esphome/feature-requests #283: Support for KlikAnKlikUt (KaKu) RF433 Switches

bluesheep, 2019-06-13:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

OttoWinter, 2019-06-14:
FYI, technically already supported through raw remote codes.

Juggels, 2019-08-16:
I've been trying to get this working by using the raw codes, but haven't had any luck yet so far 🙁

After looking at the logs with a remote_receiver component, I have managed to see the codes that are being sent by the original KaKu button. Then, in order to send (e.g.) a group on request I have tried to do the following based on the logic as it is implemented in NewRemoteTransmitter:

```yaml
# Set up remote transmitter
remote_transmitter:
  pin: 
    number: D9
    mode: OUTPUT
  carrier_duty_percent: 100%

switch:
  - platform: template
    name: Raw Code Power Button
    id: test
    turn_on_action:
      # Start pulse
      - remote_transmitter.transmit_raw:
          code: [3138, -1294, 731, -267, 1739, -1270, 735, -275, 1733, -1282, 722, -1275, 732, -1274, 731, -1278, 728, -278, 1728, -282, 1724, -281, 1726, -286, 1720, -290, 1717, -289, 1718, -1284, 723, -1287, 719, -287, 1719, -292, 1713, -1289, 718, -1287, 720, -1296, 711, -1288, 719, -1294, 712, -1293, 714, -1302, 702, -298, 1712, -298, 1708, -1299, 715, -1298, 709, -1292, 715, -1297, 710, -1296, 715]
      # Address (45878526)
      - remote_transmitter.transmit_raw:
          code: [3090, -1293, 714, -292, 1715, -1293, 714, -300, 1707, -1300, 707, -1309, 697, -1298, 709, -1299, 708, -300, 1707, -304, 1702, -297, 1710, -299, 1704, -304, 1710, -319, 1685, -1301, 706, -1302, 705, -298, 1708, -299, 1708, -1304, 703, -1300, 707, -1300, 705, -1309, 697, -1300, 705, -1303, 704, -1307, 701, -302, 1705, -300, 1707, -1307, 700, -1300, 708, -1304, 702, -1303, 704, -1301, 706]
      - remote_transmitter.transmit_raw:
          code: [3085, -1298, 708, -304, 1704, -1303, 703, -301, 1707, -1303, 704, -1302, 705, -1301, 706, -1303, 704, -299, 1708, -303, 1703, -305, 1702, -304, 1703, -301, 1706, -300, 1707, -1301, 706, -1300, 707, -301, 1706, -302, 1706, -1300, 707, -1299, 707, -1300, 708, -1305, 701, -1301, 706, -1305, 702, -1310, 694, -312, 1697, -310, 1700, -1305, 702, -1307, 699, -1299, 709, -1303, 703, -1302, 706]
      - remote_transmitter.transmit_raw:
          code: [3087, -1306, 700, -300, 1707, -1296, 711, -298, 1709, -1306, 702, -1297, 709, -1302, 706, -1309, 698, -301, 1706, -299, 1709, -307, 1700, -300, 1702, -303, 1707, -304, 1703, -1298, 709, -1297, 711, -299, 1708, -296, 1711, -1314, 694, -1301, 706, -1301, 705, -1298, 709, -1298, 710, -1297, 709, -1302, 705, -300, 1708, -298, 1709, -1304, 703, -1301, 707, -1299, 708, -1300, 707, -1303, 703]
      - remote_transmitter.transmit_raw:
          code: [3083, -1295, 713, -298, 1708, -1296, 711, -293, 1713, -1296, 711, -1295, 713, -1295, 712, -1303, 704, -296, 1712, -294, 1713, -294, 1714, -294, 1712, -295, 1711, -300, 1707, -1302, 705, -1299, 708, -300, 1707, -296, 1711, -1301, 707, -1304, 703, -1296, 711, -1314, 692, -1304, 703, -1306, 698, -1301, 709, -298, 1710, -300, 1707, -1301, 706, -1300, 707, -1298, 709, -1309, 698, -1297, 711]
      - remote_transmitter.transmit_raw:
          code: [3088, -1297, 711, -292, 1710, -1304, 703, -301, 1710, -1295, 712, -1300, 707, -1293, 714, -1300, 707, -296, 1712, -292, 1715, -295, 1712, -293, 1687, -349, 1685, -299, 1708, -1297, 709, -1295, 713, -299, 1708, -297, 1710, -1298, 709, -1296, 711, -1300, 708, -1293, 714, -1311, 697, -1293, 713, -1296, 711, -294, 1713, -296, 1711, -1297, 710, -1293, 715, -1298, 709, -1294, 714, -1294, 712]
      - remote_transmitter.transmit_raw:
          code: [3087, -1290, 717, -291, 1716, -1289, 718, -292, 1715, -1293, 714, -1294, 713, -1293, 714, -1290, 717, -296, 1711, -293, 1714, -292, 1716, -294, 1712, -294, 1713, -293, 1714, -1297, 709, -1293, 715, -290, 1717, -291, 1715, -1296, 712, -1292, 715, -1292, 715, -1292, 715, -1293, 711, -1310, 710, -1289, 710, -294, 1711, -293, 1714, -1294, 714, -1297, 709, -1293, 715, -1292, 716, -1294, 712]
      - remote_transmitter.transmit_raw:
          code: [3087, -1292, 716, -291, 1716, -1289, 718, -290, 1717, -1292, 715, -1291, 716, -1295, 713, -1294, 712, -291, 1716, -295, 1707, -299, 1713, -293, 1713, -292, 1716, -290, 1717, -1295, 712, -1295, 712, -294, 1713, -293, 1714, -1294, 713, -1296, 711, -1293, 714, -1298, 709, -1297, 711, -1294, 712, -1294, 713, -299, 1708, -298, 1709, -1295, 712, -1295, 712, -1300, 707, -1294, 714, -1302, 705]
      - remote_transmitter.transmit_raw:
          code: [3087, -1293, 714, -291, 1716, -1293, 714, -292, 1715, -1291, 716, -1292, 715, -1294, 713, -1295, 712, -291, 1716, -291, 1715, -291, 1700, -340, 1685, -292, 1715, -290, 1716, -1295, 713, -1296, 710, -292, 1715, -291, 1716, -1292, 715, -1293, 715, -1291, 715, -1293, 715, -1296, 706, -1302, 710, -1293, 714, -294, 1713, -292, 1715, -1296, 711, -1291, 716, -1292, 715, -1292, 715, -1292, 716]
      # Group bit (1)
      - remote_transmitter.transmit_raw:
          code: [3089, -1293, 714, -291, 1716, -1292, 714, -291, 1717, -1291, 716, -1295, 712, -1294, 714, -1293, 714, -292, 1714, -296, 1706, -301, 1719, -308, 1692, -289, 1718, -292, 1714, -1293, 714, -1297, 711, -290, 1716, -293, 1714, -1296, 711, -1298, 709, -1292, 715, -1301, 707, -1292, 715, -1292, 714, -1297, 711, -292, 1715, -295, 1712, -1294, 713, -1291, 716, -1293, 714, -1297, 711, -1293, 713]
      # On/Off (1)
      - remote_transmitter.transmit_raw:
          code: [3093, -1291, 715, -289, 1719, -1293, 713, -289, 1718, -1289, 718, -1291, 716, -1293, 715, -1290, 716, -293, 1714, -294, 1713, -291, 1716, -291, 1716, -298, 1709, -295, 1711, -1294, 713, -1295, 713, -293, 1714, -297, 1710, -1293, 714, -1291, 716, -1298, 709, -1293 [... cut ...]
          code: [3093, -1292, 715, -297, 1709, -1295, 712, -290, 1718, -1290, 717, -1292, 714, -1294, 713, -1294, 709, -297, 1714, -293, 1715, -293, 1713, -291, 1716, -299, 1708, -296, 1711, -1295, 712, -1293, 714, -292, 1715, -293, 1714, -1294, 713, -1295, 713, -1290, 717, -1294, 712, -1294, 713, -1299, 709, -1291, 716, -294, 1713, -291, 1715, -1292, 716, -1293, 714, -1292, 715, -1297, 710, -1293, 714]
```

Unfortunately, that doesn't trigger any of the KaKu devices that are listening for that code. When I run a simple Arduino project with the RemoteSwitch library, it does work.

Is there anything else I could/should try?

bluesheep, 2019-08-16:
I'm in the same position, I'm sorry for not responding to the previous message. I have been unable to get the raw transmissions to work.

glmnet, 2019-08-17:
That is one large code for a single switch. 

Are you sure you’re not trying to do several switches?

Juggels, 2019-08-17:
I agree it seems rather big, but that’s what `remote_receiver` caught when pushing the original button. The ‘new’ KaKu protocol being weird as it is, I thought it could be worth the try.

andrasbiro, 2020-01-03:
I looked into this, and I'm fairly sure it won't be possible to support kaku with the rc_switch protocol. Instead, it would be nice to see something similar to Portisch's flexible implementation:
https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Define-new-protocol-in-RF_Protocols.h
(written for the EFM8 on the sonoff gateway, also would be nice to have support for it)

This seems to be a superset of rc_switch - I'm thinking of porting it myself but I don't think I will have the time and motivation to finish it :)

andrasbiro, 2020-01-04:
Okay, I realized that it is possible to make it work, see above. The remaining issues to fully support kaku:
- documentation recommends 250us filter, that's way too high for 433 protocols,  kaku can go down to ~150us. I'm using 50us, which converts to 20kbps chiprate - that is unusual, but 10kbps is not. 
- tx requires sync to be sent at the beginning, not the end. I have no clue how the current implementation works, unless with repeating tx without wait time. If you detect the sync first, you should send the sync first. This come up earlier: https://github.com/esphome/issues/issues/120
- kaku requires a footer at the end of the frame. It could be actually received and transmitted as a 1 bit (rcswitch bit, not kaku bit), but that would be the 65th bit of the frame. Instead, it would be nice to have a configurable footer, similar to the sync. As a hack, I tried sending a sync at the beginning and the end, and this did work.

github-k8n, 2023-05-24:
It seems the merged PR only enabled support for NEXA protocol, while KaKu is still unsupported.
There's more information on the KaKu protocol here:
https://manual.pilight.org/protocols/433.92/switch/kaku.html

(On a side-note, the KaKu devices are sometimes received/identified as NEXA devices and sending works if you send the same NEXA code AND repeat it (doesn't work without repeating) which confirms the issue described by @andrasbiro regarding the sync (beginning vs end).

github-k8n, 2023-05-24:
Keep in mind that there is "old" KaKu (which gets identified by and works with rc_switch_raw) and "new" KaKu protocol which seems to be related to NEXA. My guess is that the link above is the new one (I hope), just as a heads up in case there is some confusion regarding those two.

github-k8n, 2023-05-24:
I have three Intertechno ITLR-3500 switches and I could send one of them if it helps to progress here.
(also I'd be happy to test new versions of the remote receiver/transmitter components anytime.. using a modded Sonoff RF bridge but also have an ESP32 with a RXB6 receiver)

Virusmater, 2025-02-23:
@andrasbiro do you have an YAML example where you could make KaKu transmitting work?

andrasbiro, 2025-02-25:
@Virusmater To my surprise, it looks like I have: https://pastebin.com/PjAVMeLD

At that time, I was working with two completely different 433 switch, the "second one" is the tanagy_ prefix. I don't remember much else.

Virusmater, 2025-03-04:
My setup is ESP32 with CC1101

I tried different types of the code formats: tried to read raw, Nexa, rf codes and then simulate them, tried rflink, openmqtt, and some other tools. Nothing worked for me or I couldn't figure it out how.

Only thing that worked for me was to read RAW signal using Flipper Zero, then convert it into ESPHome raw format and then send it as RAW.

I don't know what the reason is, but I think I will keep it like this. I would be happy to answer the questions from people who would like to decrypt the protocol and integrate it into ESPHome.