esphome/feature-requests #285: Add template fan

br0nd, 2019-06-16:
**Describe the problem you have/What new integration you would like**
I have a variable speed pool pump that I would like to see as a fan in Home assistant

**Please describe your use case for this integration and alternatives you've tried:**
My variable speed pool pump has 3 speeds controlled using 2 relays.. 
Off - Both relays off
Low - Relay 1 on, Relay 2 off
Medium - Relay 1 off, Relay 2 on
High - Relay 1 on, Relay off

I currently am using four different template switches.. to represent these states and to set the appropriate GPIO outputs.. 
I would like to set this as a template fan.. something along the lines of..

`fan:
  - platform: template
    name: "Pool Pump"
    speed: 
      off:
        - output.turn_off:
            id: pool_pump_relay_1
        - output.turn_off:
            id: pool_pump_relay_2 
      low:
        - output.turn_on:
            id: pool_pump_relay_1
        - output.turn_off:
            id: pool_pump_relay_2  
      medium:
        - output.turn_off:
            id: pool_pump_relay_1
        - output.turn_on:
            id: pool_pump_relay_2  
      high:
        - output.turn_on:
            id: pool_pump_relay_1
        - output.turn_on:
            id: pool_pump_relay_2`

----
I tried using the  fan platform.. and then writing an automation using duty_cycle_sensor, pulse_width, pulse_counter.. with on_value_range to read the fan dummy PWM GPIO output and then kickoff additional GPIO pin outputs to activate relays, but have not been able to figure it out.. I tried something like this 

`output:
  - platform: ledc
    id: dummy_pool_pump
    pin: GPIO5
fan:
  - platform: speed
    output: dummy_pool_pump
    name: "Pool Pump"
  - platform: duty_cycle
    pin: GPIO5
    id: pool_pump_duty_cycle
    on_value_range:
      - above: 50
        then:
          - switch.turn_on: pool_pump_high
      - below: 50
        then:
          - switch.turn_on: pool_pump_off
`
**Additional context**

OttoWinter, 2019-06-17:
FYI, you can also just use a simple custom output platform, then use the value received there to set `digitalWrite` commands.

OttoWinter, 2019-06-17:
Looks like https://github.com/esphome/feature-requests/issues/60 is very similar - will close in favor of #60

closed 2019-06-17 by OttoWinter