esphome/feature-requests #287: Support for HP303B - Digital Pressure Sensor

rtonin, 2019-06-18:
**Describe the problem you have/What new integration you would like**

Request support for the HP303B - Digital Pressure Sensor.

**Please describe your use case for this integration and alternatives you've tried:**

The HP303B - Digital Pressure Sensor is a well known sensor used in the Wemos (Lolin) Barometric Pressure Shield and it would be great to be able to use it in ESPHome.

**Additional context**

Wemos BP Shield Page - https://wiki.wemos.cc/products:d1_mini_shields:barometric_pressure_shield
HP303B Datasheet - https://cdn.sos.sk/productdata/59/18/835e5d50/hp303b.pdf
LOLIN_HP303B_Library - https://github.com/wemos/LOLIN_HP303B_Library

Skynet2029, 2019-08-09:
Take a look https://github.com/dmptrluke/esphome_hp303b

huskyte, 2022-03-17:
Could it be that the above link/solution doesn't work as of the dependencies anymore (hasn't been updated) AND the HP303B is still not included in ESPHome as a standard component?

max246, 2023-05-17:
Would you want this to be included into esphome or just leave it as external ?

Judge-Tread, 2023-05-18:
@max246 I would LOVE for this to be included with esphome, as it's driving me CRAZY trying to add it as a custom sensor (using the above link/git)

max246, 2023-05-18:
> @max246 I would LOVE for this to be included with esphome, as it's driving me CRAZY trying to add it as a custom sensor (using the above link/git)

Cool I will add on my list

Judge-Tread, 2023-05-18:
You are an absolute diamond - two VERY patient mods in the discord have been trying to help me add this, but I'm going round in circles :(

max246, 2023-05-19:
> You are an absolute diamond - two VERY patient mods in the discord have been trying to help me add this, but I'm going round in circles :(

I pushed a PR but will make it work during the next few days. We can work on Discord to get it working in case there is any issues.

Judge-Tread, 2023-05-19:
Blimey that was quick! When people say "add to the list" I usually expect weeks before anything starts happening! 

I'll keep my eyes peeled

shish, 2023-06-17:
I couldn't make the existing components work, so I made my own - I hope I can drop it as soon as support is merged upstream, but in the meantime, mentioning it here in case anybody finds it useful :)

https://github.com/shish/esphome-projects/tree/master/components/hp303b

ralphhughes, 2023-09-24:
Is there any update on support for HP303B making it into ESPHome?

max246, 2023-09-24:
> Is there any update on support for HP303B making it into ESPHome?

Yeah Sorry  got busy doing some work stuff. I am close with the pr to support i2c an SPI...