esphome/feature-requests #288: Something like "on tag" but for BLE, with some nice variable

nerdosity, 2019-06-18:
**Describe the problem you have/What new integration you would like**
I would like to have an event that can fire an automation when a BLE device is found, and this automation should have access to another variable: distance from BLE device.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to be able to integrate my BLE devices with MQTT Room component of home assistant.
So we should need to format a packet like this:

{
  "id": "123testid",
  "name": "Test Device",
  "distance": 5.678
}
and publish it to some topic.

**Additional context**
https://www.home-assistant.io/components/mqtt_room/

I tried https://github.com/happy-bubbles/presence and https://github.com/mKeRix/room-assistant, they both work well but Happy Bubble project is dead and room-assistant is based on raspberry devices, it would cost much less and would be much easier using ESP32 instead of raspberry

animavitis, 2019-09-30:
I was doing it some time ago
`sensor:
   - platform: ble_rssi
    mac_address: 84:A4:66:89:75:8E
    id: rssi_sygonix
    internal: true
    on_value:
    - mqtt.publish_json:
        topic: home_presence/room_1
        payload: |-
          root["id"] = "84:A4:66:89:75:8E";
          root["name"] = "sygonix";
          root["distance"] = (0.89)* pow(id(rssi_sygonix).state/-59,7.7095) + 0.11;`

nerdosity, 2019-10-17:
That's good, but it's just for THAT value. It would be easier to get all values pushed to Home Assistant and manage everything just from there.

tlrobinson, 2019-12-01:
I’m looking to do something similar for integration with [find3](https://github.com/schollz/find3). I want to publish all BLE devices, their RSSI, and the device name to a MQTT topic. Also it should happen periodically rather than just when the device is first seen.

kmdm, 2021-02-08:
Just to say I implemented something like this for iBeacons using the UUID instead of the BT MAC (which can be random). It sends all detected iBeacons to MQTT. 

https://gist.github.com/kmdm/c01c0693b1380d826af200a3e9b2a656

It uses a different distance formula (the first one Google showed me 😆) and I have no issue changing it at all. (NB: The C++ could probably be better!)

I'd just like to mention what I've done, get any feedback and give my 👍 for this being a "first class" feature instead of something needing to be implemented in lambdas.

closed 2022-07-01 by nagyrobi