esphome/feature-requests #289: SDM630

pepeEL, 2019-06-19:
**Describe the problem you have/What new integration you would like**

Is any chance to add support for Energy Unit Metere SDM630 ? This will be a good features....
**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

lrybak, 2019-06-19:
The Eastron SDM120/SDM630 are popular, widely used RS48m/modbus powers meters
http://www.eastrongroup.com/productsview/73.html

The support is already implemented in Tasmota firmware - https://github.com/arendst/Sonoff-Tasmota/pull/2735

lrybak, 2019-07-26:
@OttoWinter do you think it's possible to implement this in esphome?

pepeEL, 2019-08-01:
Any chance.... ?

jarcikw, 2019-08-01:
I am interested too ..

glmnet, 2019-08-01:
Is it modbus?
I believe it should be a modbus hub then with modbus sensors.

lrybak, 2019-09-01:
Yes it is modbus

jarcikw, 2019-10-18:
@OttoWinter is there any chancce to implement SDM Meters ?

pepeEL, 2019-11-18:
Any news or chance ?

TheSDTM, 2020-01-20:
I have implemented Esphome custom component for SDM energy meters. Here you can find files and configuration example: https://github.com/TheSDTM/esphome-SDM-enery-meter

jarcikw, 2020-01-20:
That is gresy news :)

dcec, 2020-01-21:
@TheSDTM if i have 3 SDM on the same bus is possible to use your code?

TheSDTM, 2020-01-22:
> @TheSDTM if i have 3 SDM on the same bus is possible to use your code?

In this case you should modify code in sdm/sdm_esphome.h to use several devices. Actually, you need to define 2 more group of sensors, use almost the same code for reading, but specify slave ID in readval func. Take a look at library that I have used: https://github.com/reaper7/SDM_Energy_Meter#reading

jarcikw, 2020-02-15:
Hi @TheSDTM ,
I try to connect with SDM630 and I have probably some problem with communication
[image]
I have switched TX witth RX in config but without result

my config:
```
substitutions:
  name: korytarz_sdm630
esphome:
  name: korytarz_sdm630
  platform: ESP8266
  board: esp01_1m
  includes:
    - sdm/SDM_Config_User.h
    - sdm/SDM.h
    - sdm/sdm_esphome.h

uart:
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 9600
  stop_bits: 1
```

Previously i have configured with Tasmota and it was working fine 
[image]

Do I need to change GPIO configuration in some files?

jarcikw, 2020-02-15:
@OttoWinter  is there any chance to integrate (SDM meters) with Esphome ?

glmnet, 2020-02-17:
For all those who are pinging Otto, etc to get this implemented the best thing you can do is up vote (thumbs up) the first comment of this thread, this is how FR are later prioritized.

dudu2082, 2020-12-28:
> Hi @TheSDTM ,
> I try to connect with SDM630 and I have probably some problem with communication
> [image]
> I have switched TX witth RX in config but without result
>
> my config:
>
> ```
> substitutions: [...]

Hi guys.
I had the same problem like @jarcikw .
Also I received : "[D][sensor:092]: 'Voltage': Sending state nan V with 1 decimals of accuracy".

If I set:
tx_pin: tx (or GPIO1 - i use wemos d1 mini)
rx_pin:rx (or GPIO3)
and upload it the wemos can't connect to wifi.

When i set:
tx_pin: rx (or GPIO3)
rx_pin: tx (or GPIO1)
Then it works but values are "nan".
Here https://github.com/reaper7/SDM_Energy_Meter#reading i read problems part.
I try to change baud rate to 2400 (i change it also in SDM), change stop_bite to 2 (In problems it was one of option - I have SDM 630 v2).
I change those settings also in SDM_Config_User.h file and sdm_esphome.h.
Still I receive "nan" values. 
I use this converter https://www.aliexpress.com/item/4000281685430.html 
Rx from coverter is connected to TX of wemos d1 mini (GPIO1), Tx from converter is connected to Rx of wemos d1 mini (GPIO3).

@TheSDTM any idea where is the problem?
Modbus bus in correct connected to converter and SDM630.

I spend all day to figure it out.

reaper7, 2020-12-29:
@dudu2082 - try this custom component:
https://github.com/reaper7/esphome_sdm_custom

Attached example (powermetersdm.yaml) is a configuration that works for me (on ESP01)
and uses standard RX TX pins to communicate with the converter.
Remember that you have to change the baudrate and other settings (if needed) in:
https://github.com/reaper7/esphome_sdm_custom/blob/master/powermetersdm.yaml#L48
https://github.com/reaper7/esphome_sdm_custom/blob/master/custom_components/sdm/sdm_esphome.h#L5
https://github.com/reaper7/esphome_sdm_custom/blob/master/custom_components/sdm/SDM_Config_User.h#L24

dudu2082, 2020-12-29:
@reaper7 thanks. I tried. But it also doesn't work. If set tx_pin: TX and rx_pin: RX then wemos can't connect to wifi. I checked it also on wemos d1 mini pro with external antenna. 
But problem probably is somewhere else. Leds on converter doesn't blink. 
I tried separate power supplies, I powered converter by separate power supply, and also I test another converter (I bought 2 of them). Also without blink. Connection to SDM are correct. B- from converter to B- of SDM. A+ with A+. Ground with ground. 
Voltage between B- and A+ is 4,2 V. I powered Wemos and converter by 5 V power supply.
Any other ideas where should I look?

reaper7, 2020-12-29:
of course you changed user / pass in yaml ?

start with the basic version of esphome, which simply returns only it's own status on the website
then if it works, add more modules one by one

dudu2082, 2020-12-29:
@reaper7 of course I change my wifi and password.
It's normal start with those settings:
`[18:39:29][C][uart_esp8266:075]: UART Bus:
[18:39:29][C][uart_esp8266:077]:   TX Pin: GPIO3
[18:39:29][C][uart_esp8266:080]:   RX Pin: GPIO1
[18:39:29][C][uart_esp8266:081]:   RX Buffer Size: 256
[18:39:29][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[18:39:29][C][uart_esp8266:084]:   Data Bits: 8
[18:39:29][C][uart_esp8266:085]:   Parity: NONE
[18:39:29][C][uart_esp8266:086]:   Stop bits: 1
[18:39:29][C][uart_esp8266:090]:   Using software serial
[18:39:29][C][uptime.sensor:030]: Uptime Sensor 'SDM Uptime Sensor'
[18:39:29][C][uptime.sensor:030]:   Unit of Measurement: 's'
[18:39:29][C][uptime.sensor:030]:   Accuracy Decimals: 0
[18:39:29][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[18:39:29][C][logger:185]: Logger:
[18:39:29][C][logger:186]:   Level: DEBUG
[18:39:29][C][logger:187]:   Log Baud Rate: 115200
[18:39:29][C][logger:188]:   Hardware UART: UART0
[18:39:29][C][status:034]: Status Binary Sensor 'SDM Status'
[18:39:29][C][status:034]:   Device Class: 'connectivity'
[18:39:29][C][restart:021]: Restart Switch 'SDM Restart'
[18:39:29][C][restart:021]:   Icon: 'mdi:restart-alert'
[18:39:29][C][adc:026]: ADC Sensor 'SDM VCC'
[18:39:29][C][adc:026]:   Unit of Measurement: 'V'
[18:39:29][C][adc:026]:   Accuracy Decimals: 2
[18:39:29][C][adc:026]:   Icon: 'mdi:flash'
[18:39:29][C][adc:029]:   Pin: VCC
[18:39:29][C][adc:051]:   Update Interval: 60.0s
[18:39:29][C][total_daily_energy:022]: Total Daily Energy 'SDM Daily Energy'
[18:39:29][C][total_daily_energy:022]:   Unit of Measurement: 'kWh'
[18:39:29][C][total_daily_energy:022]:   Accuracy Decimals: 3
[18:39:29][C][total_daily_energy:022]:   Icon: 'mdi:counter'
[18:39:29][C][homeassistant.time:010]: Home Assistant Time:
[18:39:29][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[18:39:29][C][captive_portal:169]: Captive Portal:
[18:39:29][C][ota:029]: Over-The-Air Updates:
[18:39:29][C][ota:030]:   Address: 192.168.1.32:8266
[18:39:29][C][api:095]: API Server:
[18:39:29][C][api:096]:   Address: 192.168.1.32:6053
[18:39:29][C][wifi_signal.sensor:009]: WiFi Signal 'SDM WiFi Status'
[18:39:29][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[18:39:29][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[18:39:29][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[18:39:32][D][adc:056]: 'SDM VCC': Got voltage=3.10V
[18:39:32][D][sensor:092]: 'SDM VCC': Sending state 3.09766 V with 2 decimals of accuracy
[18:39:40][D][SDM:021]: Start reading
[18:39:42][D][SDM:030]: V=nan V, I=nan A, P=nan W, E=nan kWh, F=nan Hz, PF=nan
[18:39:42][D][sensor:092]: 'SDM Voltage': Sending state nan V with 1 decimals of accuracy
[18:39:42][D][sensor:092]: 'SDM Current': Sending state nan A with 3 decimals of accuracy
[18:39:42][D][sensor:092]: 'SDM Power': Sending state nan W with 1 decimals of accuracy
[18:39:42][D][sensor:092]: 'SDM Energy': Sending state nan kWh with 1 decimals of accuracy
[18:39:42][D][sensor:092]: 'SDM Frequency': Sending state nan Hz with 1 decimals of accuracy
[18:39:42][D][sensor:092]: 'SDM Power Factor': Sending state nan  with 2 decimals of accuracy
[18:39:42][D][SDM:039]: End reading
[18:39:46][D][sensor:092]: 'SDM Uptime Sensor': Sending state 59.69000 s with 0 decimals of accuracy
[18:39:51][D][sensor:092]: 'SDM WiFi Status': Sending state -73.00000 dB with 0 decimals of accuracy
[18:39:55][D][SDM:021]: Start reading
[18:39:57][D][SDM:030]: V=nan V, I=nan A, P=nan W, E=nan kWh, F=nan Hz, PF=nan
[18:39:57][D][sensor:092]: 'SDM Voltage': Sending state nan V with 1 decimals of accuracy
[18:39:57][D][sensor:092]: 'SDM Current': Sending state nan A with 3 decimals of accuracy
[18:39:57][D][sensor:092]: 'SDM Power': Sending state nan W with 1 decimals of accuracy
[18:39:57][D][sensor:092]: 'SDM Energy': Sending state nan kWh with 1 decimals of accuracy
[18:39:57][D][sensor:092]: 'SDM Frequency': Sending state nan Hz with 1 decimals of accuracy
[18:39:57][D][sensor:092]: 'SDM Power Factor': Sending state nan  with 2 decimals of accuracy
[18:39:57][D][SDM:039]: End reading`

I tried also invert rx and tx cables on wemos but still leds on cnoverter doesn't blink. 
Also if I powered converter by 3.3 V from Wemos board.

reaper7, 2020-12-29:
You have connected something to wemos usb? 
this port must be empty! connect only external 5V power supply to wemos 5V and GND

watch the logs only in the http console or cmd line

dudu2082, 2020-12-29:
Yes. Last attempt was with powered wemos from phone charger by usb.
When I powered it by pins on wemos I have this:
`[19:24:41][C][uart_esp8266:075]: UART Bus:
[19:24:41][C][uart_esp8266:077]:   TX Pin: GPIO3
[19:24:41][C][uart_esp8266:080]:   RX Pin: GPIO1
[19:24:41][C][uart_esp8266:081]:   RX Buffer Size: 256
[19:24:41][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[19:24:41][C][uart_esp8266:084]:   Data Bits: 8
[19:24:41][C][uart_esp8266:085]:   Parity: NONE
[19:24:41][C][uart_esp8266:086]:   Stop bits: 1
[19:24:41][C][uart_esp8266:090]:   Using software serial
[19:24:41][C][uptime.sensor:030]: Uptime Sensor 'SDM Uptime Sensor'
[19:24:41][C][uptime.sensor:030]:   Unit of Measurement: 's'
[19:24:41][C][uptime.sensor:030]:   Accuracy Decimals: 0
[19:24:41][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[19:24:41][C][logger:185]: Logger:
[19:24:41][C][logger:186]:   Level: DEBUG
[19:24:41][C][logger:187]:   Log Baud Rate: 115200
[19:24:41][C][logger:188]:   Hardware UART: UART0
[19:24:41][C][status:034]: Status Binary Sensor 'SDM Status'
[19:24:41][C][status:034]:   Device Class: 'connectivity'
[19:24:41][C][restart:021]: Restart Switch 'SDM Restart'
[19:24:41][C][restart:021]:   Icon: 'mdi:restart-alert'
[19:24:41][C][adc:026]: ADC Sensor 'SDM VCC'
[19:24:41][C][adc:026]:   Unit of Measurement: 'V'
[19:24:41][C][adc:026]:   Accuracy Decimals: 2
[19:24:41][C][adc:026]:   Icon: 'mdi:flash'
[19:24:41][C][adc:029]:   Pin: VCC
[19:24:41][C][adc:051]:   Update Interval: 60.0s
[19:24:41][C][total_daily_energy:022]: Total Daily Energy 'SDM Daily Energy'
[19:24:41][C][total_daily_energy:022]:   Unit of Measurement: 'kWh'
[19:24:41][C][total_daily_energy:022]:   Accuracy Decimals: 3
[19:24:41][C][total_daily_energy:022]:   Icon: 'mdi:counter'
[19:24:41][C][homeassistant.time:010]: Home Assistant Time:
[19:24:41][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[19:24:41][C][captive_portal:169]: Captive Portal:
[19:24:41][C][ota:029]: Over-The-Air Updates:
[19:24:41][C][ota:030]:   Address: 192.168.1.32:8266
[19:24:41][C][api:095]: API Server:
[19:24:41][C][api:096]:   Address: 192.168.1.32:6053
[19:24:41][C][wifi_signal.sensor:009]: WiFi Signal 'SDM WiFi Status'
[19:24:41][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[19:24:41][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[19:24:41][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[19:24:47][D][SDM:021]: Start reading
[19:24:49][D][SDM:030]: V=nan V, I=nan A, P=nan W, E=nan kWh, F=nan Hz, PF=nan
[19:24:49][D][sensor:092]: 'SDM Voltage': Sending state nan V with 1 decimals of accuracy
[19:24:49][D][sensor:092]: 'SDM Current': Sending state nan A with 3 decimals of accuracy
[19:24:49][D][sensor:092]: 'SDM Power': Sending state nan W with 1 decimals of accuracy
[19:24:49][D][sensor:092]: 'SDM Energy': Sending state nan kWh with 1 decimals of accuracy
[19:24:49][D][sensor:092]: 'SDM Frequency': Sending state nan Hz with 1 decimals of accuracy
[19:24:49][D][sensor:092]: 'SDM Power Factor': Sending state nan  with 2 decimals of accuracy
[19:24:49][D][SDM:039]: End reading`

dudu2082, 2020-12-29:
Of course I changed baud rate and stop bit in meter to 9600 and 1.

reaper7, 2020-12-29:
as I say...try with simple code, without sdm...get wifi connection first

STOP, You are using software serial ??
```
[uart_esp8266:090]: Using software serial
```

dudu2082, 2020-12-29:
What you mean Software serial? I connect converter to wemos. And SDM to converter.
Without connection to so SDM or even to converter if I set tx_pin: TX and rx_pin: RX then wemos can't connect to wifi.
If I set tx_pin: RX and rx_pin: TX then wemos can boot up and connect to wifi.

dudu2082, 2020-12-29:
I don't know if its important but on my converter is pullup resistor to Txd.

dudu2082, 2020-12-29:
Why on SDM_Config_User.h is:
#if defined ( ESP8266 ) || defined ( ESP32 )
    #define SDM_RX_PIN                        13
    #define SDM_TX_PIN                        15
I connected converter to GPIO 3 and GPIO1. Those GPIO-s are marked Rx and Tx. Should I connect it to GPIO 15  nad GPIO13?
Should I then replace pullup resistor from my converter?

reaper7, 2020-12-29:
as I wrote above, i use the same converter as-is

> Why on SDM_Config_User.h is:
> #if defined ( ESP8266 ) || defined ( ESP32 )
> #define SDM_RX_PIN 13
> #define SDM_TX_PIN 15
> I connected converter to GPIO 3 and GPIO1. Those GPIO-s are marked Rx and Tx. Should I connect it to GPIO 15 nad GPIO13?
> Should I then replace pullup resistor from my converter?

you are reading it wrong, do not be influenced by these settings, pay attention to the markers #if defined,
everything is set up correctly here.

But look at your log:
```
UART Bus: [18:39:29][C][uart_esp8266:077]: TX Pin: GPIO3 [18:39:29][C][uart_esp8266:080]: RX Pin: GPIO1 
```
I don't know why TX is a GPIO3 and RX is a GPIO1
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
TX should be a GPIO1 and RX GPIO3

try to set in yaml file real gpios number instead TX/RX
```yaml
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  stop_bits: 1
```

dudu2082, 2020-12-29:
>I don't know why TX is a GPIO3 and RX is a GPIO1 in Your settings
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
TX should be a GPIO1 and RX GPIO3

When I do that wemos don't want to connect to wifi.

On GPIO 3 and 1 leds on converter didn't blink.

Few minutes ago I tried:
uart:
  rx_pin: GPIO13
  tx_pin: GPIO15
  baud_rate: 9600"
  parity: none
  data_bits: 8
  stop_bits: 1

To GPIO 13 I connect tx from converter. To GPIO15 I connect rx from converter.
So connections are tx->rx and rx->tx. So this is correct.
But I need to add 100 kOhm resistor beetwen GPIO 13 and 3,3v.

Now wemos boots up, connect to wifi and rx led blinks on converter when wemos send "questions" to sdm.

On startup was this warning:
"[21:38:59][W][uart:041]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0."
So I set in logger baud rate to 0. 
Warning disappeared.

Unfortunately still I receive "nan" values and tx led didn't blinks.

`[21:43:26][C][uart_esp8266:075]: UART Bus:
[21:43:26][C][uart_esp8266:077]:   TX Pin: GPIO15
[21:43:26][C][uart_esp8266:080]:   RX Pin: GPIO13
[21:43:26][C][uart_esp8266:081]:   RX Buffer Size: 256
[21:43:26][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[21:43:26][C][uart_esp8266:084]:   Data Bits: 8
[21:43:26][C][uart_esp8266:085]:   Parity: NONE
[21:43:26][C][uart_esp8266:086]:   Stop bits: 1
[21:43:26][C][uart_esp8266:088]:   Using hardware serial interface.
[21:43:26][C][uptime.sensor:030]: Uptime Sensor 'SDM Uptime Sensor'
[21:43:26][C][uptime.sensor:030]:   Unit of Measurement: 's'
[21:43:26][C][uptime.sensor:030]:   Accuracy Decimals: 0
[21:43:26][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[21:43:26][C][logger:185]: Logger:
[21:43:26][C][logger:186]:   Level: DEBUG
[21:43:26][C][logger:187]:   Log Baud Rate: 0
[21:43:26][C][logger:188]:   Hardware UART: UART0
[21:43:26][C][status:034]: Status Binary Sensor 'SDM Status'
[21:43:26][C][status:034]:   Device Class: 'connectivity'
[21:43:26][C][restart:021]: Restart Switch 'SDM Restart'
[21:43:26][C][restart:021]:   Icon: 'mdi:restart-alert'
[21:43:26][C][adc:026]: ADC Sensor 'SDM VCC'
[21:43:26][C][adc:026]:   Unit of Measurement: 'V'
[21:43:26][C][adc:026]:   Accuracy Decimals: 2
[21:43:26][C][adc:026]:   Icon: 'mdi:flash'
[21:43:26][C][adc:029]:   Pin: VCC
[21:43:26][C][adc:051]:   Update Interval: 60.0s
[21:43:26][C][total_daily_energy:022]: Total Daily Energy 'SDM Daily Energy'
[21:43:26][C][total_daily_energy:022]:   Unit of Measurement: 'kWh'
[21:43:26][C][total_daily_energy:022]:   Accuracy Decimals: 3
[21:43:26][C][total_daily_energy:022]:   Icon: 'mdi:counter'
[21:43:26][C][homeassistant.time:010]: Home Assistant Time:
[21:43:26][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[21:43:26][C][captive_portal:169]: Captive Portal:
[21:43:26][C][ota:029]: Over-The-Air Updates:
[21:43:26][C][ota:030]:   Address: 192.168.1.32:8266
[21:43:26][C][api:095]: API Server:
[21:43:26][C][api:096]:   Address: 192.168.1.32:6053
[21:43:26][C][wifi_signal.sensor:009]: WiFi Signal 'SDM WiFi Status'
[21:43:26][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[21:43:26][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[21:43:26][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[21:43:39][D][SDM:021]: Start reading
[21:43:41][D][SDM:030]: V=nan V, I=nan A, P=nan W, E=nan kWh, F=nan Hz, PF=nan
[21:43:41][D][sensor:092]: 'SDM Voltage': Sending state nan V with 1 decimals of accuracy
[21:43:41][D][sensor:092]: 'SDM Current': Sending state nan A with 3 decimals of accuracy
[21:43:41][D][sensor:092]: 'SDM Power': Sending state nan W with 1 decimals of accuracy
[21:43:41][D][sensor:092]: 'SDM Energy': Sending state nan kWh with 1 decimals of accuracy
[21:43:41][D][sensor:092]: 'SDM Frequency': Sending state nan Hz with 1 decimals of accuracy
[21:43:41][D][sensor:092]: 'SDM Power Factor': Sending state nan  with 2 decimals of accuracy
[21:43:41][D][SDM:039]: End reading` 

SDM set like yaml:
9600
8 bits
parity: none
stop bit: 1

SDM should now send some data.
Why still I have "nan"?

reaper7, 2020-12-29:
maybe SDM (in SDM630) ADDR is not 001 ?
I have no more ideas...

P.S.
> "[21:38:59][W][uart:041]: You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0."
> So I set in logger baud rate to 0.
in my first post and in the examples I gave from the beginning, there was such a parameter

dudu2082, 2020-12-29:
Address is 001.
If I swapp wires (connect RX to RX and tx to tx) then wemos can't connect to wifi and one of leds on converter is turn on constantly.

reaper7, 2020-12-29:
turn on constantly possibly because of the your pull-up resistor?
whatever ... I don't understand why it doesn't work on standard 1/3 pins, my example is adapted to these pins

well, if you decided on those pins 13/15 
then you have to change line:
https://github.com/reaper7/esphome_sdm_custom/blob/master/custom_components/sdm/SDM_Config_User.h#L68
from
```
  //#define SWAPHWSERIAL                      0
```
to
```
  #define SWAPHWSERIAL                      1
```

and https://github.com/reaper7/esphome_sdm_custom/blob/master/custom_components/sdm/sdm_esphome.h#L5
from
```
  SDM sdm = SDM(Serial, 9600, NOT_A_PIN, SERIAL_8N1, false);
```
to
```
  SDM sdm = SDM(Serial, 9600, NOT_A_PIN, SERIAL_8N1, true);
```

dudu2082, 2020-12-30:
Still nothing.
Today I use another ESP (NodeMcu v3). 
I connected converter to RX and TX. To custom_component i put fresh files from your github. 
I compile and upload it.
Now I have blinking only 1 led. If I swap wires second led blinking but 1 is off.  On wemos d1 leds was off.
In both versions I have "nan" values.
I think that I could have:
 - broken 2 converters
 - or modbus in SDM is broken.
 Meter is brand new from Aliexpress.
I am waiting for another meter with modbus (no sdm).
I will connect to it and will see if both leds will blinking.

nagyrobi, 2022-06-29:
See https://github.com/KinDR007/SDM630-ESPhome

closed 2022-06-29 by nagyrobi